vvip-68/GyverPanelWiFi #1: "Погода" в Украине работать будет?

80Stepko08, 2020-08-30:


vvip-68, 2020-08-30:
Понятия не имею. Пробуйте.
Зайдите на страничку yandex.ru 
В строке поиска наберите чего-нибудь, нажмите кнопку "Искать". Откроется страничка с результатами поиска.
Смотрите что в адресной строке браузера. Там что-то типа
https://yandex.ru/search/?lr=62
62 - это код региона для которого яндекс отдает погоду. В Украине что-то возвращается?
Далее  этот код (в данном случае 62) подставляется в этот запрос:
https://yandex.ru/time/sync.json?geo=62
Если в ответ получено что-то вроде этого - 
{"time":1598775715587,"clocks":{"62":{"id":62,"name":"Красноярск","offset":25200000,"offsetString":"UTC+7:00","showSunriseSunset":true,"sunrise":"05:49","sunset":"19:47","isNight":false,"skyColor":"#68bff7","weather":{"temp":18,"icon":"bkn-d","link":"https://yandex.ru/pogoda/krasnoyarsk"},"parents":[{"id":11309,"name":"Красноярский край"},{"id":225,"name":"Россия"}]}}}
погода работать будет. Если нет - не будет.

vvip-68, 2020-08-30:
Попробовал через Tor зайти. Сказала, что я в германии, код региона - 115326.
Погоду вернуло. Думаю, что для Украины тоже будет работать

closed 2020-08-30 by vvip-68