vvip-68/GyverPanelWiFi #2: Матрица 26х14

serg-74, 2020-08-31:
Здравствуйте.
Прочитал, что проект "GyverMatrixWiFi", который я использовал устарел и Вы рекомендуете переходить на данный проект. Будет ли проект работать с матрицей 26х14, и если да, то с какими ограничениями? ("Также подходит с некоторыми ограничениями для матриц с размерами 16x16")

vvip-68, 2020-09-01:
Если 26 это ширина - будет нормально работать. Ограничения в 16х16 связаны с тем, что часы на таком размере можно отобразить только шрифтом 3x5 - для него требуется 15 колонок. Для крупного шрифта 5x7 требуется вроде бы как раз 26 колонок (как я посчитал). Так что для такого размера - ограничений нет.

closed 2020-09-01 by vvip-68

fastdron, 2020-10-25:
Оооо, как раз одна колонка не влазит - где в коде переписать часы ? Вы извините меня, я в этом деле новичек и без Вашей помощи не справлюсь - не могли бы вы исправленный кусочек кода например тут в комент выложить?

vvip-68, 2020-10-25:
файл clock.ino процедура DrawClock() - строки 284-291 - убрать отрисовку колонки x+13
Далее в строках 295 и 291 сместить позицию на одну влево