vvip-68/GyverPanelWiFi #3: Предложение

serg-74, 2020-09-02:
Есть предложение на какой-нибудь пин ESP вывести управляющий сигнал для подачи питания на матрицу, т.к. даже когда матрица выключена, то светодиоды продолжают потреблять ток, а т.к. матрица большая, то и потребляемый ток уже не маленький. Питать матрицу можно через мощный полевой транзистор (MOSFET), также можно использовать компьютерный БП (ESP повесить на 5В дежурного режима, а управляющим сигналом включать БП через зеленый провод PS-ON). В настройках можно предусмотреть константу, чем будет включаться матрица "0" или "1" (для упрощения схемотехники).

vvip-68, 2020-09-02:
Реализовать сможете? Схему подключения нарисовать?
Готовьте все, оформляйте Pull request. Какой транзистор нужен, скажем на 1000 диодов (у меня такая матрица). Как подключить?

vvip-68, 2020-09-03:
Хорошо, я добавил вывод управляющего сигнала вкл/выкл питания матрицы
Файл a_def_hard.h

- Задается возможность использования / не использования этой фичи, а также назначение управляющего пина
#define USE_POWER 1           // 1 - использовать отключение питания матрицы через MOSFET; 0 - не использовать 
#define POWER_PIN D1         // D1 управляющий пин вкл/выкл матрицы через MOSFET

Файл a_def_soft.h, стр 571, 572 - назначение уровня сигнала на пине для вкл / выкл питания
#define    POWER_ON              HIGH            // Для включения питания матрицы (через MOSFET) подавать на пин POWER_PIN высокий уровень
#define    POWER_OFF             LOW             // Для вЫключения питания матрицы (через MOSFET) подавать на пин POWER_PIN низкий уровень

Основной файл скетча, строки 71-73 - инициализация пина на выход
  #if (USE_POWER == 1)
    pinMode(POWER_PIN, OUTPUT);
  #endif

a_main.ino - стр 214-222 - подача управляющего сигнала на пин в зависимости от состояния флага isTurnedOff

Но мне попрежнему нужна схема включения, а так же номиналы резистора и обвязки, а так же тип MOSFETа, способного выдержать ток 30-60 ампер и полностью включающегося уже при напряжении 4 вольта.

serg-74, 2020-09-03:
В похожем проекте я использовал компьютерный БП (дежурный режим 5В на ESP, питание матрицы от рабочих 5В с управлением по PS-ON).
На дружественном форуме нашел вот такую схему на IRF4905 - https://community.alexgyver.ru/threads/wifi-lampa-budilnik-obsuzhdenie-proekta.1411/post-32877. Сам не проверял, но пишут, что работает. Транзистор IRF4905 по даташиту 74А при 25 градусах, 52А - при 100 градусах.

closed 2020-09-04 by vvip-68

vvip-68, 2020-09-04:
А тут беда вот в чем - для надежного открытия ключа нужно 4.5 В или более. А Из за согласования уровня логики ленты и ESP8266, который работает на логике 3.3 вольта питание ленты и контроллера желательно не более 4.5 вольт, иначе лента может глючить. Как-то все слишком на грани... Недооткроется ключ при нормальном для ESP питании - будет греться и сгорит. Дашь больше питания - начнутся глюки на ленте.

serg-74, 2020-09-04:
На сколько я понял, по даташиту для открытия надо 4В (что в принципе тоже на грани). Кстати, для согласования логики я использовал схему с этого же форума, т.к. схема согласования с полевым транзистором у меня не работала - глюки хоть и меньше, но все равно были. Если использовать эту схему - то на матрицу можно подавать 5В и ключ должен надежно отпираться.
[image]

vvip-68, 2020-09-04:
А объяснить можешь как это работает? 
Спойлер - у меня не работает. И вот почему. 
На саму ноду можно подавать аж до 9 вольт (кажется) и контроллер будет работать Можно и 4 вольта подать - и тоже будет работать. И при любом входном напряжении уровень логики на выходных пинах будет 3.3 вольта.
Проблема в том, что чем бОльшее напряжение питания подается на ЛЕНТУ - тем выше напряжение логич. 1Ю которую ожидает лента. То есть подав на ленту (и на контроллер) 4.4 вольта, лента будет воспринимать на DIN входе  уровень 1 уже с 3.1 вольта. Повышая питающее напряжение на ленте до 5 вольт, уровень 1 на входе уже ожидается 3.3 в, а это максимум что может дать контроллер. Повысив до 5.25 вольт питание на ленте она на DIN будет ожидать уровень логической 1 примерно 3.5-4 вольта, что уже не может выдать контроллер. Отсюда глюки на матрице.
Таким образом при подаче питания 5В напрямую на ленту и ограничивая диодом питание на ESP - ничего не дает. Несогласование уровня логич 1 остается - будут глюки. Диод бессмысленен. Вероятно можно ограничивать питание всей системы через диод (который должен держать ток 30-60А) - но проще просто с БП не давать выше 4.8 вольта.

serg-74, 2020-09-04:
Суть в диоде, в таком включении, масса матрицы "опускается" относительно массы контроллера на 0.6 вольта. Соответственно, уровень единицы контроллера, относительно массы матрицы, получается 3.3 + 0.6 ~= 3.8 -3.9 вольта.
Получается уровень "1" уже будет в допустимых пределах и уровень "0" еще будет в допустимых пределах.
У меня реально не стало глюков при питании от 5В.
Более подробно можно почитать на форуме:
https://community.alexgyver.ru/threads/wifi-lampa-budilnik-obsuzhdenie-proekta.1411/post-39457

vvip-68, 2020-09-04:
Хм... Именно такая схема (диод на минусе) у меня заработала. Сейчас проверил - почти до 6 вольт поднимал напряжение питания - работает без глюков. Вот тут уже можно и ключ ставить - будет работать. Сейчас с ключом поэкспериментирую...

fastdron, 2020-10-24:
Использую вместо мосфета обычный релейный модуль с АлиЭкспресс , все работает как часы

vvip-68, 2020-10-25:
Ну, мне просто еще не пришел. Али такой медлительный... А по сути там все то же самое - оптронная развязка и реле

fastdron, 2020-10-27:
> Ну, мне просто еще не пришел. Али такой медлительный... А по сути там все то же самое - оптронная развязка и реле

По сути- да, одно и то-же,с той лишь разницей что нет необходимости "доставать паяльник" 
Набросал по быстрому схему подключения,которую использую .
[image]

fastdron, 2020-10-27:
https://a.aliexpress.com/_9G7adr это ссылка на реле (выбираем 5в)

vvip-68, 2020-10-27:
А вот такой шит не сподручней будет? Специально для Wemos D1 Mini сделан - матрешкой садится.
https://aliexpress.ru/item/32859624847.html

fastdron, 2020-10-27:
Даже не знал о его существовании) спасибо за наводку. использовал просто то- что было в наличии. Сейчас закажу и такой, спасибо

fastdron, 2020-10-27:
А как этот шит определяет управляющую ногу? или там нужно вручную паять перемычку?

vvip-68, 2020-10-27:
Так он еще не пришел. Описания нигде не нашел. Но раз он специально разработан для матрешки - думаю и управляется и по уровням логики согласован. Вероятнее всего управляющий сигнал просто разведен на какую-то ногу и надо в прошивке указать этот пин вместо D1, если не совпадают

vvip-68, 2020-10-27:
Судя по крупной фотке - дорожка как раз с D1 идет 
[image]

fastdron, 2020-10-27:
да, по фото видно что D1. ну напишите ,когда дойдет...тоже прикуплю,еще раз спасибо за совет и вообще за ЛУЧШУЮ реализацию проекта!!!!!

fastdron, 2020-10-28:
[image]
забыл кнопку на схему добавить)