vvip-68/GyverPanelWiFi #4: Предложение №2

80Stepko08, 2020-09-07:
Есть такая штука, как субпиксель, на форуме https://community.alexgyver.ru/threads/wifi-lampa-budilnik-proshivka-firelamp_jeeui-gpl.2739/ Он делает эффект плавнее и красивее

vvip-68, 2020-09-08:
Что-то по приведенной ссылке ничего конкретного не нашел, а перечитывать все 128 страниц темы - как-то не особо...

80Stepko08, 2020-09-08:
https://github.com/DmytroKorniienko/FireLamp_JeeUI/tree/dev

vvip-68, 2020-09-09:
Посмотрел. Утянул процедуры отрисовки субпиксельной графики. На будущее.
Пока же у меня я не вижу куда это применить. Часть эффектов исходной лампы где это используется у меня отсутствует.
Часть эффектов что у меня есть и я видел на роликах в форуме как они выглядят с субпикселями (те же стрелки) - мне не нравится такой вариант. К тому же именно пиксельность придает некоторый шарм, от которого мне не хотелось бы избавляться. Но за идею - спасибо.

closed 2020-09-09 by vvip-68