vvip-68/GyverPanelWiFi #5: ESP32

UrbanCraftz, 2020-09-15:
Доброго дня
Никак не получаеться скомпилировать ваш проект на есп32
Пробовал v 1.03 и 1.04
Плата:ESP32 Dev Module

Ошибка: exit status 1
statement has no effect [-Werror=unused-value]

Ругаеться на строку:

` if (timeToGetWeather) { weather_cnt = 0; weather_t == 0; refresh_weather = true; }`
если убрать 1 знак =,
строчку компилит,но упирается в
`          showTextNow && (specialTextEffect >= 0))    // Воспроизводится бегущая строка на фоне указанного эффекта
`
exit status 1
suggest parentheses around '&&' within '||' [-Werror=parentheses]
 
Пробовал компилировать под другие варианты плат,ничего не меняеться
Менеджер плат обновлен,все библиотеки из папки проекта.

vvip-68, 2020-09-15:
Ну, с 'weather_t == 0' тут явный косяк, должно быть одинарное '=';
У вас в настройках компилятора (комбобокс в настройках "Сообщения компилятора") стоит скорее всего "Подробно", что включает опции компиляции "Трактовать предупреждения как ошибки". Поставьте там "Ничего".
Я в своем коде где мог поправил такие предупреждения (обновите код), но пока не измените эту опцию - компилиться не будет, ибо ругается на предупреждения как в сторонних библиотеках, так в и ядре самого esp32? а там уже всё это исправить я не могу.

vvip-68, 2020-09-15:
[image]

closed 2020-09-16 by vvip-68

UrbanCraftz, 2020-09-16:
Спасибо огромное,все компилируеться без проблем.