vvip-68/GyverPanelWiFi #6: How can I edit the GyverPanelWiFi.aia file ?

dorinik, 2020-09-22:
Congratulations on your work!
I want to edit the GyverPanelWiFi.aia file.
I tried with MIT app inventor, but without success ...
How can I solve this problem?
Thanks!

yusufsulaiman, 2020-09-23:
@dorinik u can use thunkableclassic

vvip-68, 2020-09-23:
@yusufsulaiman, yes, you right. It is application for Thunkable Classic.
@dornic, for edit this aia file in Thunkable Classic you MUST have account in this system. However registration of new user was closed by owners a couple year ago. Only old-registered users can use this service.
Second issue is - application reach its maximal size. Addition 1-2-3 new block into application makes application stop to compile. Compiler begin retrieve an error of compilation.

dorinik, 2020-09-23:
Thank you for the answers !
I don't want to add new blocks, I just want to translate messages from Russian into English.
Please convert the GyverPanelWiFi.aia file to the format recognized by the MIT app inventor, because I don't have an old account on Thunkable Classic.
Thanks!

vvip-68, 2020-09-23:
Do you know a way to convert aia-file from Thinkable to MIT? I know nothing about this feature / tool.
Ok, because size of application reach limit, which do not allow add new features, I can try to translate labels in UI of application to English. But you know, part of russian words come from device from firmware - list of effects for example...

dorinik, 2020-09-23:
I'm already working on translating the firmware into English.
Please translate all the words that appear in the application in Russian into English.

dorinik, 2020-09-23:
vvip-68, can you send me the keystore exported from Thunkable?

vvip-68, 2020-09-23:
I will update project after completion. aia and apk files. I think it will be done before today night.
Could you please after that check the translation and take screenshots with inapropriate translation and send them to me.
English is not native language for me. So, translation can be tampered by sense...

dorinik, 2020-09-23:
Thanks a lot !

vvip-68, 2020-09-23:
I`ve commited English version of APK file.
Please check it and send me corrections for wrong translation

dorinik, 2020-09-23:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

vvip-68, 2020-09-23:
1. I used Full HD screen of my smartphone (1920x1080). I tried to fix sizes of elements, but not sure that it will normally viewed on screens with smaller resolution.
2. I corrected missprints in "Effects" page - "Use effect in demo mode" and "Text over effect" (and "Вариант" as label of combobox.
3. Other texts in cyrillic (content of comboboxes and checkbox label) are coming from firmware. You said you are changing code of firmware by yourself.
a_def_soft.h - 20-29 - list of effect names + line 33 -"Марио" - name of animation picture
a_main.ino - 1171-1821 - content of combo box 'Kind of" for effects parameters and checkbox label "Ceгменты"

Do you need help with translation of these strings?

dorinik, 2020-09-23:
The screenshots are from a Samsung SM-A805F (2256x1080).
The translation of the firmware in English is not completed, I still have work ...
Thanks for your patience and time for the translation!
Respect!

dorinik, 2020-09-23:
In a_def_soft.h lines 20-28 - are changed to English, only "escaped" line 33 - "Марио"

vvip-68, 2020-09-23:
Probably you use increased font size. Screenshots look as coming from hot HD screen, max - 720 pix...
I use Samsung SM-J730FM with 1920x1080, and text looks more small then yours on standard (unchanged) settings and there was no issue with screen elements sizes and positioning...

dorinik, 2020-09-23:
With the changes you made, it fits perfectly on the screen now!

dorinik, 2020-09-23:
In order not to take up space, the attached print screens (432x960) are reduced compared to the initial resolution (1080x2400)

vvip-68, 2020-09-23:
I told not about real size of pics, I told about proportions of text size with comparison of size of screen. I`ve got similar picture as on your screenshots when I changed (increased) font size in phone settings up to 2 grade over standard settings for font size (seems the ingrement is 150% over normal / ordinal size)

dorinik, 2020-09-23:
All that's left is this:
[image]

vvip-68, 2020-09-23:
Updated.

closed 2020-09-24 by vvip-68

Fab8888, 2022-08-16:
vvip-68, would you mind to send me the translate English version app inventor .aia file and apk file, I use much time to translate it, but it's quite difficult to translate the effect page drop down menu, I really don't know how to find the list words, thanks

vvip-68, 2022-08-16:
@Fab8888 , dorry I have no time to do that at all.
List of effects comes from sketch from here:

[image]

Fab8888, 2022-08-16:
Thank you so much, I almost done for the translation, but I have a litter bit cannot do that very well, such as setting page the  header  "Matrix" 16 x16,  and the effects page Patterns Option drop down menu,  but it is a minor things.
By the way, really thanks.

vvip-68, 2022-08-16:
@Fab8888 , Matrix 16x16

[image]

[image]

Patterns effect names:

[image]

vvip-68, 2022-08-16:
@Fab8888 , could you share your work (aia + apk) for other users, when it is completed?

Fab8888, 2022-08-17:
Really thanks, pretty sure, I can share it.
https://tmpfiles.org/dl/12481/gyverpanelwifi.zip

One more question, if I want to send some Chinese words into the text page. Should I change the fonts.h from sketch ? 
or the app have any drawing function to immediately to send the MCU Matrix LED? Just like Divoom pixel.

vvip-68, 2022-08-17:
@Fab8888 , - File not found (404)
Yes fonts.h contains bitmap image of each character of used font. 
There are 3 set of letters (latin and cyrillic) for char image size 5x7, 8x13 and 10x16 points.
If you want have other-language characters, you need draw it into bitmap arrays and probably change some procedures in runningText.ino sketch like getFont(), drawLetter() and some other...

vvip-68, 2022-08-17:
Here is some information about fonts
https://github.com/vvip-68/GyverPanelWiFi/wiki/Большие-буквы-в-бегущей-строке

Text in russian, but you may use google-translate to translate it into your language.

Fab8888, 2022-08-19:
I upload again, this time in sendspace. If can't download, tell me again.
https://www.sendspace.com/file/qsxmiy

I just find some features do not include in GyverPanelWiFi.
I need the draw function. The GyverMatrixWiFi have it.
https://github.com/vvip-68/GyverMatrixWiFi/tree/master/Android
but there's aia file only supported in thunkable classic version. (Thunkable classic cannot access now)
Do you have any way to let me adjust the aia file? thanks.

vvip-68, 2022-08-30:
@Fab8888 , drawing and game application  for GyverPanelWiFi project is located in the same folder and named GyverPlayerWiFi.aia. Developed in AppInventor. Application was separated into two parts, because of big size of resulted code, that made a) very slow response in editor. b) Inventor compiler sometime returned compilation error 'file size too big'.

Fab8888, 2022-09-01:
Thanks for your reply.
One more question, when I use GyverPanelWiFi and the firmware, it is always dimming to light off, I need to increase the brightness bar. However, it is do the same things again and again.

When I use GyverMatrixWiFi and the firmware, it's no problem. Do you know what is wrong about that?

vvip-68, 2022-09-01:
@Fab8888 , yes it is known effect.
Most possible reasons:
1. Wrong initialization of of button (wrong type of button - initialization for tact button when really used sensor button or vise verse) or sketch tuned for button usage, but really no button contacted.
2. Usage of wrong version of MCU core, instead of recommended. Some versions of core mapped 2/D2 or 4/D4 to same pin out, so button and lead out mapped to same pin and output led signal on out pin reads by sketch as input button 'hold' signal that used by sketch as change brightness. Same effect also may occurs when user manually set output led to the same pin as button input pin.

Fab8888, 2022-09-04:
Thanks, I fixed the pin.
May I save a word to be a starting demo, and loop to play.

ChamakChy, 2023-05-30:
@Fab8888 can you please upload the translated .apk file the link given shows as expired.

vvip-68, 2023-05-31:
I have only this things, this is in free access

[image]