vvip-68/GyverPanelWiFi #8: Возможность управления через REST

shostko, 2020-09-29:
Доброго дня.

Было бы замечательно, если бы прошивка поддерживала управление не только через сокет (как я понял по коду), но и через команды отправленные по REST. 
Это бы добавило возможностей по использованию прошивки. В частности для меня это бы позволило настраивать управление через "сервер умного дома" и управлять матрицей через Алису голосом.

Например `POST /set?cmd=$8_0_N`
Тут я пробел заменил на подчеркивания для удобства, в теории конечно можно и пробел заенкодить.
Так же через `;` можно было бы отправлять сразу набор команд.
Можно принимать и `GET` запросы для удобства тестирования в браузере.

PS. Нечто подобное реализовано в [WLED](https://github.com/Aircoookie/WLED/wiki/HTTP-request-API), которую я сейчас использую в некоторых местах.

PPS. Сам не большой спец в программировании под ардуино/есп, но чем смогу - готов помочь.

vvip-68, 2020-09-29:
Мысль интересная, но не первоочередная. Алисы у меня нет, с браузера управлять МНЕ не удобно / не пригодится. Была мысль сделать управление через MQTT, но это сначала надо анроид-программирование освоить, чтобы написать чисто андроидное приложение с поддержкой протокола. Вот когда всё это освою, можно и про REST подумать. Но, если сделаете что - оформляйте Pull Request, рассмотрим. Если вещь рабочая - добавим.

kostyamat, 2020-09-29:
Тут это есть https://github.com/DmytroKorniienko/FireLamp_JeeUI/tree/dev пока в ветке dev.
Обсуждение идет тут https://community.alexgyver.ru/threads/wifi-lampa-budilnik-proshivka-firelamp_jeeui-gpl.2739/
А для совсем новичков бинарная прошивка тут https://community.alexgyver.ru/threads/wifi-lampa-budilnik-proshivka-firelamp_jeeui-gpl.2739/page-131

shostko, 2020-09-30:
> Была мысль сделать управление через MQTT
меня бы такой вариант так же устроил, но имхо усилия на реализацию значительно большие, чем на REST

> но это сначала надо анроид-программирование освоить, чтобы написать чисто андроидное приложение с поддержкой протокола. 
А вот тут я смогу помочь, это собственно моя профессия =)

vvip-68, 2020-09-30:
> меня бы такой вариант также устроил (какой умный дом не умеет в MQTT?).
но имхо усилия на реализацию значительно большие, чем на REST

Зато управление из любой точки, а не так, чтобы либо в локальной сети только, либо белый IP на роутере иметь и порты пробрасывать.  По андроиду - будут вопросы, обращусь. Котлин используете?

shostko, 2020-09-30:
> По андроиду - будут вопросы, обращусь. Котлин используете?

Обращайтесь. Конечно, котлин - основной и офф язык для андроид разработки сейчас.
 
> А может - давайте так: - я разбираюсь с REST на ESP, поднимаю сервер, делаю обработку и т.п, а вы пишете базу приложения под Андроид на котлин, которая управляет текущей версией уйтройства. Потом я беру за основу ваше приложение, разбираюсь с ним как с примером, добавляю поддержку MQTT и далее развиваю с добавлением новых фич, когда появятся...
> 
> Ну, или не REST, а - я добавляю поддержку MQTT в прошивку, а вы пишите приложение под андроид которое будет этим управлять через MQTT. Ну, как?

Написать приложения с управлением и на REST, и через MQTT, через сокет для меня не будет проблемой. Только тогда я бы попросил обрисовать необходимый функционал и UX. Т.к. сам им пользоваться вряд ли буду - не хочется придумывать ради придумывания, лучше использовать реальный опыт реального пользователя.

> А я смотрю, у вас проектах уже есть управление LED на ESP8266 через MQTT
> Так в чем проблема была сюда поддержку добавить?..

не то, что бы это было проблемой,  за какой-то промежуток времени я бы добавил поддержку конечно. но в вашем проекте довольно много настроек и нюансов, вам наверняка будет быстрее внести изменения, чем мне во всем разобраться. 

а в том форке у меня только добавлена библиотеку WiFiManager (в оригинале у bruchautomation настройки сети хардкодились в прошивке), непосредственно MQTT я там не настраивал.

vvip-68, 2020-09-30:
> Написать приложения с управлением и на REST, и через MQTT, через сокет для меня не будет проблемой. Только тогда я бы попросил обрисовать необходимый функционал и UX. Т.к. сам им пользоваться вряд [...]

Ну, придумывать всё равно придется. Можно, конечно, повторить убогий интерфейс андроид-приложения из проекта (см. apk файл в комплекте проекта в в папке Android), которое собрано из блоков конструктора в Thunkable, тогда придумывать ничего не нужно. Но в силу ограничений конструктора - внешний вид соответствует возможностям этого Thunkable. Наверняка при "свободном" программировании можно и получше сделать. 

А вообще, все это похоже будет долгий разговор, если возьмемся за такою доработку. И тут делать длиннопост уже будет неудобно. Может - другой канал связи, например в VK? Тогда личными сообщениями сюда можно: https://vk.com/iivanovii. Или предложите другой способ - Viber, etc...

closed 2020-10-01 by vvip-68

vvip-68, 2020-12-13:
Добрый день, Сергей
Удалось ли вам интегрировать всё это хозяйство в умный дом?
Все ли нормально и удобно с API для MQTT?
А то тут люди делают интеграцию в Home Assistant - говорят не хватает некоторых возможностей - например отправки актуального состояния в отдельный топик при загрузке матрицы и при изменении параметра. Еще ошибки нашли - типа сбрасывается авторежим после отправки команды получения актуальных настроек $6 7|
Если у вас все получилось и работает - не могли бы вы написать инструкцию - как интегрировать все это дело в Алису?
И мне и другим кто повторяет проект думаю было бы полезно. Я бы инструкцию в wiki выложил...

shostko, 2020-12-14:
Добрый день.
Да, но я не интегрировал прям весь функционал, только минимально необходимое мне на ближайшее время.
Я так же использую Home Assistant (Алиса просто выступает еще одним способом вызовы команд), в моем случае не потребовалось вносить изменения в прошивку, мне хватило информации приходящей в топике NFO. но вполне допускаю, что чего-то может не хватать и делать опрос из умного дома с определенной периодичностью не лучшая идея.

Про ошибку вижу, что уже разобрались.

Насчет инструкции - думаю пока рановато, но своей текущей реализацией вполне могу поделиться. 
(она у меня пока не закоммитана, но для такого дела выложу WIP)

vvip-68, 2020-12-14:
Спасибо за предложение. Но я в этом ничего не понимаю. Без инструкции все равно непонятно что с реализацией делать.
Будет готовый конечный вариант - вот тогда и инструкцию бы, и пример реализации (ссылка на доступный репозиторий). А так без инструкции - просто ради любопытства посмотреть что это такое...

shostko, 2020-12-14:
вот мой конфиг на данный момент: https://github.com/shostko/home/blob/4401eaa85280b38de2fccf92f209564abef7f3f2/packages/balcony_panel.yaml

тут только несколько скриптов для отключения гирлянды, смены режима manual/auto (это без обратной связи, для чего у вас просили добавления)
и еще возможность переключения режимов в ручном режиме через выпадающий список (тут уже обратную связь я сделал, через топик NFO)

[image]

vvip-68, 2020-12-14:
В общем, похоже - требуется взаимодействие. Поскольку я не знаю чего именно нужно. Вот только пожелания добавить отправку изменения состояния параметров...

shostko, 2020-12-14:
В данном случае у меня даже не интеграция, а просто конфиги и автоматизации на голый Home Assistant (ну и + mqtt).
Да, нужно будет коммуницировать для полноценной интеграции.

В целом обычно при работе с MQTT предполагают, что устройство будет само обновлять значение определенного топика, при смене каких-то данных. Поэтому часто используются отдельный топик для команд и отдельный для обновления состояния (обратной связи). Подписчиков может быть больше одного в момент времени, и если первый меняет какие-то параметры, второй тут же получает эти изменения из топика stat.
Это если совсем коротко