vvip-68/GyverPanelWiFi #9: Не компилируется при активации mp3 и ТМ1627.

Geonm778, 2020-10-08:
Не могу скомпилировать при активации проигрывателя DFPlayer mp3 и TM1627 (версия 1.05). Выдается много ошибок. Например
C:\GyverPanelWiFi_v1.05\GyverPanelWiFi_v1.05.ino: In function 'void setup()':
GyverPanelWiFi_v1.05:90:16: error: 'loadSettings' was not declared in this scope
   loadSettings();
И т.д. Проект больше не предполагает работу с лампой 16х16, mp3 проигрывателем и дисплеем? Просто схема с ними в проекте присутствует. При этом без mp3 и ТМ1627 компиляция проходит нормально.

vvip-68, 2020-10-08:
Там была ошибка - не убрал отладочный код, он не давал скомпилиться при включенном MP3. Исправил.
Еще вылез косяк с отрицательными температурами. Тоже исправил.
Возьмите последний коммит, там всё исправлено - должно собираться

closed 2020-10-08 by vvip-68

Geonm778, 2020-10-08:
Прошу прощения, но нет. Не компилируется. Устанавливаю вариант часы на столе (2), устанавливаю необходимые параметры матрицы, прописываю
#define USE_SD 0
#define USE_MP3 1             
#define USE_TM1637 1
#define USE_WEATHER 0        
#define USE_POWER 0          
#define USE_MQTT 0 
#define SRX D4              
#define STX D3               
#define DIO D5                
#define CLK D7               
 и при попытке скомпилировать вылетает ошибка: 'loadSettings' was not declared in this scope

vvip-68, 2020-10-09:
Geonm778, у меня с вашими параметрами собирается без проблем.
" 'loadSettings' was not declared in this scope" - это всего лишь последняя ошибка в списке, причем порожденная другими ошибками. Приведите весь лог со всеми ошибками.
И - параметры с "часы на столе" - это для моей конкретно поделки, с моим набором оборудования, углом подключения и размером матрицы. Вам нужно все параметры все равно под себя настроить / проверить, что это все соответствует вашему изделию. Впрочем, думаю вы все это и так понимаете. Просто, на всякий случай...

Geonm778, 2020-10-09:
Добрый вечер! Компиляция заканчивается ошибкой, если указать что матрица свернута в трубу (а у меня как раз лампа!  Т.е.  файле a_def_hard.h если будет в настройках ЛЮБОГО варианта матрица свернута в трубу:

#define DEVICE_TYPE 0         // Использование матрицы: 0 - свернута в трубу для лампы; 1 - плоская матрица в рамке   

то все, ошибка при компиляции!

vvip-68, 2020-10-10:
Geonm778, исправил

Geonm778, 2020-10-10:
Отлично! Спасибо!