vvip-68/GyverPanelWiFi #10: Протокол обмена

frusa, 2020-10-10:
Добрый день, не нашел где с вами связаться, кроме как тут. Я решил сделать wifi панель от AlexGyver, и ваша реализация мне очень понравилась. Делаю я не для себя, а для знакомого, у которого Apple Iphone, из-за чего он не может полностью пользоваться вашей реализацией. В связи с этим я хотел бы попросить у вас правила обмена, протокол, описание обмена мк с телефоном, хочу сделать приложение под Windows. Готов всячески сотрудничать в этом направлении.

vvip-68, 2020-10-10:
frusa, так все ж на странице проекта в разделе Wiki описано...
Интересуюсь потом результатами, выложу в хранилище.
Также интересуюся программой для Android. Ну и для iPhone тоже

frusa, 2020-10-10:
> 
> 
> frusa, так все ж на странице проекта в разделе Wiki описано...
> Интересуюсь потом результатами, выложу в хранилище.
> Также интересуюся программой для Android. Ну и для iPhone тоже

frusa, 2020-10-10:
Да, все увидел. прошу прощения, ток осваиваю гит ) 
Спасибо

closed 2020-10-10 by frusa