vvip-68/GyverPanelWiFi #14: Управление играми

fastdron, 2020-10-24:
Очень понравилась Ваша реализация проекта , спасибо за труды!
Хотел уточнить у Вас - нет ли возможности управлять играми ? 
И ещё - как растянуть тетрис на всю матрицу 
Ш-25
В-11
На окне использую)

vvip-68, 2020-10-25:
Управлять играми возможности нет. Это "Гирлянда на окно", а не игровая приставка.
Выкинуто по причинам: а) мне это не нужно б) код управления занимает огромное количество  места в программе для смартфона, которая ограничена по размерам, было выкинуто, чтобы иметь возможность сделать управление другими настройками.
Растянуть тетрис - всё находится в g_tetris.ino 
Разбираетесь как идет отрисовка и проверки положения плашек, переписываете под себя так как нужно вам.

closed 2020-10-25 by vvip-68

fastdron, 2020-10-25:
Понял, спасибо большое за пояснения. Мне ,по сути, эти игры Толе не особо нужны) просто хотел уточнить ,думал может я чего не досмотрел ) .