vvip-68/GyverPanelWiFi #15: Матрица 25*12 - не помещается один пиксель часов

fastdron, 2020-10-25:
Доброго времени суток , прошу помощи Вашей в правке больших часов ,не влазит один столбик , можно например мигающие точки отредактировать чтоб они были по ширине в один пиксель? И если это возможно - ткните носом где это в коде ибо я от программирования совсем далёкий) . Заранее спасибо! Ваша реализация этого проекта -круче всех!!!

vvip-68, 2020-10-25:
файл clock.ino процедура DrawClock() - строки 284-291 - убрать отрисовку колонки x+13
Далее в строках 295 и 291 сместить позицию на одну влево

closed 2020-10-25 by vvip-68

fastdron, 2020-10-25:
Спасибо ,побежал пробовать👍

vvip-68, 2020-10-25:
Там, похоже, еще проверки на ширину матрицы нужно править 26 на 25 - например в clock.ino строка 197, custom.ino 179