vvip-68/GyverPanelWiFi #16: Управление кнопкой

fastdron, 2020-10-25:
Подскажите почему не работает индикатор тм1637
Собрана матрица 25*12 в скетче выбран          
  Device id ==4 и настроены все пункты, раскоментил #define Dio D5 и # define CLK D7  подключил модуль к соответствующим пинам ,перепрошил D1 mini (правда без очистки памяти) и все равно часы не выводятся и на кнопку принажатии 4 или 5 раз не выводится ip адрес. Что я не так делаю? Или при использовании панели в качестве гирлянды нет возможности подключать индикатор часов и выводить ip?

vvip-68, 2020-10-25:
есть, но честно говоря, особо не проверялось.
и TM1637 точно не может работать с SD картой - они используют одни и те же пины SPI

fastdron, 2020-10-25:
Прошу прощения, мой косяк - не поставил 1 в def-hard для активации модуля. С этим понятно, а вот как вывести ip адрес по нажатии кнопки на панель?

vvip-68, 2020-10-25:
"На панель" - на матрицу или на TM1637?
Впрочем - у меня это работает по четырехкратному / пятикратном нажатию кнопки в зависимости от выбранного при компиляции DEVIC_TYPE. По крайней мере так задумывалось. Что, отвалилось? Сейчас посмотрю...

fastdron, 2020-10-25:
при выбраном типе девайса 4(гирлянда на балкон) не выводится ip на саму матрицу(гирлянду) ,так же пробовал подключить ТМ1637 -время не завилось(может я не донастроил в скетче где-то-пока только разбираюсь))) мигают просто точки а вместо цифр просто полоски (на тм1637) , и ip тоже не выводится на 1637 при нажати 4или 5 раз на кнопку

vvip-68, 2020-10-25:
Так, с выводом IP - поправил, выложил.
С точками и тире - а время то с сервера получилось?
Такое поведение - признак того, что прошивка не смогла получить текущее время с сервера NTP

fastdron, 2020-10-25:
Спасибо огромное, сейчас обновлю и проверю. Вы уж простите что я тут засоряю коменты))) ,сам я только начинаю разбираться с ардуино и электроникой, и на Git научился писать буквально вчера)))))))) Хотел уточнить у Вас - как правильно и проще всего обновить прошивку и добавить Ваши исправления? нужно заново прошивать и настраивать весь скетч , или копировать отдельно измененные файлы и вставлять в скетч? зарание прошу прощения за глупые вопросы) ,просто меня этому никто еще не научил))

vvip-68, 2020-10-25:
Я таки настоятельно рекомендую поставить и настроить SmatrGit. Он показывает что в каких файлах изменилось и позволяет забирать изменения отдельных файлов, не требуя перенастройки ваших исправлений.
Ну или то же самое вручную - на гите смотреть что в каких файлах изменилось при этом коммите и брать только измененные файлы или только измененные строки. Но SmartGit удобнее

fastdron, 2020-10-25:
"Ну или то же самое вручную - на гите смотреть что в каких файлах изменилось при этом коммите и брать только измененные файлы или только измененные строки. "
только что просто строки измененные перезалил в скетч вручную- муторно немного))))
НО ТЕПЕРЬ ВЫВОД IP на матрицу работает по 4кратному нажатию кнопки! Суппер! сейчас заново подкину TM 1637 и посмотрю что с ними) спасибо огромное. также SmartGit начну изучать по Вашему совпету.

fastdron, 2020-10-25:
а нельзя как нибудь сделать чтобы отображение ip на матрице пррокручивалось всего один раз и возвращался ранее выбраный эффект или демо режим?

vvip-68, 2020-10-25:
Можно. Разобраться и переписать код.

fastdron, 2020-10-25:
Понял,ну я это точно не осилю,пока что))))))) Спасибо . отпишусь еще по поводу tm1637 ,как раз сейчас подключаю

vvip-68, 2020-10-25:
Часы то на матрице отображаютс?

fastdron, 2020-10-25:
"Раз уж у вас девайс (кнопка) под рукой и вы (один раз в год может быть) нажмете 4 раза, чтобы посмотреть IP - что вам мешает после этого нажать кнопку 3 раза, чтобы вернуться в демо-режим?" Полностью согласен! это был просто уточняющий вопрос

fastdron, 2020-10-25:
"Часы то на матрице отображаются?"
на матрице отображаются большие часы(пришлось указать размер матрицы на 1 столбик шире (26 вместо 25 иначе были только малые часы) , еще не пробовал правки по часам по Вашему совету, сейчас разберусь с тм1637 и после полезу править часы на матрице)))

fastdron, 2020-10-25:
по поводу подключения TM1637 - как только прошиваю d1mini с активированым в dev_hard модулем ТМ1637- устройство перестает конектится к моей локалке и в монитор порта выкидывает ip адрес подключения к моему роутеру 2,2,2,2 ...

fastdron, 2020-10-25:
LED-Panel-WiFi v.1.05.2020.1012

Загрузка строк выполнена.
Ипользованы адреса EEPROM 800 - 962
Свободно ячеек 3133

Подключение к ASUS
.......
WiFi подключен. IP адрес: 2.2.2.2

UDP-сервер на порту 2390
NTP-сервер ua.pool.ntp.org -> 255.255.255.255
Отправка NTP пакета на сервер ua.pool.ntp.org
Таймаут NTP запроса!
NTP-сервер ua.pool.ntp.org -> 255.255.255.255
Отправка NTP пакета на сервер ua.pool.ntp.org
Таймаут NTP запроса!
NTP-сервер ua.pool.ntp.org -> 255.255.255.255
Отправка NTP пакета на сервер ua.pool.ntp.org
Таймаут NTP запроса!
NTP-сервер ua.pool.ntp.org -> 255.255.255.255
Отправка NTP пакета на сервер ua.pool.ntp.org
Таймаут NTP запроса!

fastdron, 2020-10-25:
и на самих тм1637 соответственно полоски-время не подтянулось - матрица глючит ужасно,на кнопку почти не реагирует, эфекты тормозят.... что это может быть?

vvip-68, 2020-10-25:
Не знаю. Только что собрал проект с включенным USE_TM1637 - никаких проблем нет.
Сам индикатор, правда, не подцеплен - неохото было с проводами возиться паять...

fastdron, 2020-10-25:
Отключение в def_hard не решает прорблему ,только сброс eeprom помог

fastdron, 2020-10-25:
сейчас проверю без индикатора

fastdron, 2020-10-25:
ОООО, интересненько))) без физически подключенного индикатора но с #define USE_TM1637 1 все ок,не виснет ничего

fastdron, 2020-10-25:
прбую другой индикатор

vvip-68, 2020-10-25:
Мне кажется физическое подключение к пинам никак не должно влиять на прошивку...
Что, ж, видимо мне придется доставать паяльник...

fastdron, 2020-10-25:
а не может быть проблемы из за несогласования уровней? я индикатор питаю от 5 в от бп, скорее всего в этом проблема)))) вот я тупень)))

vvip-68, 2020-10-25:
Думаю, что нет

fastdron, 2020-10-25:
именно в этом был трабл))) перекинул питание на 3.3 в - все работает ,не виснет,подключается!!! но часы на индикаторе какие-то битые, сейчас парочку других подкину ,посмотрю

fastdron, 2020-10-25:
а как вывести ip на тм1637 напеомните -это 5кратное нажатие кнопки? если так- то не работает

fastdron, 2020-10-25:
с другим индикатором часы работают ! Класс_ВЫ ЛУЧШИЙ!!!)))

vvip-68, 2020-10-25:
> а как вывести ip на тм1637 напеомните -это 5кратное нажатие кнопки? если так- то не работает
Так же как на матрицу - вывод на индикатор и на матрицу должен идти параллельно.
На индикатор - попеременно цифры тетрад

fastdron, 2020-10-25:
на индикатор не идет вывод ip паралельно с матрицей. на матрице все ок. В скетче есть гдето флаги откл этого? я мог уже рание чтото откл и забыть)))

fastdron, 2020-10-25:
Обновил еще раз прошивку, проверил все параметры, залил - и О чудо!!))))) есть вывод! спасибо огромное

vvip-68, 2020-10-25:
Спаял, проверил - все работает.
https://drive.google.com/file/d/1ImupDAcpdq9ada9xP8HB-hewNHmkvUNG/view?usp=sharing

fastdron, 2020-10-25:
доступ закрыт на диске. Но в любом случае уже работает и у меня! это очень удобно будет смотреть ip - матрица то у меня на улицу будет повернута) а часы выведу на корпусе устройства и там смтогу смотреть ip в случае необходимости! Спасибо огромное вам за помощь!!!!!!!!! тут все... есть вопросы по выводу текста по наступлению события и по спец режимах, если у вас будет минутка -поможете разобраться?

vvip-68, 2020-10-25:
Да открыл уже доступ. Кнопку вовремя не нажал...
Помогу

fastdron, 2020-10-25:
в отдельную тему вопросы по событиям писать или тут?

fastdron, 2020-10-25:
это у вас матрица 16 на 16 на видео?

vvip-68, 2020-10-25:
из трех 16х16 Это мой тестовый стенд
Если длиннопост (ак этот) - лучше вообще  в личку сюда https://vk.com/iivanovii

fastdron, 2020-10-25:
а телеграмм нету у вас? @Fastdron

vvip-68, 2020-10-25:
Нет

closed 2020-10-25 by vvip-68