vvip-68/GyverPanelWiFi #17: Предложение по доработке обратного отсчёта до события

fastdron, 2020-10-26:
Добрый день. Было бы не плохо вести обратный отсчет до события ,когда остаётся меньше минуты - показывать обратный отсчёт секунд на матрице (если указан макрос который нужно выводить по наступлению события).
Например: 
Задано событие 1 января 2021г-
Указано в макросах вывести текст "С новым 2021 годом" (закреплено в событиях)-
Когда до наступления события остаётся меньше минуты - предлагаю просто выводить обратный отсчёт секунд ,не переключаясь на другие бегущие строки или эффекты , и когда отсчёт дойдет до нуля - вывести макрос события "С Новым 2021 годом"
Включать этот отсчёт секунд предлагаю только в случае если в событии указан макрос с текстом вывода, если там ,ничего не указано - то просто не включать отсчёт (логика остаётся прежней )

fastdron, 2020-10-26:
Ещё ,если это вообще возможно - добавить возможность задавать события не только по дням , но и указать конкретное время события до минут (или ,хотя бы часов)

fastdron, 2020-10-26:
Например чтоб можно было указать в событии " До начала матча осталось " и указать конкретную дату и время)

vvip-68, 2020-10-26:
Это требует довольно серьезной переработки алгоритма работы со строками. У меня сейчас нет ни времени ни желания погружаться во все это. Да я честно говоря и не представляю как это можно сделать. Сейчас просто через указанные промежутки времени извлекается очередная строка, смотрится является ли она строкой события и отображается тот или иной текст. Ваше же предложение требует где-то хранить список даты и времени событий (а их может быть несколько), на каждой итерации цикла loop проверять не осталось ли указанного времени до события, какая строка с этим связана и выводить ее. 
То есть: а) - полностью переработать алгоритм проверки и вывода строк текста. б) вводить новые сущности - список дат и связь со строками и хранение в памяти всего этого. в) даже если бы это было легко сделать в прошивке - каким образом заносить эти даты (сообщать об этом прошивке)? Программа на смартфоне достигла своего предела и не позволяет добавлять туда что-то новое.
Все исходники открыты - вы просто можете озадачиться этим и реализовать то, что нужно именно вам. Если нет навыков программирования - хороший повод научиться этому. Так же из за ограничений программы (см. выше) - будет повад разобраться в программировании под андроид и написать программу для телефона, которая будет взаимодействовать с прошивкой.

closed 2020-10-26 by vvip-68

fastdron, 2020-10-26:
На счёт научится программированию - тут я полностью согласен) тема очень интересная, именно этим сейчас и занимаюсь. 
Я понял что это очень сложная доработка - а если просто выводить последние 60 секунд в виде обратного отсчёта (не прерывая отсчёт эффектами или другими строками) это возможно реализовать?

vvip-68, 2020-10-26:
Думаю да, при желании всё возможно. Хорошая учебная задача. Пробуйте.

fastdron, 2020-10-26:
Буду разбираться)

vvip-68, 2020-10-26:
Кстати, если хорошо подумать - то программу на телефоне можно и не править, наверное. Просто каждый раз при загрузке массива строк из EEPROM в массив этих строк textLines[36] (или при сохранении изменений строк) - нужно пробежаться по всем строкам массива, найти те, которые содержат дату (и время!) события ({R26.10.2020 2330}) распарсить дату и время и записать в какой-нибудь другой массив - например как массив unsigned long - тот же тип, что возвращает millis() - а именно время в unixtime. Плюс еще массив индексов этих строк, чтобы знать какую строку извлекать и отображать при подходящем событии. Далее на каждом проходе loop() делать пробежку по распарсенному массиву этих unixtime и если unixtime[i] - millis() < 60000 (минута в милисекундах или сколько вам там надо) - отображать текст бегущей строки, индекс которой сохранен во втором массиве соответствия времени события и отображаемой строки. Как-то так.

vvip-68, 2020-10-26:
Про millis() выше я ерунду написал. Это время работы контроллера после рестарта. Unixtime же текущее надо брать с библиотеки time. На каждом проходе брать текущее время 
```time_t t = now();```
и вычислять сколько осталось до события

fastdron, 2020-10-28:
Подскажите по поводу режимов по времени - есть возможность включать матрицу в заданное время не один из эфектов ,а именно демо режим?

vvip-68, 2020-10-29:
Да, в списке действий режимов по времени называется "Случайный режим".

fastdron, 2020-10-29:
Понял, спасибо ! Думал что "случайный режим" включает один из режимов рандомно и крутит его.

vvip-68, 2020-10-29:
Что стоило выбрать и проверить-посмотреть что получится?...

vvip-68, 2020-10-29:
> Ещё ,если это вообще возможно - добавить возможность задавать события не только по дням , но и указать конкретное время события до минут (или ,хотя бы часов)

Так, добавил время в макрос события - теперь можно писать {R01.11.2020 7:25#E}

>Было бы не плохо вести обратный отсчет до события ,когда остаётся меньше минуты - показывать обратный отсчёт секунд на матрице (если указан макрос который нужно выводить по наступлению события).

Так, сделал событие вида {P01.11.2020 7:00#E#45#25#345}
Здесь - 01.11.2020 - дата события. Может быть опущена, указано только время, что будет означать "каждый день"
Любая часть даты может быть заменена звездочками, например **.11.2020 - значит весь ноябрь
**.**.2020 - весь 2020 год, 01.**.2020 - в 2020-ом году каждое 1-е число 09.11.**** - каждое 11-е сентября и так далее

07:00 - время наступления события. Если опущена - значит 0:00; 
Дата и время разделяются одним пробелом

#E - строка-заменитель, которая отображается ПОСЛЕ наступления события.
#45 - за сколько секунд ДО наступления события отображать эту строку с обратным отсчетом
#25 - сколько секунд ПОСЛЕ наступления события отображается строка-заместитель. Если она не указана - ничего не отображается
#345 - в какие дни недели отображать 1-пн..7-вс

от #Е и далее - может быть опущено. Тогда значения по умолчанию - нет строки ПОСЛЕ, 60 сек ДО 60 сек после и все дни недели - т.е 1234567

Например можно указать для строки 'D':  ```Побудка через {P07:00#E}```
Что эквивалентно ```Побудка через {P07:00#E#60#60#1234567}```
И для строки E ```C добрым утром!```

Итак, строка ```Побудка через {P07:00#E#60#60#12345}``` будит каждый будний день в 6:59 начинать отображать обратный отсчет бегущей строкой, а с 7:00 до 7:01 - 'C добрым утром!э

То же самое можно и с новым годом и "До начала матча осталось..." и так далее.
Пробуй. Если найдешь ошибки - пиши.

fastdron, 2020-10-29:
Спасибо за доработку бегущий строки!!!! ВЫ мега-крут)))

fastdron, 2020-10-29:
это не просто лучшая реализация  проекта Алекса - это уже ВАШ!!!! крутой и неповторимый проект! Респект за труды,старания и отзывчивость

fastdron, 2020-10-29:
Сейчас обновлюсь и протестирую все новые фишки ,класс

fastdron, 2020-10-29:
укажите еще в wiki о проблеме с tm1637- как у меня были зависания и глюки при подключении к 5в линии питания, после переключения на 3.3 в -проблема ушла. Возможно это единичный случай-но,думаю, людям не мешало бы знать. спасибо

vvip-68, 2020-10-29:
Насколько я помню - есть два варината этих модулей - одни рассчитаны на 5 вольт, другие на 3.3
Нужно смотреть какой покупаешь и с какой ноги ему питание даешь...

fastdron, 2020-10-29:
еще заметил небольшой баг(фитчу)))) в обратном отсчете секунд до наступления события:
секунды то отсчитываются быстро и текст осталось ** "секунд" или "секунды" не всегда попадает в "тему"))))) ,где можно его заменить на сокращение "сек"?
Просто иногда бывает что попадает на 26 сек. и пишет остьалось 26 секунды. Это вообще не критично ,спасибо вам за реализацию.

vvip-68, 2020-10-29:
utility.ino - строки от 488 до 521

fastdron, 2020-10-29:
#345 - в какие дни недели отображать 1-пн..7-вс

еще хотел уточнить у Вас по поводу дней недели- если указана конкретная дата ,то дни недели не учитываются? {P01.11.2020 7:00#E#45#25#345} вот например тут зачем 3й 4й и 5й день указан если задана четкая дата? 
Я так понимаю что #345 пригодятся в том случае если дата упущена?

vvip-68, 2020-10-29:
Если конкретная дата - все равно учитываются. Безобразно, зато единообразно.
Если укажешь конкретную дату 30.10.2020? но в днях недели не будет пятницы (5) - строка не отобразится.
И да - все эти дни недели именно для нечетких дат придуманы.

fastdron, 2020-10-29:
"Если укажешь конкретную дату 30.10.2020? но в днях недели не будет пятницы (5) - строка не отобразится"
То-есть при :
{P30.10.2020 7:00#E#45#25#1234567} - строка отобразится с заданными параметрами
{P30.10.2020 7:00#E#45#25#5} - строка отобразится с заданными параметрами
{P30.10.2020 7:00#E#45#25#} - строка отобразится с заданными параметрами

{P30.10.2020 7:00#E#45#25#123467} - строка не отобразится вообще
верно?

vvip-68, 2020-10-29:
Верно

fastdron, 2020-10-29:
> #E - строка-заменитель, которая отображается ПОСЛЕ наступления события.
> #45 - за сколько секунд ДО наступления события отображать эту строку с обратным отсчетом
> #25 - сколько секунд ПОСЛЕ [...]

{P01.11.2020 7:00#E#45#25#345}
01.11.2020 ЗА 45 СЕКУНД ДО 7:00 ОТОБРАЗИТСЯ СТРОКА ,ПРОДОЛЖИТЕЛЬНОСТЬ 25 СЕК, (С УЧЕТОМ ЧТО 1Е ЧИСЛО ПОПАДАЕТ В 3,4 ИЛИ 5 ДЕНЬ НЕДЕЛИ)
а что если не указать к-во секунд продолжительности (после наступления события) ?
{P01.11.2020 7:00#E#45#345}- в таком случае система будет "думать" что ей задали выводить строку за 45 сек до и 345 сек после времени события? и  считать что заданы все 7 дней недели (#1234567)?

vvip-68, 2020-10-29:
Именно так

fastdron, 2020-10-29:
понял ,спасибо. и еще контрольный для закрепления)))) Вы уж не сердитесь что достаю вопросами своими) -Я только учусь)
и так : {P01.11.2020 7:00#E#45} - в таком примере строка будет воспроизведена за 45 секунд до 7-00 ,продолжительность воспроизведения строки составит 60 сек(по умолчанию), (дни недели будут учитываться все(1234567по умолчанию),но строка прокрутится только один раз с задаными параметрами 01,11,2020го) .
Я все правильно понял?

fastdron, 2020-10-29:
Протестировал малость событие вида {P01.11.2020 7:00#E#45#25#345} ,обратный отсчёт срабатывает как нужно в заданное время, с этим все хорошо -но выявил проблему -после завершения события и после прокручивания строки заместителя все равно продолжат прокручиваться строка события(но уже без обратного отсчёта) интервал вывода строк указан 60 сек в приложении , ещё если поставить интервал вывода бегущих строк в приложении 10секунд - то события вида событие вида {P01.11.2020 7:00#E#45#25#345} вообще не отрабатывают (это не страшно, 10сек никто ставить не будет) . Как бы заставить отключатся строку события после завершения, чтоб она больше не крутилась как обычная строка ?

fastdron, 2020-10-29:
И ещё вопрос -не будут- ли конфликтовать события {R} и {P} ?
Например строка" До нового года" {R01.01.21#N}  - не будет конфликтовать со строкой {P01.01.2021 #2#60#60#1234567} ? У кого из них приоритет и будет ли обратный отсчёт?

vvip-68, 2020-10-30:
1. Да, строка должна показываться только в интервал за 45 сек до и 60 сек после, 1 ноября 2020 года в районе 7:00
2. Озадачил. В коде runningText строки 243-248 стоит 
```        bool disabled = (text.length() == 0) || (i == 0 && text[0] == '#') || (text[0] == '-') || (text.indexOf("{-}") >= 0) || (text.indexOf("{P") >= 0);
        if (disabled) continue;
```
То есть строка с наличием макроса "{P" обрабатывается так же как "отключенные" строки - а значит не должны показываться как обычные строки. Если показывается как обычные - я не понимаю как это происходит.
И со вторым тоже непонятно. Проверка наступления события {P} в коде стоит ПЕРЕД проверкой необходимости других строк м по идее - если есть активное {P} - другие строки вообще не должны показываться, а ты говоришь что всё наоборот. Чудеса.
3. При наличии одновременно {R} и {P} судя по коду и по задумке - {R} вообще должно игнорироваться, т.к. {P} имеет более высокий приоритет. И {R} вообще не должна отрабатываться. Но я уже ни в чем не уверен, раз у тебя все работает не так как задумывалось.

В общем все должно работать как ты пишешь как должно, но если у тебя все работает с точностью до наоборот - ХЗ. Будем разбираться

vvip-68, 2020-10-30:
> При наличии одновременно {R} и {P} судя по коду и по задумке - {R} вообще должно игнорироваться, т.к. {P} имеет более высокий приоритет. И {R} вообще не должна отрабатываться. 

И вообще , строка с наличием {P} не должна выводиться по интервалам как регулярная строка. При этом и макрос {R} и макрос {P} каждый вставляет свой хвост остатка от своего времени. Поэтому одновременное наличие этих двух макросов не имеет смысла. Как регулярная строка она отображаться не будет, будет только в наступление события {P}, но заменены на "остаток" будут оба макроса.

vvip-68, 2020-10-30:
> Такой сценарий использования пригодился бы как раз для события типо Нового Года , {R} бы задавался вывод бегущей строки "До нового года осталось ** дней (часов ..) "например каждых пол часа круглосуточно, а {P} бы активировала обратный отсчёт в бегущей строке и выводила строку заместитель)

Поскольку каждый макрос оставляет свой хвост "остатка" конструкция с одновременным наличием макросов R и P смысла не имеет. Для отображения "...осталось..." с макросом R должна быть своя строка, которые будет показываться через интервалы времени, для события около указанного времени с макросом {P} должна быть отдельно своя строка.
{P} не должна отображаться по интервалам, только около привязанного события

vvip-68, 2020-10-30:
Ну вот, почти час погонял я во всех режимах, которые ты описываешь. И с интервалом 10 секунд, и с прошедшим событием и все такое. Ни одного эффекта, описываемого тобой не наблюдается. Все отрабатывает как было задумано. И {P} как регулярная строка не выводится. Ни до ни после события. И прерывает другие строки, даже если они уже отображаются к моменту наступления события {P}. И если строка имеет оба макроса R и P - она не выводится как регулярная, выводится только при наступлении события и, конечно же, печатает оба хвоста остатка...
Вопрос - что я делаю не так?
Или - что ты делаешь, чтобы получить описанные несоответствия?

fastdron, 2020-10-30:
> Давай полный баг-репорт со всеми настройками и что сделать по шагам, чтобы получить описанные тобой ошибки.

Сейчас на работу иду - через 8 часов более подробно опишу и протестирую, матрица тестовая дома остаётся))) ! Спасибо за помощь!

fastdron, 2020-10-30:
А как лучше будет сделать этот баг-репорт? Просто более подробно описать свои действия и настройки матрицы и прошивки? Или в скетче есть возможность писать логи и именно они бы помогли Вам разобраться? Выкл извините за порой глупые вопросы) правда только пару недель как всем этим увсекся) Esp8266 только с алишки недавно забрал, опыта вообще нет)

vvip-68, 2020-10-30:
Попробуй просто написать что в какой последовательности делать, чтобы у меня тоже получилась эта ошибка. Я делал так как это понимаю я и у меня никаких несуразностей не выявилось. Но у меня может быть глаз "замылен" - я неосознанно делаю так, как оно ДОЛЖНО быть, а сторонний пользователь может действовать по своей логике, автором не предусмотренной.

Не всегда такие ошибки следует исправлять и ставить "защиту от дурака" на любой случай. Всегда можно ОСОЗНАННО сделать что-то так, что программа "сломается". Найти способ сломать можно почти всегда. Однако, если что-то делается вроде как правильно - а результат неожиданный - тут стоит посмотреть - может нужно что-то подправить в коде.
Без фанатизма, чтобы код проверок не был слишком большим и сложным, сложнее чем код самого алгоритма, тем более на безопасность это не влияет.

Можешь попробовать сам выловить в коде что работает не так.
Просто в контрольных точках кода пишешь Serial.println() в котором выводишь в монитор порта контрольную информацию / значения переменных. Потом анализируя что выведено в лог - смотреть что пошло не так.

fastdron, 2020-10-30:
> Можешь попробовать сам выловить в коде что работает не так.
> Просто в контрольных точках кода пишешь Serial.println() в котором выводишь в монитор порта контрольную информацию / значения переменных. Потом анализируя что выведено в лог - смотреть что пошло не так.

Ок, спасибо большое ! После работы займусь. Правда жена уже гонит с балкона(кабинет у меня там свой для работы) ,говорит что пропадаю там часто😁))))

fastdron, 2020-10-30:
Могу по памяти описать действия свои. 
1.Текст ввожу в основном с приложения 
2.Настройки deh_hard
#if (DEVICE_ID == 4)
/*
 * Wemos D1 mini
 * Физическое подключение:
 * Матрица зигзаг, левый нижний угол, направление вправо
 * Пин ленты    - D2
 * Пин кнопки   - D4
 * В менеджере плат выбрано NodeMCU v1.0 (ESP-12E)
 */ 
#if defined(ESP8266)
#define WIDTH 25              // ширина матрицы
#define HEIGHT 11             // высота матрицы
#define DEVICE_TYPE 1         // Использование матрицы: 0 - свернута в трубу для лампы; 1 - плоская матрица в рамке   
#define MATRIX_TYPE 0         // тип матрицы: 0 - зигзаг, 1 - параллельная
#define CONNECTION_ANGLE 0    // угол подключения: 0 - левый нижний, 1 - левый верхний, 2 - правый верхний, 3 - правый нижний
#define STRIP_DIRECTION 1     // направление ленты из угла: 0 - вправо, 1 - вверх, 2 - влево, 3 - вниз
#define USE_SD 0              // поставьте 0, если у вас нет SD-card модуля
#define USE_MP3 0             // поставьте 0, если у вас нет звуковой карты MP3 плеера
#define USE_TM1637 1        // поставьте 1, если используется дополнительный индикатор TM1637, 0 если индикатора нет
#define USE_WEATHER 0         // 1 - использовать получение информации о текущей погоде; 0 - не использовать 
#define USE_POWER 1           // 1 - использовать отключение питания матрицы через MOSFET; 0 - не использовать 
#define USE_MQTT 0            // 1 - использовать управление по MQTT-каналу; 0 - не использовать 

#define LED_PIN 2             // D2 пин ленты
#define PIN_BTN D4            // кнопка подключена сюда (D4 --- КНОПКА --- GND)
#define POWER_PIN D1          // D1 управляющий пин вкл/выкл матрицы через MOSFET; POWER_ON - HIGH, POWER_OFF - LOW
#define SD_CS_PIN D8          // пин выбора SD карты - ChipSelect; Также SD карта использует D5 (CLK), D6 (MISO), D7 (MOSI)
//#define SRX D4                // не используется, но требуется для компиляции скетча (USE_MP3==1)
//#define STX D3                // не используется, но требуется для компиляции скетча (USE_MP3==1)
#define DIO D5                // D5 TM1637 display DIO pin - не используется, но требуется для компиляции скетча (USE_TM1637 == 1)
#define CLK D7                // D7 TM1637 display CLK pin - не используется, но требуется для компиляции скетча (USE_TM1637 == 1)

#endif
#endif

3.Прописан свой домашний wifi в 1й подсети
4.в приложении включен режим "демо" 
5. Интервал смены эффектов 20сек.(в приложении)
6. Интервал смены бегущих строк 20 сек.(в приложении)
7. Создаю Макрос "2" с текстом "С НОВЫМ ГОДОМ" (фон черный)
8. Создаю макрос "1" с текстом "До Нового Года {P30.10 2020 10:00#2} " указываю цвет фона черный.пишу этот текст и сохраняю в 9:57 (за 2мин до предполагаемого начала вывода обр. отсчёта) все ввожу с помощью приложения. 
Результат : 
- через 20 сек выводится текст "До Нового Года" (как в макросе "1") но без обратного отсчёта , спустя ещё 20 сек выводится повторно тот же текст
Даже не знаю что сделал не так)
- вчера (по-моему) делал все тоже но указал более подробно событие "{P30.01.20 10:00#2#60#60#1234567} в таком виде обратный отсчёт начинался вовремя и не прокручивался ранее чем за минуту до наступления события (надо ещё проверить, не могу утверждать на 100%) но была другая проблема - после обратного отсчёта текст макроса "1" продолжал крутится с заданным в приложении интервалом для вывода бегущих строк ,но уже без обратного отсчета -как обычная строка

fastdron, 2020-10-30:
> Попробуй просто написать что в какой последовательности делать, чтобы у меня тоже получилась эта ошибка

Обязательно опишу подробно

vvip-68, 2020-10-30:
Да, похоже, такое происходит, когда забиты всего две строки и ни одна из них в настоящий момент не должна воспроизводиться. Если есть еще строки, которые могут быть отображены - такого не происходит.  Сейчас посмотрю как это исправить

vvip-68, 2020-10-30:
Исправил.

fastdron, 2020-10-30:
> похоже, такое происходит, когда забиты всего две строки

да- забивал всего две строки для тестов.

fastdron, 2020-10-31:
> Исправил.

добрый день. все протестировал,проверил,пощелкал... -ВСЕ СУПЕР!!!!! текст выводится как надо и когда нужно! спасибо. теперь пробую как то побороть ПОГОДУ)))) в Украине у меня что-то совсем не хочет работать.

vvip-68, 2020-10-31:
А в чем проблема? У вас Яндекс не банят? Какой город?
https://yandex.ru/time/sync.json?geo=115326
Что возвращает?
Что возвращает в адресной строке браузера, если набрать yandex.ru
А потом просто нажать на кнопку "Поиск"? Какой код стоит после ?lr=xxx - какой xxx?

fastdron, 2020-10-31:
> Что возвращает в адресной строке браузера, если набрать yandex.ru

[image]

fastdron, 2020-10-31:
Мой провайдер "Киевстар" банит ,скотина(((

> Какой город?

Днепр

Мне копать в сторону openwrt?

fastdron, 2020-10-31:
Ещё хотел попросить вас заснять ваши готовые устройства на видео (если не затруднит). Ну прям очень понравился проект и вообще вся темя про адресные светодиоды. Планирую 11.11 затариться на алишке матрицами, лентами, светодиодами , и собрать что-то типо часов настенных , также хочется картину-панель на стену и лампу- если у Вас это все есть в готовом виде - хотелось бы посмотреть как это вживую смотрится👍🍻

vvip-68, 2020-10-31:
Ну, значит, погода у тебя работать не будет, увы...
Я не знаю что такое openwrt и как им пользоваться
Ну, может сниму как-нибудь. У меня на самом деле не так много девайсов то - троек часов (они все примерно одинаковые, достаточно одно снять) и гирлянда на окне висит. Ну и на столе три матрицы 16x16 склеены как тестовый стенд - отлаживал не нем эффекты для гирлянды. Она у меня 50x20. А уж видео вообще может только на гирлянды смысл имеет. Да и то сомнительно - достаточно фотки.

vvip-68, 2020-10-31:
Во, вот тут видео часов откопал:
https://www.instagram.com/p/CE_zXpQCiGG/
https://www.instagram.com/p/CE6YmDUjDl-/

fastdron, 2020-10-31:
Шикарно смотрятся часы. Корпус печатали ?

vvip-68, 2020-10-31:
Не часы - чисто светильник с леруа типа такого (у вас леруа то хоть не забанили?)
https://leroymerlin.ru/product/svetilnik-vstraivaemyy-svetodiodnyy-kvadrat-18-vt-belyy-svet-82026534/
Гирлянда - просто светодиоды, без рассеивателей. Брал просто "струну" светодиодную типа такой:
[image]
Шаг сетки получился 10x10 см
У меня балкон шириной 6 метров. На тюлевую штору нитки пришил, чтобы окно можно было открыть, сдвинув шторку.

vvip-68, 2020-10-31:
[image]

vvip-68, 2020-10-31:
Кстати, светодиодная струна такого типа мне категорически не понравилась и брать не советую.
Проводки, вероятно, имеют плохую изоляцию  (просто голый провод-косичка, вероятно покрытый лаком. Кажется он временами / местами где-то замыкает, поэтому встречались линии просто светящиеся всеми цветами радуги на полной яркости вместо того, что им шлет контроллер. И где-то искажает сигнал. 20 линий не вытянул - при включении 20-1 линии с 16 по 20 начинают сбоить вспышками. Отключил 20-ю - меньше стало, но все равно. Плюс "синхронизация" периодически срывается. Хотел чтобы часы отображались - так они периодически рассыпаются и яркие точки по всему полю матрицы. Пришлось на гирлянде часы отключить - на эффектах срывы не так заметны. Не видно как цифры прыгают.

fastdron, 2020-10-31:
> у вас леруа то хоть не забанили?

Хахаааха))) Нет - с этим норм, полно строительных гипермаркетов)

> У меня балкон шириной 6 метров.

Не хило) и до самого пола застеклён? У меня тоже почти 6м но перегородили на два отдельных,на одном теперь мой "кабинет"))) , в теории можно через перегородку прикинуть перемычку и тоже раскинуть гирлянду на все 5,8м, (правда высота окна всего 120см) . Соседи будут в восторге😂

vvip-68, 2020-10-31:
У нас новый дом - фасадное остекление от пола до потолка. 3 метра в высоту, 6 метров в ширину.

fastdron, 2020-10-31:
Я вообще планирую сам паять, купил ленту, порежу на светодиоды, затем полоски из прозрачного орг-стекла шириной 2см ,на них наклею светодиоды шагом 19см, спаяю проводами. В одной такой "палке" будет 11 светодиодов, и таких палок пока планировал 25 (на пол балкона) . Крепить их буду на присоски с алишки. Верх и низ этих столбиков тоже будет зафиксирован на полоске огрг стекла

vvip-68, 2020-10-31:
[image]

fastdron, 2020-10-31:
> У нас новый дом - фасадное остекление от пола до потолка. 3 метра в высоту, 6 метров в ширину

Представляю какруто смотрится👍👍👍. Цепляет гирлянду к НГ или постоянно висит?

vvip-68, 2020-10-31:
Она первый год. Сам же ж понимаешь - только доделал. Там посмотрим. Впрочем, не вижу необходимости снимать. Надоест - просто выключу

fastdron, 2020-10-31:
Вообще красота! Реально украшение всего дома) к Вам ,наверное уже рекламщики обращаются вывести рекламу на балконе😂😂😂)))?

[58 more replies]