vvip-68/GyverPanelWiFi #18: 120x7 matrix

yusufsulaiman, 2020-10-27:
hi, thanks for sharing
i made matrix 120x7, but runningtext is not showup
what should i do?
thanks

vvip-68, 2020-10-27:
I do not know. Maybe you doing something wrong. I tried run text line on my test matrix with defined size 16x7 - text running well.
Of course tall cyrillec letters like 'Ц' and 'Щ' are truncated, however text is shown.
Does anything suspect in port monitor logs?

yusufsulaiman, 2020-10-27:
It works for 60x7, but it did not work for 120x7. Is it due to eeprom runningtext.thanks
[image]
[image]
[image]

vvip-68, 2020-10-27:
What wrong with eeprom running text? As far as I understand eeprom capacity for store text is about 1.5K chars.
And in my understanding do not related with ability to display text on matrix.

yusufsulaiman, 2020-10-27:
this is when i send runningtext
UDP пакeт размером 31 от 192.168.4.2, порт 2390
Содержимое: $6 0|1|ghwjshshsbbd            
Сохранение строк выполнено.
Ипользованы адреса EEPROM 800 - 849
Свободно ячеек 3246
Answer to 192.168.4.2:2390 >> $18 TE:1|TI:20|CT:0|ST:69|C2:255,255,255|OM:1623|TS:520000000000000000000000000000000000|TA:1|TX:[ghwjshshsbbd]|TY:[1:1 > 'ghwjshshsbbd'];

Exception (3):
epc1=0x40208501 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40024b48 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd30 end: 3fffffc0 offset: 01a0
3ffffed0:  00000000 00000007 00000078 4020c974  
3ffffee0:  00000000 3fff5b2e 0000000c 00000000  
3ffffef0:  3fff5eb2 00000001 0000000c 40211339  
3fffff00:  014ba1a5 3fffdad0 3fff7400 000c000f  
3fffff10:  00000001 00000001 00000000 00000001  
3fffff20:  ffffff8f 3fff6870 3fff8d38 401063e3  
3fffff30:  3fff5be8 3fff5b2e 3fff8d04 3fff61a4  
3fffff40:  00000000 3fff5b2e 3fff3c5c 402148ba  
3fffff50:  6c432000 7364756f 80000027 3fff61a4  
3fffff60:  3fffdad0 3fff6870 3fff5f5c 4021fa55  
3fffff70:  3fffdad0 00000000 3fff5f5c 4021df36  
3fffff80:  00000000 00000000 00000001 3fff61a4  
3fffff90:  3fffdad0 00000000 3fff6164 40214bcb  
3fffffa0:  3fffdad0 00000000 3fff6164 4021c668  
3fffffb0:  feefeffe feefeffe 3ffe8590 40101115  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld

vvip-68, 2020-10-27:
Do you have latest version of firmware? A have more then 10 working devices with this firmware with different sizes of matrix. All of them display text (receive text from application, save it into eeprom) without any fault. This stack information is not informational for me. It may be decoded on your size using Exception decoder (https://github.com/me-no-dev/EspExceptionDecoder) having your build of firmware.
I am not able to debug this on my side, because I am not able to reproduce error.

vvip-68, 2020-10-27:
A see context of logs you provided - it contains words and structure of log phrases that do not present in latest version of firmware. So, I think you use enough outdated version of firmware that may contain errors fixed in later versions. Try to update your firmware to latest version. Send me log text from moment of start microcontrollet and until exceprion output.

yusufsulaiman, 2020-10-27:
ok. thanks

vvip-68, 2020-10-27:
Did you solve isuue finally?

closed 2020-10-27 by vvip-68