vvip-68/GyverPanelWiFi #19: Обратная связь (вывод состояния матрицы в приложение)

fastdron, 2020-10-28:
Доброго времени суток- не могли бы Вы,если это не слишком сложно, попробовать впихнуть в приложение небольшуб строчку или индикатор некий ,на котором бы отображалось текущее состояние матрицы( вкл.выкл , и какой именно режим активен,вернее эфект) , Было бы очень удоюно для балконных матриц которые на улицу повернуты, можно было бы не выходя на улицу или балкон посмотреть в каком режиме устройство- демо или ручное  и тд...

fastdron, 2020-10-28:
[image]
 что-то типо того))

vvip-68, 2020-10-29:
Приложение на телефоне уже достигло своего предела и практически любое изменение / добавление нового приводит к тому, что оно перестает собираться совсем. В качестве рекомендации могу предложить включить поддержку MQTT в приложении? настроить его - https://github.com/vvip-68/GyverPanelWiFi/wiki/%D0%9D%D0%B0%D1%81%D1%82%D1%80%D0%BE%D0%B9%D0%BA%D0%B0-%D0%BF%D0%BE%D0%B4%D0%BA%D0%BB%D1%8E%D1%87%D0%B5%D0%BD%D0%B8%D1%8F-%D0%BA-MQTT-%D1%81%D0%B5%D1%80%D0%B2%D0%B5%D1%80%D1%83

Далее на компе зайти на панель http://client.mqtt.4api.ru/ подключиться к MQTT-серверу (к которому подключено и приложение). В панели будут видны все сообщения, и состояние в котором сейчас находится матрица.

[image]

При некоторой сноровке можно наловчиться отправлять команды для управления матрицей
Система команд описана тут https://github.com/vvip-68/GyverPanelWiFi/wiki/API-%D1%83%D0%BF%D1%80%D0%B0%D0%B2%D0%BB%D0%B5%D0%BD%D0%B8%D1%8F-%D1%83%D1%81%D1%82%D1%80%D0%BE%D0%B9%D1%81%D1%82%D0%B2%D0%BE%D0%BC

Приложение для телефона на native Android SDK в планах, но пока не хватает знания и времени. Можете разобраться и написать. Будет полный контроль над матрицей, что захотел - добавил-сделал. И осознания - "Я молодец!, Я сделаль!"
И респект от остальных пользователейю

closed 2020-10-29 by vvip-68

fastdron, 2020-10-30:
> Приложение для телефона на native Android SDK в планах

Где-то тут встречал вашу переписку с одним из пользователей-вроде он обещал помочь с приложением. Или дело так и закончилось только разговором?

vvip-68, 2020-10-30:
Слился. После этого даже а ВКонтакте не появлялся.

fastdron, 2020-10-30:
Поспрашиваю у своих на работе, есть у нас пару программистов нормальных, может кто и поможет

vvip-68, 2020-10-30:
Да там, мне думается, не особо сложно. Больше - рутина чем творчество.
И синтаксис у Котлина какой-то слегка замысловатый. 
Плюс часть операторов / переменных / конструкций, которые во всех языках есть и принято называть примерно одинаково - авторы Котлина зачем-то назвали по иному.  
Типа во всех языках функция не возвращающая значений - void function() в котлине - unit function() Смысл тот же - зачем-то назвали иначе. Объявление переменных - везде объявление переменной - var - в Котлине val
Ну и еще до кучи по мелочи. Когда с языками много работал - осмыслить бессмысленное достаточно нелегко.
Ну а сами конструкции примерно одинаковые...