vvip-68/GyverPanelWiFi #21: GyverPanelWiFi_v1.06

AxxiMator, 2020-11-06:
Здравствуйте. Очень понравился проект.
Очень хотелось бы увидеть в проэкте версию часов со ШРИФТОМ 6*8 - могу даже оказать помощь с кодом для данного шрифта.
Данный шрифт очень актуален для матриц Ws2812 размером 8*32. И цифры красивее смотрятся.
Только как его  интегрировать в код - не знаю.
Заранее благодарен. =Александр+

vvip-68, 2020-11-06:
Все доступные символы шрифта определены в файле fonts.h (разложите Hex в binary - увидите образы символов).
Делаете по аналогии файл со своим шрифтом. Далее в a_def_soft.h в строках184,185 меняете ширину/высоту символа в соответствии с вашими размерами - должно заработать. Если нет - в файле funningText.ino возможно придется отладить getFont() и drawLetter()

vvip-68, 2020-11-06:
Данная версия не предполагается для эксплуатации в отсутствии интернета. Он сейчас есть практически везде, где есть электричество. Изначальная идея - как можно меньше компонентом, чтобы всё работала с одной маленькой платкой без лишних наворотов и дополнительного монтажа - припаял 3 провод и готово. . Исходные коды и схемы открыты - вы можете легко интегрировать энергонезависимые часы какие нужны вам для ваших потребностей.

closed 2020-11-06 by vvip-68