vvip-68/GyverPanelWiFi #22: Changing custom list of effects

Engineerator, 2020-11-11:
Hello, thank you for this project!

I would like to change the list of effects that switch by themselves. The comments in custom.ino says to put the code in this structure "case number: effect; break;"

Where exactly do I put this code if I wanted to add "case MC_SNOW: snowRoutine(); break;"?

vvip-68, 2020-11-12:
I do not understand what you to achive with these changes.
1. Call of ```doEffectWithOverlay(aMode);``` do a lot of other preparations before show effects.
2. Your construction ```void customRoutine(byte aMode) { case MC_SNOW: snowRoutine(); break; }``` as you provide it is wrong by syntax and is not compileable.  ```case MC_SNOW: snowRoutine(); break;``` is part of ```switch``` operator, which is analog of ```if()``` cascade and do selection which effect must be processed depend on control variable aMode - all of them placed inside ```processEffect()``` procedure.
3. If you receive compiled scetch without error - I surprised about it. The blank matrix is became your current mode (thisMode) either is not in range ow known modes constants or there is no code which call one of effect processing procedure.

Engineerator, 2020-11-12:
Sorry what I meant was `void customRoutine(byte aMode) { switch (aMode) { case MC_SNOW: snowRoutine(); break; } }`.
I also meant that I ran the default sketch only changing the parameters to fit my matrix and the program worked normally.

I ran the default sketch and only changed the processEffect() procedure to `void processEffect(byte aMode) {
  switch (aMode) {
        case MC_SNOW:                snowRoutine(); break;
  }
}` but my matrix became blank.

vvip-68, 2020-11-12:
Obviously it because aMode was not MC_SNOW at that moment and showRoutine() was not called.
If you need only show MC_SNOW effect and by some reason do not want use sketch normally, set processEffect() procedure as
```
void processEffect(byte aMode) { snowRoutine();  }
```

closed 2020-11-12 by vvip-68

Engineerator, 2020-11-12:
Thank you, I was able to run this code. I plan to add more effects but when I add another effect `void processEffect(byte aMode) {
  switch (aMode) {
    case MC_SNOW:                snowRoutine(); break;
    case MC_FIRE:                fireRoutine(); break;
  }
}` my matrix becomes blank again.

vvip-68, 2020-11-12:
Put into ```processEffect(byte aMode)``` procedure following line: 
```
Serial.println("aMode=" + String(aMode));
```
and see which value has the switch variable aMode inside procedure and which branch of switch operator (or none of them) is executed.

Engineerator, 2020-11-12:
I put this code into processEffect(byte aMode) and the serial monitor printed out random values each minute.
aMode=12
aMode=15
aMode=31

vvip-68, 2020-11-12:
It is because time of show for each effect is one minute by default. Each minute ID of effect changed to next. 
12 - MC_NOISE_CLOUD
15 - MC_NOISE_RAINBOW
31 - MC_ANALYZER
None of branch of your reduced switch operator belongs to these codes. You have only branches for MC_SNOW=2 and MC_FIRE=6. 
Because you have no procedures drawing images on matrix for codes except MC_SNOW and MC_FIRE - you have blank matrix.
I do not understand why you surprised and what do you want finally.

You have a car. You removed some important gears from engine and transmission and wonder why your car stopped to move.
You removed some important parts of code from sketch and wonder why it stopped to work.
I do not understand why you surprised.

Engineerator, 2020-11-13:
In [GyverMatrixBT ](https://github.com/AlexGyver/GyverMatrixBT/blob/master/firmware/GyverMatrixOS_v1.12/custom.ino#L67) it is possible to change the list of effects the way you wanted. It is also possible in [GyverMatrixWiFi](https://github.com/vvip-68/GyverMatrixWiFi/blob/master/firmware/GyverMatrixOS_v1.15/custom.ino#L127). I'm trying to change the list so that I can add or remove effects that I want and don't want to use so that they switch by themselves without using the app.

vvip-68, 2020-11-13:
List of available effects is defined in a_def_soft.h, lines 36-79. The line 81 contains maximum number of effects.
Lines 19-29 contains list of effects (in defined order corresponded the list of definition in lines 36-79 for transfer into smartphone application. If you need to remove some effects from processing, reoder and renum them in 36-79, correct line 81 with number less then last used effect ID, reorder list of effects in 19-29.

custom.ino line 382 - nextModeHandler() and line 416 - prevModeHandler() sequentially increment/decrement ID for current effect from 0 to MAX_EFFECT or get random number in range 0..MAX_EFFECT depends on random switch on/off.

If you do not use some effect at all, you can remove effect ID definition at all and remove parts of code related with removed ID.
If you need add your own effects - take as a sample one of existing effects, do search among code for MC_XXXX and do almost same as for sampled effect

Engineerator, 2020-11-14:
Perfect! Thank you very much!!! I was able to change the custom list of effects.

When I turn off the [random switch](https://github.com/vvip-68/GyverPanelWiFi/blob/master/firmware/GyverPanelWiFi_v1.06/a_def_soft.h#L560), the program starts with a random effect. From that random effect the program switches to the next effect on the list. Is this possibly a glitch? How could the program be changed so that it starts with the first effect on the list?

vvip-68, 2020-11-14:
Main project file, lines 267-272 retrieve saved mode from EEPROM (if it was set manually) or get random mode on start.
Put following lines instead of:
```
thisMode = 0;
setEffect(thisMode);
```

Engineerator, 2020-11-15:
Thank you! I was able to start the program with the first effect on the list.

Engineerator, 2020-11-16:
I'm trying to use the project to read .out animation files from an SD card. How can I change the program so that it starts with the first file on the SD card and selects the next file on the list?

Currently if the program has more than three effects on the list, it will pick a random file and continue picking random files. I think that [this ](https://github.com/vvip-68/GyverPanelWiFi/blob/master/firmware/GyverPanelWiFi_v1.06/sdcard.ino#L121)line in the sdcard file chooses the next random file on the SD card. However I'm not sure how to change it so that it selects the next file on the list.

vvip-68, 2020-11-16:
If you need sequential files - you need simple increment file_idx until it reach countFiles value. After it reach countFiles value - reset file_idx back to zero. It is obviously, isn`t it?

Engineerator, 2020-11-22:
I'm trying to understand. So far I came up with this:
```
     for(currentFile < 0 || currentFile >= countFiles; currentFile < countFiles; currentFile++){
      file_idx++;
      if(file_idx == countFiles){
        file_idx = 0;
      }
     }
```
I replaced lines 115-125 in the sdcard file with this code. It works, but the first effect is skipped and I don't know why. For example: I have 3 files on an SD card: Effect1.out, Effect2.out and Effect3.out. The program begins with Effect2.out and then plays Effect3.out and then starts from the beginning Effect1.out, Effect2.out, etc. How can I change this code so that it begins with the first effect on the list?

Engineerator, 2020-11-22:
I'm trying to understand. So far I came up with this:
```
     for(currentFile < 0 || currentFile >= countFiles; currentFile < countFiles; currentFile++){
      file_idx++;
      if(file_idx == countFiles){
        file_idx = 0;
      }
     }
```
I replaced lines 115-125 in the sdcard file with this code. It works, but the first effect is skipped and I don't know why. For example: I have 3 files on an SD card: Effect1.out, Effect2.out and Effect3.out. The program begins with Effect2.out and then plays Effect3.out and then starts from the beginning Effect1.out, Effect2.out, etc. How can I change this code so that it begins with the first effect on the list?

Thank you very much for your help!!!

vvip-68, 2020-11-22:
It is very strange construction - use cycle 'for' without index variable. I do not understand logic how it works. 
But If you use this construction and you said it works...

Obviously your currentFile and file_idx initially have values equal to zero.
Next, in the very first stage (ever before first usage)  you increment values and they become 1 and next use them in first time.
As solution - try in line 113 use following: currentFile = -1; 
and initialize file_idx as: 
file_idx = -1;

Engineerator, 2020-11-22:
The problem is solved! Thank you very much!!!

Evgeni198, 2020-12-02:
> Проблема решена! Большое спасибо!!!
Здравствуйте, может Вы мне поможете. Решил убрать ненужные эффекты например "Часы" убрал его из строки 36. Также в строке #define MAX_EFFECT уменьшил на один эффект и компиляция не проходит :

[image]

Engineerator, 2021-05-19:
Hello @vvip-68 please delete this issue. Thank you.

Engineerator, 2021-05-21:
@vvip-68 Delete this issue. Thank you.

Engineerator, 2021-05-21:
@vvip-68 Please delete this issue. Thank you.

Engineerator, 2021-05-21:
@vvip-68 Please delete this issue thank you.