vvip-68/GyverPanelWiFi #23: Ночные часы

AGRO666, 2020-11-13:
А как должны работать Ночные часы? Ставлю по времени включение Ночных часов, время наступает и панель тухнет полностью. Панель 18х10, если это важно.

vvip-68, 2020-11-13:
Должны отображаться часы на черном фоне минимальной яркостью одного из 7 выбранных цветов - красный, синий, зеленый, желтый, голубой, розовый, белый. Естественно, если время инициализировано и известно системе. Если время неизвестно - ничего отображаться не будет - просто черный экран.

AGRO666, 2020-11-13:
Ага, нашел, работает оказывается, просто на матрице с тонированным стеклом не очень видно.

closed 2020-11-13 by AGRO666

vvip-68, 2020-11-13:
clock.ino строка 1132 - функция getNightClockColorByIndex()
Увеличте там яркость ночных часов на какую вам нужно.