vvip-68/GyverPanelWiFi #24: Ошибка в апк

ClausZP, 2020-11-13:
Приключении настроек времени выпадает ошибка прикрепил скриншот. Телефон Xiaomi 9t(k20) прошивка miui 12.
[image]

ClausZP, 2020-11-13:
[image]

vvip-68, 2020-11-13:
Ну да, бывает. У вас какие-то неинициализированные настройки в прошивке и/или ошибка передачи из прошивки в приложение. Приложение на смартфоне - слабо управляемый конструктор, забитый до своего предела и не поддающийся практически уже модификации. Поэтому - что есть, то есть. Пробуйте еще раз. В большинстве случаев - получается.

closed 2020-11-13 by vvip-68

ClausZP, 2020-11-14:
Ошибка возникает просто при переходе во вкладку, если не нажимать "end application" окошко закрывается, и дальше работает. Спасибо за проделанный труд, и быструю реакцию.

vvip-68, 2020-11-14:
У меня такой ошибки не возникает - поэтому я даже проверить не смогу что может быть не так. 
Если вы что-то меняли в прошивке, связанное с этим - откатывайте назад. Или на прошивке, скачанной с репозитория без изменений тоже вот так выходит?