vvip-68/GyverPanelWiFi #25: Проблема с Wifi

mcvall, 2020-11-15:
Приветствую! Плата NodeMCU. Проблема с Wifi.
Что я делаю. Подключился со смартфона к точке доступа "PanelAP", все ок, управляется, в IDE в мониторе порта команды поступают при любых действиях в приложении. Далее в приложении ввожу название сети роутера, пароль, IP-адрес роутера, сохраняю и все! Точка доступа "PanelAP "отключается и приложение переходит в статус "не подключен", при этом в мониторе порта мне говорят, что плата подключилась к роутеру и принимает данные о погоде. Приложение становится неактивным, в настройках WIFI смартфона больше нет точки доступа "PanelAP". Решает проблему только перезагрузка платы

vvip-68, 2020-11-16:
Там же в мониторе порта пишется какой адрес получила плата при подключении к сети WiFi
Впрочем, вы его должны были ввести в приложении там же где имя сети и пароль.
Совершенно непонятно зачем вы туда вводите IP адрес роутера.
Понятно же, что плата получит тот же адрес что у роутера, в ctmb будет два устройства с одинаковым IP и ничего работать не будет.
Далее в приложении в настройках подключения указываете этот IP адрес который получила плата.

mcvall, 2020-11-16:
Давайте сначала.
- В скетче меняю значение #define EEPROM_OK
- В настройках сети в скетче ничего не меняю
- Заливаю скетч в плату
- Ок
- Открываю монитор порта
- Захожу в настройки Wifi на смартфоне, подключаюсь к точке доступа "PanelAP"
- Запускаю приложение, все работает, в мониторе порта появились данные
- В приложении в настройках подключения ставлю галочку напротив "Влючить точку доступа"
- Сохнанить
- Приложение отваливается, статус "Не подключено"
- Захожу в настройки Wifi на смартфоне, подключаюсь к точке доступа "PanelAP"
- В приложении в настройках подключения ввожу Имя сети, пароль, IP НЕ ВВОЖУ
- Сохнанить
- Приложение отваливается, статус "Не подключено"
- Захожу в настройки Wifi на смартфоне, подключаюсь к точке доступа "PanelAP"
- В приложении в настройках подключения напротив "IP" прописался адрес 192.168.0.175
- В мониторе порта написано, что плата подключилась к Wifi с таким же адресом 192.168.0.175
- Приложение отваливается, статус "Не подключено" 
- Захожу в настройки Wifi на смартфоне, точки доступа "PanelAP" УЖЕ НЕТ
- В мониторе порта все ок, плата подключена к роутеру, данные передаются
- Приложение не подключается!

После перезагрузки платы все по новой.
Как оно должно вообще работать, не пойму. Постоянно перезагружать плату и коннектится к точке доступа "PanelAP", чтобы управлять со смартфона? Но тогда ведь не будет интернета на смартфоне.

vvip-68, 2020-11-16:
Оно должно работать так:
1. Чтобы все начисто и не влияло то, что в предыдущий раз наколбасили - В скетче меняю значение #define EEPROM_OK
2. Залили скетч в плату. Монитор порта открыт.
3. Захожу в настройки Wifi на смартфоне, подключаюсь к точке доступа "PanelAP"
Запускаю приложение, все работает, в мониторе порта появились данные
4. В приложении в настройках соединения вводите имя вашей локальной сети, пароль доступа в эту сеть и в поле IP адрес - свободный адрес в вашей сети - например тот же 192.168.0.175
Нажимаете кнопку "Сохранить". Может быть несколько раз - UDP, знаете ли - без гарантированной доставки пакетов...
5. В результате приложение должно со временем "отвалиться" (не подключено) на матрице бегущей строкой должен отобразиться IP адрес который получила плата, этот же адрес - в мониторе порта. А также всякая шняга про получение времени, погоды и т.п.
6. На страничке подключения приложения нажимаете кнопку "Новое подключение", задает ему имя - например "Моя панель", указываете там IP адрес, который получила ваша панель - 192.168.0.175. Порт не меняете (вы же его не меняли в прошивке). Нажимаете кнопку "Сохранить" и/или "Подключить". 
7. Смартфон подключается к матрице, управляете матрицей из смартфона.
8. На этом всё.

vvip-68, 2020-11-16:
Почему рекомендуется вводить адрес там где вы написали "IP адрес не ввожу":
В этом случае при загрузке микроконтроллера он получает адрес в сети по DHCP - то есть фактически случайный. Вы видите в его в мониторе порта. Далее настраиваете приложение, чтобы оно подключалось к этому адресу. Все подключается и работает. Однако в один прекрасный момент ваш роутер может выдать плате другой адрес. И если монитор порта не включен - вы это не увидите. А вот приложение просто перестанет подключаться по "старому" адресу который вы настроили и который запомнило приложение.

mcvall, 2020-11-16:
Ура! Заработало. Спасибо большое!

closed 2020-11-16 by vvip-68

jdc85, 2021-12-12:
Приветствую. Был проэкт на вемос д1мини, всё работало версия 1.07. Сейчас перешил на 1.12. Установил приложение версии 1.12, пытаюсь подключится не получается. К точке доступа PanelAP подключаюсь, ip тоже нормальный. Стоит андроид 10.Что может быть не так? Спасибо большое

vvip-68, 2021-12-12:
"Не так" может быть, то, что не выполнена настройка подключения к сети - не задано имя сети и/или пароль доступа к этой сети.

jdc85, 2021-12-12:
Немного разобрался, на втором телефоне с 10 андрюшей работает. Не пойму в чем проблема. И тот и другой рутованные Xiaomi.

jdc85, 2021-12-12:
Всё решил вопрос. Стояла подмена Host файла в Magiske. Видимо из-за рекламы я его заблокировал! Удалил патч и всё вернулось на круги своя! Спасибо большое! Автору удачи в его делах и с Наступающим Новым Годом!

Demons1984, 2022-04-09:
Всем привет, Захожу в настройки Wifi на смартфоне, жамкую подключится к точке доступа "PanelAP". А не подключается, пишет получение IP  ESP8266 Подскажите в чем может быть проблема?

Bahyty, 2022-12-25:
Подскажите, в чем может быть проблема: подключилось, на экране все отображается, от кнопки переключается, но в приложении нет настроек самой гирлянды - эффекты, яркость, бегущая строка и пр.

vvip-68, 2022-12-25:
Возможно вы используете не то приложение. Их там три. Попробуйте другое.

Bandit625, 2023-11-22:
Добрый вечер! Помогите пожалуйста, уже вторую неделю сижу, перечитываю, пробую, меняю и ничего не получается. Включаю плату, подключаюсь к ней. Захожу в приложение, подключаюсь, иногда бывает успеваю что то настроить и программа вылетает. Пробую подключиться но не получается. В чём может быть проблема?

vvip-68, 2023-11-23:
Нет ответа на ваш вопрос. Про такие случаи никто не писал. Что у вас за телефон? Сколько в нем памяти? Есть ли другой телефон, чтобы попробовать на нем?