vvip-68/GyverPanelWiFi #26: Рисование через приложение

masterpro100, 2020-11-16:
Здравствуйте! Проект просто супер! Запись эффектов на microSD карту - то, что так было мне нужно. 

    Есть пожелание по режиму "Рисование" , которое было в проекте GyverMatrixWiFi. Очень хотелось бы видеть его в этом новом проекте. Детям очень нравилось рисовать картинки на всё окно балкона. Я понимаю, что ответ будет таким же, как и про вопрос про игры, но всё же, вдруг есть возможность его добавления.

vvip-68, 2020-11-16:
Ответ будет таким же как и про игры. Приложение, которым всё это управляется - написано в "конструкторе" Thunkable и достигло своего предела, когда никаких элементов в него уже добавлять не получается. Оно просто перестает компилиться.
Используйте ту прошивку в которой можно рисовать и играть в игры. Там, правда нет поддержки SD карты...
Предлагаю сделку: вы пишите на native Android (на Kotlin например) приложение, которое всем управляет, в т.ч. и позволяет рисовать - я добавляю функции рисования в прошивку. Идет?

closed 2020-11-16 by vvip-68

masterpro100, 2020-11-16:
Спасибо за быстрый ответ! Дети пусть в альбомах рисуют, а гирляндой будем просто любоваться! Спасибо за проделанную работу, очень крутой проект!

vvip-68, 2020-11-16:
Соберите две гирлянды. Ну то есть - одну на окно большую (там точки далеко друг от друга - рисование не так красиво будет) и другую - малого размера из ленты 30 диодов на метр. Ячейка будет 3х3 см. Если решетку еще сделаете (да хоть бы из ватмана), чтобы границы "клеток" четкими были - для рисования вообще атас будет. Ну, а для повседневной жизни это будут часы. Возможно с будильником. Например 30x20 - отличный размер. 600 клеток.  
Вот тут, например - https://aliexpress.ru/item/2036819167.html
если выбрать WS2812BECO White PCB отрезки 1m IP30 CHINA - 97 руб за полоску из 30 диодов. Закажите 20 полосок (600 диодов) - выйдет меньше чем 2000. Вроде 27.11 еще какая-та черная пятница намечается. В распродажу 11.11 полоска стоила 87 руб...

masterpro100, 2020-11-16:
Отличный совет, обязательно воспользуюсь! Гирлянда уже собрана, а экран из ленты собирал на заказ, теперь и для себя наконец-то соберу.

fastdron, 2020-11-17:
Привет. По приложению, скоро на работе прогер обещал помочь. А пока что сам пробую различные варианты, но не хватает знаний) Пробую пока что прикрутить Blynk и управлять из внешки не по MQTT а с приложухи blynk, уже можно вкл выкл, и режимы переключать. Хотел ещё спросить по поводу Ваших последних правок - теперь вместо "ночных часов" прото "часы" но вот беда - через приложение нельзя задать цвет однотонный часам, кнопки не работают,часы просто белые

vvip-68, 2020-11-18:
По рисованию.  И, возможно, играм. Думаю, что добавлю в прошивку такую возможность. Для рисования и игр будет отдельное приложение (переделанное из GyverMatrixOS). Работа идет.
Просто часы. Народ часто просил просто часы без всяких эффектов. На черном фоне.  Ночные часы включаются с кнопки на первом экране (или автоматически по расписанию), Дневные часы - из эффектов. 
Цвет дневных часов: там очень много эффектов, на которых белые часы (или часы заданного цвета) не видны. Например - белые часы на снеге. Или на эффектах типа Лава или Облака. Для этих эффектов используются контрастные часы фиксированного цвета. Например - зеленые или желтые. И тут уже у людей массово возникают вопросы - "а почему я задаю цвет часам (по кнопке Цвет монохромных часов), а он все равно показывает другим цветом?". А заданный пользователем цвет - он на меньшинстве эффектов только доступен был. Чтобы не сводить с ума пользователей и избежать таких массовых вопросов - часы в таких эффектов было решено сделать белыми, а кнопку из приложения убрать.

vvip-68, 2020-11-18:
Могу для дневных часов попробовать сделать отдельную настройку - только для этого эффекта - ползунком "Вариант" менять цвет отображения дневных часов.

masterpro100, 2020-11-18:
Отличная новость! Прошивка будет одна, а приложения два, это очень круто! Буду внимательно следить за вашим проектом. Благодарю за вашу работу, что не забрасываете проект, а развиваете дальше!

vvip-68, 2020-11-21:
@masterpro100, я выложил прошивку в которой поддерживается рисование на матрице и 4 игры - Лабиринт, Змейка, Тетрис, Арканоид. Вроде бы у меня все работает. Тестируйте. Если найдете ошибки - пишите.
Для управления всем этим - дополнительно к GyverPanelWiFi.apk нужно поставить приложение GyverPlayerWiFi.apk
Они лежат в проекте в папке Android рядом

masterpro100, 2020-11-21:
Премного благодарен, отдельное спасибо от моих детей! Вы отличный специалист!

masterpro100, 2020-12-03:
Доброе утро! Спасибо за прошивку с играми и рисованием. У меня всё работает отлично, кроме игр. С приложения GyverPanel рисует, картинки преобразует. А вот в играх ничего не происходит, гирлянда не светится. Есть ли ограничения в минимальных размерах гирлянды (тестирую на размере 15*10 пикселей)? Или может где-нибудь в прошивке требуется раскомментировать строку с играми?

vvip-68, 2020-12-03:
Нет, ничего не нужно раскомментировать. Все должно работать и так. И у меня работает.
Ограничений на размер гирлянды нет. С приложнения "GyverPanel" вообще-то не должно рисовать - там нет такого функционала. Для рисования было сделано отдельное - GyverPlayer.
Когда вы открываете GyverPlayer, переходите на вкладку "Игры" и нажимаете на панельку одной из 4 игр - что происходит? Должно нарисоваться поле игры и она встать в режим "пауза". При нажатии любой кнопки управления в приложении - начаться действие. Может быть станет понятнее что происходит, если посмотреть что в это время происходит в мониторе порта, какие там сообщения?

masterpro100, 2020-12-03:
Понял, попробую посмотреть в мониторе порта. А про приложение - я ошибся, конечно в GyverPlayer. При переходе на вкладку "Игры" ничего не происходит. Гирлянда не светиться, на кнопки управления не реагирует.

vvip-68, 2020-12-03:
Ну, то есть вот вы что-то, к примеру, рисовали-рисовали, потом перешли на вкладку "Игры", нажали на одну из игр и у вас - хлоп, и черный экран?
А не определен ли у вас DEVICE_ID **4** ?

masterpro100, 2020-12-03:
Режим DEVICE_ID 4, а какой нужно? 

Если включен какой-либо эффект, то просто пауза ставиться.

masterpro100, 2020-12-03:
Наверно мне нужно как панель на стене сделать - DEVICE_ID 3, правильно?

masterpro100, 2020-12-03:
Блин, прошу прощения, не внимательно прочитал, мне действительно надо было выбрать DEVICE_ID 3. Спасибо! Сейчас скину рубль в качестве благодарности!

vvip-68, 2020-12-03:
@masterpro100 - ну то есть мы из тех, кто нажимает кнопку "загрузить" особо не вчитываясь в комментарии к настройкам, где черным по белому в строках 22-27 чуть ниже определения этого самого DEVICE_ID (хотел сказать - крупными буквами) написано

```
// Внимание!!! 
// DEVICE_ID == 4 имеет размер матрицы - 1000 светодиодов, что уже довольно много для ESP8266, и в данном варианте прошивки при всех включенных возможностях
// прошивка начинает "падать" по причине недостатка памяти - функция malloc() не может выделить память. 
// Для устройства с этим DEVICE_ID в целях устранения этих проблем в прошивке отключены некоторые возможности - например отключены игры Лабиринт, Змейка, Тетрис, Арканоид.
// Вполне вероятно, что какие-то возможности тоже [будут] урезаны. Для больших матриц рекомендуется использовать ESP32 у которого памяти значительно больше.
// Если вы повторяете этот проект для матриц меньшего размера - берите за основу блок для других DEVIVE_ID и настраивайте параметры под себя.
'''

vvip-68, 2020-12-03:
Да, с DEVICE_ID равным 3 игры будут работать

masterpro100, 2020-12-03:
Да, к сожалению я такой:) Проверил, всё работает. Сейчас счёт пополню, вышлю благодарочку!

masterpro100, 2020-12-03:
А можно мне на почту (masterpro84@mail.ru) скинуть номер телефона (если привязан к карте), или номер карты. Не могу пополнить счёт на PayPal. Мне удобнее на карту перевести.

vvip-68, 2020-12-03:
@masterpro100 а откуда вы paypal взяли?
А там же есть ниже еще одна ссылка - она открывается или я где накосячил и исправлять нужно?
Если не открывается - пишите - тогда на почту отправлю

vvip-68, 2020-12-03:
@masterpro100 -  Благодарю за благодарочку :)

masterpro100, 2020-12-03:
Я дважды не внимательный:) Ссылка рабочая, просто не совсем понял, что это по оплате:) Перевел, спасибо!

vvip-68, 2020-12-03:
@masterpro100 и вам спасибо :)

masterpro100, 2020-12-03:
Ещё последний вопросик задам и всё. Нет ли возможности сделать изменение размеров гирлянды с приложения, например GyverPlayer? Ну это чтобы, если вдруг захочется увеличить размеры гирлянды на следующий год, или использовать один контроллер для гирлянды и панели на стене с разным количеством пикселей, не пришлось перепрошивать Wemos.

vvip-68, 2020-12-03:
То есть вы хотите один и тот же вемос к разным гирляндам цеплять?
А в чем смысл? Экономия 130-150 рублей на вемосе? Одна гирлянда работает, другая лежит?
Поигрались с одной, подключили вемос к другой?

Какая-то копеечная экономия. Которая по факту для тех кто повторяет - породит массу вопросов - спаял, подключил - ничего не работает вместо эффектов какая-то каша.

Сделать можно - но смысла в этом не вижу никакого.

masterpro100, 2020-12-03:
Тоже верно! Смысл в том, что, например, отправил я родственнику в другой город контроллер прошитый, т.к. родственник далёк от владения компьютером. Он подключил ленту и радуется. А в следующем году он решил увеличить ширину гирлянды, добавив ленту с алиэкспресс. А контроллер нужно новый приобретать, прошить и ему отправлять. Сам контроллер стоит копейки, а почта за пересыл берёт около 300 руб. Общая экономия была бы уже 500 руб. Но это редкий случай, и раз в год уж можно выложить 500руб. Лучшее - враг хорошему! Вопрос отпадает!

vvip-68, 2020-12-03:
То есть родственник настолько далёк от темы, что сам прошить контроллер не может, а вот купить подходящие диоды на али и подключить их в работающую систему может? Как-то не вяжется.
В любом случае - есть интернет и программы удаленного доступа к компьютеру. Вы можете удаленно зайти к нему на комп и выполнить все требуемые действия, дав команду родственнику - включить USB шнурок одним концом в компьютер, другим - в микроконтроллер. Все остальное можно сделать удаленно.

masterpro100, 2020-12-03:
Ваша правда! Полностью согласен.

vvip-68, 2021-01-05:
>А как насчет возможности сохранить свой рисунок, чтобы использовать в след. раз готовый?
На ту же SD карту например...

Как раз в разработке... Ещё не готово

simonovich, 2021-01-06:
Ещё в копилку: проигрывание gif анимации

vvip-68, 2021-01-06:
Этим я на 99% заниматься не буду. Библиотек, работающих с гифками мне не встречалось, сам я потрошить этот формат не буду. Кроме прочего - обычный размер матриц 16ч16 - таких гифок практически енте. А брать гиф не в размер матрицы, расшифровывать кадр и масштабировать его с сохранением нормального качества (бикубическими сплайнами) - точно работа не для хилого микроконтроллера...

simonovich, 2021-01-06:
Ok, понял
Это для инфы https://github.com/marcmerlin/FastLED_NeoMatrix