vvip-68/GyverPanelWiFi #28: Ошибка при компиляции a_def_pass.h: No such file or directory

bluegreensugar, 2020-11-18:
Собираю на wemos d1 mini, при компиляции выдает ошибку a_def_pass.h: No such file or directory

vvip-68, 2020-11-18:
В файла a_def_soft.h комментарии в строках 287-289 читали?

closed 2020-11-18 by vvip-68

Crezidarkman, 2022-11-03:
а можно как-то получить этот фаил a_def_pass.h, чтобы на базе него свой сделать. Заранее благодарю!

vvip-68, 2022-11-03:
// ------------- WiFi & MQTT parameters --------------
#undef  NETWORK_SSID
#undef  NETWORK_PASS
#undef  DEFAULT_IP

#define NETWORK_SSID "SSID_NAME"                    // Имя WiFi сети
#define NETWORK_PASS "SSID_PWD"                     // Пароль для подключения к WiFi сети

#undef  DEFAULT_MQTT_SERVER
#undef  DEFAULT_MQTT_USER
#undef  DEFAULT_MQTT_PASS
#undef  DEFAULT_MQTT_PORT
#undef  DEFAULT_MQTT_PREFIX

#define DEFAULT_MQTT_SERVER   "mqtt.by"             // MQTT сервер по умолчанию
#define DEFAULT_MQTT_USER     "name"                // Имя mqtt-пользователя по умолчанию 
#define DEFAULT_MQTT_PASS     "password"            // Пароль mqtt-пользователя по умолчанию
#define DEFAULT_MQTT_PORT     1883                  // Порт mqtt-соединения TCP по умолчанию

// API-идентификатор сервиса получения погоды
#undef  WEATHER_API_KEY
#define WEATHER_API_KEY "553cfcc9ac3fa47794e1fb936fe66fd8"

#if defined(ESP8266)  
#define DEFAULT_MQTT_PREFIX   "af7cd16a"            // Префикс топика сообщения устройств ESP8266
#endif
#if defined(ESP32)  
#define DEFAULT_MQTT_PREFIX   "af7cd12a"            // Префикс топика сообщения устройств на ESP32
#endif