vvip-68/GyverPanelWiFi #29: Постоянная перезагрузка

jkreet, 2020-11-19:
Собрал панель на базе ESP-12E.
Включается, работает нормально.
Пока не подключена к wifi, всё ок. Эффекты переключаются, работает демо-режим. Можно подключиться к ней, как к точке доступа и управлять с телефона.
Стоит только подключить к домашнему wifi начинаются проблемы.
При включении показывает случайный эффект и часы с температурой.
Затем часы уезжают влево, и через секунду - reset.
В мониторе порта сообщения по кругу:
------------------------------------------------------------------------------------------------------------------------------
ED-Panel-WiFi v.1.06.2020.1114

Загрузка строк выполнена.
Ипользованы адреса EEPROM 800 - 837
Свободно ячеек 3258

Подключение к Keenetic
...........
WiFi подключен. IP адрес: 192.168.1.102

UDP-сервер на порту 2390
NTP-сервер ru.pool.ntp.org -> 94.103.91.82
Отправка NTP пакета на сервер ru.pool.ntp.org
UDP << ip='94.103.91.82:123'; ntp sync
UDP пакeт размером 48
Разбор пакета NTP
Секунд с 1970: 1605788805
--------------------

Запрос текущей погоды
Погода получена:
Сейчас: пасмурно, +2ºC
Код иконки: 'ovc'

Цвет неба: '#58bbfe'

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4020b658 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40020173 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd00 end: 3fffffc0 offset: 0190
3ffffe90:  00000009 00000005 00000002 00000006  
3ffffea0:  00000002 00000000 00000002 00000006  
3ffffeb0:  00000012 0000001c 00000001 00000000  
3ffffec0:  ffffff9c 00000019 00000001 00000000  
3ffffed0:  3ffe8559 00000009 00000001 40214b97  
3ffffee0:  00000000 00000000 4bc6a7f0 00000000  
3ffffef0:  00000000 80ff1544 40100568 000039df  
3fffff00:  00000000 00000000 3fff0ad4 80fe8574  
3fffff10:  00000001 00000001 00000001 00000000  
3fffff20:  3fff0a9c 3fff09aa 3fff0948 3ffe8574  
3fffff30:  3fff0a9c 3fff09aa 3fff0ad4 40218c76  
3fffff40:  3fff3900 001f001f 80ff0978 3fff3600  
3fffff50:  001d02bf 800c000f 3fff3900 3fff10c8  
3fffff60:  3fffdad0 00000000 3fff1544 40224fe1  
3fffff70:  00000000 80efeffe 3fff0e80 40223536  
3fffff80:  00000000 00000000 00000001 3fff10c8  
3fffff90:  3fffdad0 00000000 3fff1088 40218fe7  
3fffffa0:  3fffdad0 00000000 3fff1088 4022186c  
3fffffb0:  feefeffe feefeffe 3ffe85d8 40100fed  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

------------------------------------------------------------------------------------------------------------------------------

kostyamat, 2020-11-19:
Судя по этому
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
сработал аппаратный вачдог. Где-то программа зациклилась.

vvip-68, 2020-11-19:
Не, когда Watchdog - оно так и пишет, что SoftReset или Watchdog reset. Тут что-то другое.
Может и в часах дело. Какой размер матрицы? На последней версии прошивки из репозитория - тоже ошибка есть? Может уже исправлено?

jkreet, 2020-11-19:
Матрица 16x16
Пробовал только последнюю версию прошивки.
По поводу ESP Exception Decoder - не силён в этом. Но если подскажете, готов попробовать.

vvip-68, 2020-11-19:
https://github.com/me-no-dev/EspExceptionDecoder
Ставишь как написано. 
В меню IDE "Инструменты"  должно появиться "ESP Exception decoder"
Выбираешь этот пункт. Появляется окно, в которое копируешь вот тот стек с ошибками.
Оно показывает стек вызовов, по которому можно примерно понять что происходит.
Лучше то что получил кинуть мне в личку в https://vk.com/iivanovii - похоже разговор будет долгий.
Ни у кого ошибки не возникает, у тебя возникает. Это странно.
Давай туда же свой блок настроек - какие фичи включены / выключены, настройки размеров матрицы, угла подключения и все прочее связанное

jkreet, 2020-11-19:
У меня нет вк.
`0x40216e67: doEffectWithOverlay(unsigned char) at Z:\arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.06/custom.ino line 284
0x40100500: millis() at C:\Users\Pasha\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 167
0x4021a98a: process() at Z:\arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.06/a_main.ino line 195
0x40100568: millis() at C:\Users\Pasha\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40106553: esp8266::polledTimeout::timeoutTemplate   >::expiredOneShot() const at C:\Users\Pasha\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/PolledTimeout.h line 218
0x40226d8c: esp8266::MDNSImplementation::MDNSResponder::_updateProbeStatus() at C:\Users\Pasha\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 1321
0x4022b140: PubSubClient::connected() (C:\Program Files at x86)\Arduino\libraries\PubSubClient\src\PubSubClient.cpp line 689
0x40205bc3: checkMqttConnection() at Z:\arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.06/mqtt.ino line 44
0x40227845: esp8266::MDNSImplementation::MDNSResponder::_process(bool) at C:\Users\Pasha\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 90
0x40225d9a: esp8266::MDNSImplementation::MDNSResponder::update() at C:\Users\Pasha\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266mDNS\src\LEAmDNS.cpp line 1337
0x4021ad27: loop() at Z:\arduino\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.06/GyverPanelWiFi_v1.06.ino line 297
0x402240d0: loop_wrapper() at C:\Users\Pasha\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 197`

vvip-68, 2020-11-19:
Ничего не понятно... Ну возможно да - по WatchDog падает - где-то что-то зацикливается
У тебя MQTT включен - он реально настроен и ты его используешь?
Взятый с репозитория проект - компилится, не вылетает?
Если проект без изменений не вылетает, а после твоих изменений проблемы - смотри где что правил, откатывай назад, вноси правки по одной и смотри на каком моменте валиться начинает...

jkreet, 2020-11-19:
Правок не вносил. 
Только переменные под свои желания поправил.
mqtt пробовал отключать - без него то же самое.

jkreet, 2020-11-19:
Ради интереса собрал, отключив погоду:
#define USE_WEATHER 0

И всё заработало.

vvip-68, 2020-11-19:
Какой город погоды? Погода - яндекс или OpenWeatherMap?

jkreet, 2020-11-19:
Москва, Яндекс.

jkreet, 2020-11-19:
Но, похоже, дело вообще не в погоде.
Я сейчас экспериментировал - переключил часы в вертикальное отображение - всё заработало как надо.
Погоду тоже подключил - и она работает.

Включил часы поверх эффектов и горизонтально - снова словил перезагрузку. Но теперь не циклическую, т.к. в eeprom не сохранилась настройка с горизонтальными часами.

И сейчас могу, переключив часы в горизонт, снова вызвать ребут.

vvip-68, 2020-11-19:
Интересно почему у меня такого не происходит?
Видимо, тебе придется это отлаживать самостоятельно...
Потом расскажешь мне в чем было дело.

Дай свои настройки все что ты исправлял в исходном проекте.

closed 2020-11-20 by vvip-68

fastdron, 2020-11-21:
Вот буквально сегодня обновил прошивку до последней актуальной - тоже все супер,не вылетает. Хотел попросить автора добавить в шрифты символ ° . Для отображения температуры очень не хватает в бегущей строке, особенно когда ноль градусов на улице - всегда дописываю после значения температуры 'С.

vvip-68, 2020-11-21:
@fastdron,  забирай обновления