vvip-68/GyverPanelWiFi #30: Не работает текстовая строка

werwolf68, 2020-11-22:
Добрый день!
Через приложение добавляю фразы для бегущей строки. 
Если на вкладке текст нажимаю просмотр, все работает. Через отжим демо - нет. Подскажите, в чем может быть проблема?

vvip-68, 2020-11-22:
Например на вкладке настройки текстовых строк не поставлена галка  "Текст поверх эффектов". 
Или интервал показа текстовых строк установлен большим и не успевает истечь, когда вам надоедает ждать появления строки (например интервал указан - "каждые в 600 секунд", а вы ожидаете только 300 секунд, потом вам надоедает ждать.
Или для (каждого) эффекта в его настройках на закладке "Эффект" не стоит галка, разрешающая отображать бегущую строку

werwolf68, 2020-11-22:
Да, действительно вы правы. 
Я отключил наложение на эффекты и в настройках отключил все галочки. (По моему мнению с эффектам текст смотрится ужасно, но решил эту особенность черным фоном). Всё сейчас отлично работает.
Огромное Вам спасибо за такой классный и большой проект.

werwolf68, 2020-11-22:
Добрый вечер. Ваш совет помог. Огромное Вам спасибо!Могу ли я Вас материально отблагодарить? Каким способом?Для меня предпочтительнее карта visa/mastercard

Отправлено из мобильной Яндекс.Почты: http://m.ya.ru/ymail

closed 2020-11-22 by werwolf68

vvip-68, 2020-11-22:
Чтобы текст поверх эффектов не сливался с фоном эффекта - у каждого эффекта есть настройка "контрастность" - это яркость эффекта относительно яркости текста / часов. Можно подобрать такую яркость, что текст будет достаточно хорошо читаться на фоне эффекта. Черный фон, конечно, тоже выход...
Ну, если есть желание - кинь чего-нибудь на карту сбербанк/виза 4276 31000 2491 1960

werwolf68, 2020-11-22:
Спасибо за объяснение, контрастность не думал что так работает здесь.А в номере карты нет ли у вас ошибки?Возможно в второй группе цифр лишний 0?

vvip-68, 2020-11-22:
Я, вроде исправил в сообщении выше - 4276 3100 2491 1960

vvip-68, 2020-11-22:
Спасибо!

247760, 2020-12-02:
Вопрос по тексту и макросам. Для проверки, я скопировал и вставил в окно приложения готовый макрос 
До Нового года осталось {P01.01.****#E#60#60}
-С Новым годом!!!!
Но **COM** порт мне пишет в ответ:
17:55:01.171 -> --------------------
17:55:01.171 -> Строки с событием {P}
17:55:01.171 -> --------------------
17:55:01.205 -> Строка: 'До Нового года осталось {P01.01.****#E#60#60}  -С Новым годом!!!!'
17:55:01.205 -> 
17:55:01.205 -> Ошибка в макросе '{P01.01.****#E#60#60}'
17:55:01.205 -> -------------^
17:55:01.205 -> --------------------

На данную минут, вообще не понимаю структуру написания макросов (но это моя проблема), хотелось бы на одном примере всё разобрать.

vvip-68, 2020-12-02:
Поизучайте пока другие макросы. Я пока посмотрю что с этим пошло не так...

werwolf68, 2020-12-02:
Добрый вечер. Я думаю это письмо адресовано не мне

247760, 2020-12-02:
@werwolf68  я написал автору тут, чтобы не открывать новые вкладки с вопросами.

vvip-68, 2020-12-02:
@werwolf68, да это в вашей теме (issue #30) хулиганит (задает вопросы) @247760 - ему и был ответ.
Ну, а гитхап естественно рассылает всем, кто в теме уведомления

vvip-68, 2020-12-02:
@247760 - я исправил. Странно, что и я и другие пользователи этот макрос проверяли и ведь он работал... Как и когда ошибка закралась? Чудеса...