vvip-68/GyverPanelWiFi #31: Бегущая строка на другом языке.

vAkUuMkA, 2020-11-22:
Здравствуйте. Есть ли возможность редактировать шрифт? Хочется добавить бегущую строку к примеру на чешском языке. И чешские буквы (такие как *ч*) отображаются не корректно. Или можно указывать региональные настройки? Спасибо.

vvip-68, 2020-11-22:
Нет, поддерживается только русский/английский.
Для дополнительных букв нужно
а) в файле fonts.h нарисовать изображение буквы по аналогии с имеющимися
б) в runningText.ino в строке 168 нужно написать соответствие между кодом символа (чешского алфавита) и индексом изображения буквы в массиве шрифтов.  Код чешской буквы, передаваемой в эту функцию вычисления индекса в массиве - getFont() определяется модификатором modif - может быть один для набора символов, фактически это первый байт двухбайтового кода UTF и кода самой буквы - font 
```
uint8_t getFont(uint8_t font, uint8_t modif, uint8_t row)
```

closed 2020-11-22 by vvip-68

vaicekas, 2020-11-30:
Здраствуйте, не нашёл есть ли вазможност переключить язык на английский при отображении времени до события? На пример вместо 30 дней - 30 days.

vvip-68, 2020-11-30:
Нет, только исправив этот вывод в скетче. Файл utils.ino, функция WriteDays() и около нее - часы минуты, названия дней недели и т.п. Если нужны оба языка - нужно создавать, например, новый макрос вместо {R...} и при его о бработке вызывать другие функции, например не WriteDays(), a WriteDaysEng()