vvip-68/GyverPanelWiFi #32: Не проходит компиляция на версии 1.07

tih66, 2020-11-23:
На последней версии не проходит проверку и соответственно не компилируется проект:

In file included from C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31:0,
                 from sketch\a_def_hard.h:314,
                 from D:\GyverPanelWiFi-master\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.07\GyverPanelWiFi_v1.07.ino:60:
C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/IPAddress.h:77:19: error: invalid use of incomplete type 'class IPAddress'
         IPAddress();
                   ^
C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/IPAddress.h:57:7: error: forward declaration of 'class IPAddress'
 class IPAddress: public Printable {
       ^
C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/IPAddress.h:77:19: error: ISO C++ forbids declaration of 'type name' with no type [-fpermissive]
         IPAddress();
                   ^
In file included from c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\memory:67:0,
                 from C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:28,
                 from C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:28,
                 from C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,
                 from sketch\a_def_hard.h:314,
                 from D:\GyverPanelWiFi-master\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.07\GyverPanelWiFi_v1.07.ino:60:
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_raw_storage_iter.h:75:7: error: expected primary-expression before 'explicit'
       explicit
       ^
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_raw_storage_iter.h:74:11: error: expected ';' at end of member declaration
     public:
           ^
In file included from c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\memory:74:0,
                 from C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:28,
                 from C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:28,
                 from C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,
                 from sketch\a_def_hard.h:314,
                 from D:\GyverPanelWiFi-master\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.07\GyverPanelWiFi_v1.07.ino:60:
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext\concurrence.h:70:5: error: expected primary-expression before 'virtual'
     virtual char const*
     ^
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext\concurrence.h:69:9: error: expected ';' at end of member declaration
   public:
         ^
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext\concurrence.h:78:5: error: expected primary-expression before 'virtual'
     virtual char const*
     ^
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext\concurrence.h:77:9: error: expected ';' at end of member declaration
   public:
         ^
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext\concurrence.h:86:5: error: expected primary-expression before 'virtual'
     virtual char const*
     ^
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext\concurrence.h:85:9: error: expected ';' at end of member declaration
   public:
         ^
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext\concurrence.h:94:5: error: expected primary-expression before 'virtual'
     virtual char const*
     ^
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext\concurrence.h:93:9: error: expected ';' at end of member declaration
   public:
         ^
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext\concurrence.h:131:13: error: invalid use of incomplete type 'class __gnu_cxx::__mutex'
     __mutex() 
             ^
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext\concurrence.h:118:9: error: forward declaration of 'class __gnu_cxx::__mutex'
   class __mutex 
         ^
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext\concurrence.h:131:13: error: expected ';' at end of member declaration
     __mutex() 
             ^
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\ext\concurrence.h:132:5: error: expected unqualified-id before '{' token
     { 

и т.д.
Не подскажите в чем загвоздка?

vvip-68, 2020-11-23:
Возможно - в устаревшей версии ядра ESP8266. У вас стоит 2.5.2
Обновитесь до последней - на текущий момент 2.7.4

tih66, 2020-11-23:
Тоже самое
In file included from C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31:0,
                 from sketch\a_def_hard.h:314,
                 from D:\GyverPanelWiFi-master\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.07\GyverPanelWiFi_v1.07.ino:60:
C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/IPAddress.h:78:19: error: invalid use of incomplete type 'class IPAddress'
         IPAddress();
                   ^
C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/IPAddress.h:58:7: error: forward declaration of 'class IPAddress'
 class IPAddress: public Printable {
       ^
C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/IPAddress.h:78:19: error: ISO C++ forbids declaration of 'type name' with no type [-fpermissive]
         IPAddress();
                   ^
In file included from c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\memory:67:0,
                 from C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:28,
                 from C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:28,
                 from C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,
                 from sketch\a_def_hard.h:314,
                 from D:\GyverPanelWiFi-master\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.07\GyverPanelWiFi_v1.07.ino:60:
c:\users\������������\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_raw_storage_iter.h:75:7: error: expected primary-expression before 'explicit'
       explicit

vvip-68, 2020-11-23:
Попробуйте взять совсем последнюю версию с репозитория.
Если ничего в ней не менять, просто скомпилить - те же ошибки?
В меню "Файл" -> "Настройки" - комбобокс "Сообщения компилятора" стоит "Ничего"?
Я не знаю причину. У меня ничего подобного не возникает.
Буду думать. Если что придумаю 0 сообщу.

tih66, 2020-11-23:
Тоже самое. Уже полностью удалил все что связано с ардуино и установил все с нуля, последняя версия вываливается в ошибку.

vvip-68, 2020-11-23:
Еще вариант - у вас в путях - кириллица. То что черными квадратиками в сообщении выше. Раньше Ардуино IDE точно не любил кириллицу в путях. Какой, кстати, версии у вас Arduino IDE? Если не последней - обновите... 
Создайте пользователя в винде не содержащего кириллицу в имени, войдите под ним. Установите Arduino IDE, ядро и библиотеки. Откройте проект из под этого пользователя.
Я не вижу причины почему эта версия может вести себя вот так. Предыдущая версия - собирается?
Ничего серьезного касательно IPAddress на которую ругается - не поменялось. Все то же самое.
В 1.07 только добавилась игра Андроид, включено интерактивное управление играми и добавлена возможность рисования на матрице из программы на смартфоне. Весь остальной код остался тем же самым. Я не вижу причин почему старая версия собиралась, а новая нет.
Единственное изменение 1.07 от 1.06 касаемое IPAddress - в файле clock.ino, строки 88-89 - 
  ip1.fromString(F("0.0.0.0"));
  ip2.fromString(F("255.255.255.255"));
Закомментируйте этот блок, и связанное с ним, попробуйте собрать.
Посмотрите как это было в версии 1.06

vvip-68, 2020-11-23:
Вот эти строки
C:\Users\������������\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/IPAddress.h:78:19: error: ISO C++ forbids declaration of 'type name' with no type [-fpermissive]
IPAddress();
навевают на мысль, что у вас в компиляторе включены какие-то строгие проверки...
Попробуйте в настройках среды выставить вот так:
[image]

tih66, 2020-11-23:
К сожалению смена пользователя на "без кирилицы" не помогла, та же самая ошибка. Настройки у меня выставлены как у Вас. Попробовал в итоге вообще заменить файл clock.ino на файл из 1.06 прошивки. не помогло.

vvip-68, 2020-11-23:
Так а 1.06 - собирается?

vvip-68, 2020-11-23:
Еще отличие 1.06 от 1.07 - в 1.07 были обновлены библиотеки ArduinoJson и ESPSoftwareSerial.
Но они вообще никаким боком к IPAddress не относятся...

tih66, 2020-11-23:
1.06 собирается

vvip-68, 2020-11-23:
Пришлите совсем полный лог ошибок  от нажатия кнопки до результата ошибок. 
На vvip-cобака-маил-точка-ру

tih66, 2020-11-23:
[image]

tih66, 2020-11-23:
это версия 1.06

vvip-68, 2020-11-23:
Ну, что-то у меня нет вариантов. Я не понимаю почему ошибки и на что ругается.
У меня версия 1.07 собирается без ошибок и прямо сейчас работает на 5 устройствах...

[image]

vvip-68, 2020-11-23:
Вообще а другом есть возможность попробовать собрать?
И - у меня Windows 10...
Хотя 1.06 же у вас и на 7-ке тоже собирается. Не знаю...

vvip-68, 2020-11-23:
Пришлите мне на почту архив полная папка вашего проекта, вот прям как он у вас есть...

tih66, 2020-11-23:
отправил

tih66, 2020-11-23:
создал пустой файл a_def_pass.h и закоментировал //#define public и проект собрался!

tih66, 2020-11-23:
[image]

vvip-68, 2020-11-23:
Да... Такой вариант я не проверял - у меня всегда этот файл есть...
Попробую понять что происходит...

vvip-68, 2020-11-23:
Хм... Видимо public - что-то где-то зарезервированное "#define public" раньше был в другом месте скетча и, видимо, не влиял на компиляцию. В 1.07 он переехал в начало скетча и все сломал...
Переопределил
#define public_project
#ifndef public_project
и всё стало собираться.

closed 2020-11-23 by vvip-68

vvip-68, 2020-11-23:
У вас в проекте в a_def_hard.h
#define DEVICE_ID   4         // 0 - тестовый стенд
Вот лучше использовать другую ID
У меня под этим номером - довольно проблемная гирлянда - во первых она большая 50x20 - как следствие - не хватает памяти на всё и скетч "валится". Во вторых - диоды проблемные - срывается синхронизация. 
Чтобы не валился и попытки исправить - для этого DEVICE_ID, например, отключены игры. Возможно что-то где-то еще...
Совершенно точно - при сборке с этим ID вместо игр будет или черный экран или замерший предыдущий эффект...

KAKTO, 2020-12-21:
Я мало понимаю в этом, но начиная с  версии 1.07 был изменён код в EspSoftwareSerial. Конкретно в файле \libraries\EspSoftwareSerial\src\SoftwareSerial.h изменены строки 130-137, в строке 132 открыта фигурная скобка.
Заменил этот файл (из версии 1.06), теперь компиляция проходит на ура и в "точке доступа" вроде всё работает.

130     int available() override;
131 #if defined(ESP8266)
132     int availableForWrite() override {
133 #else
134     int availableForWrite() {
135 #endif
136         if (!m_txValid) return 0;
137         return 1;

KAKTO, 2020-12-21:
Да, и никаких манипуляций с файлом a_def_pass.h не проводил (не создавал, отключен и MQTT отключен.