vvip-68/GyverPanelWiFi #33: // 418 - WEATHER_API_KEY

fastdron, 2020-11-23:
если раскомментить эту строку - не собирается проект, при этом всем в настройках указано USE_WEATHER == 1 и WEATHER_SYSTEM == 1 и прописан свой API, если же оставить строку "// 418 - WEATHER_API_KEY"  комментированой и указать те же настройки - все собирается хорошо и погода выводится успешно.
[image]

fastdron, 2020-11-23:
И подскажите как теперь в игры играть?)))) Пропустил инфу об этом. Скачал второе приложение на смарт, но не управляется ничего и пишет что матрица 0х0

vvip-68, 2020-11-24:
А для чего раскомментировать комментарий? В коде много где чего прокомментировано - любой раскомментируй -- собираться перестанет.

closed 2020-11-24 by vvip-68

247760, 2020-11-30:
Привет. Спасибо за проект. Сейчас я в процессе настройки. Почему в приложении осталась только погода от Яндекс?
Да я знаю что в прошивке всё есть. Просто интересно. Яндекс по коду даёт погоду сразу на всю страну, а не на то место где я (если я правильно понимаю), а вот второй сервис даёт на мой город. Написал тут, чтобы не плодить эти ветки.

vvip-68, 2020-11-30:
Где в приложении осталась только от Яндекс? Там три чекбокса - Не использовать / Яндекс / OpenWeatherMap
[image]

Яндекс дает погоду по тому месту, для которого указан код. Обычно самое мелкое деление - населенный пункт (город, деревня, село и т.д)
 Здесь указано как узнать код местности - 
https://github.com/vvip-68/GyverPanelWiFi/wiki/%D0%9D%D0%B0%D1%81%D1%82%D1%80%D0%BE%D0%B9%D0%BA%D0%B0-%D0%BF%D0%BE%D0%BB%D1%83%D1%87%D0%B5%D0%BD%D0%B8%D1%8F-%D0%B8%D0%BD%D1%84%D0%BE%D1%80%D0%BC%D0%B0%D1%86%D0%B8%D0%B8-%D0%BE-%D0%BF%D0%BE%D0%B3%D0%BE%D0%B4%D0%B5

247760, 2020-11-30:
Ну ёмаё, видимо я старую версию приложения установил, где только Яндекс был( Затупил.