vvip-68/GyverPanelWiFi #34: Отображение часов каждую минуту.

AxxiMator, 2020-11-24:
Здравствуйте. Можно ли в данный проэкт добавить возможность отображения часов+температура не поверх эффекта, а при смене минут, при этом эффект полностью отключать, выводить часы+температура на указанное время, далее эффект продолжать?
И очень бы хотелось иметь возможность отображения ПОГОДЫ между эффектами. Сейчас Я понял так , что реализован вывод картинки ПОГОДЫ только в ручном режиме.

vvip-68, 2020-11-24:
Можно забить отображение часов + температура в текст бегущей строки, поставить интервал отображения 60 секунд, в качестве фона отображения бегущей строки поставить "черный" - макрос {B#000000}
Про погоду - не понял.
Есть эффект "Погода", который так же как и остальные эффекты включается периодически.
Есть возможность в бегущей строке выводить текущую погоду и время.

AxxiMator, 2020-11-24:
Есть эффект "Погода", который так же как и остальные эффекты включается периодически.
а для погоды желательно выводить не периодически, а через заданный промежуток времени.

vvip-68, 2020-11-24:
Это вам скетч переписывать нужно будет - добавлять таймер на отображение часов, и по наступлению события таймера - выводить часы с температурой на матрицу. 
Например в custom.ino в строках 211-235 на if (clockReady) { } устанавливать / сбрасывать флаг необходимости отображения часов вместо движения, а в строке 278 вызывать процедуру drawClock() при только этом установленном флаге.

ТО же самое с погодой - добавляете таймер на необходимость отображения этого эффекта и когда он срабатывает - на заданный промежуток времени включать эффект Погода а не какой там сейчас по очереди отображается.  Это в строке 200. Если флаг отображения погоды НЕ установлен - вызывается processEffect(), а если установлен - прямо weatherRoutine()

AxxiMator, 2020-11-25:
Спасибо большое

closed 2020-11-25 by AxxiMator

247760, 2020-12-02:
Приветствую. Почти во всём разобрался кроме бегущей строки, вернее её написания. Но вопрос в другом, часы, они у меня поверх любого режима. При попытке выключить к примеру в режиме ОГОНЬ часы поверх его, меня выбрасывает в приложении к часам, и так с каждым режимом, где накосячил не понимаю.

vvip-68, 2020-12-02:
Ну, я ж не знаю что вы правили и как пытаетесь выключить. У меня такого не происходит. Я тоже не понимаю где и в чем вы накосячили.

247760, 2020-12-02:
Записал для примера с экрана, когда выбрасывает при снятии галки https://dropmefiles.com.ua/ru/L2SxaYcU
Сейчас проект сделаю с минимумов основных правок, чтобы исключить вмешательство. Переустановка приложения не помогла.

vvip-68, 2020-12-02:
Ну, это вы где-то чего-то наколбасили видимо. У меня точно такого нет.

247760, 2020-12-02:
Смею предположить, это просто версия, что дело в телефоне, а именно в версии Андроида, у меня Редми НОУТ 7 Андроид 10. Внёс минимум правок (размер, данные для сети и номер профиля). Никак галочка не снимается, а при нескольких попытках подряд, приложение само начинает прыгать во вкладку часы и к примеру Матрица, галки ставит и снимает, при этом в СОМ порт данные идут исправно, поэтому и думаю, что дело в телефоне, а другого дома нет на проверку.

vvip-68, 2020-12-02:
Вот сомневаюсь, что версия андроида может на что-то влиять.
А у вас вообще версия прошивки соответствует версии программы? 
Откуда и когда программу устанавливали? Она, знаете ли, тоже меняется время от времени вслед за прошивкой. Ну, или прошивка вслед за программой

247760, 2020-12-02:
@vvip-68 разбираюсь, вот Вы правки внесли свежие, перекачал, перенастроил и прошил. Видимо раз никто больше не писал, проблема у меня. Главное в целом работает, верняк разберусь. Ещё раз спасибо.

247760, 2020-12-03:
@vvip-68 так и не одолел я приложение, гирлянда работает как часики, даже один макрос сам замутил простой. Чистил телефон, переустанавливал, нельзя галочку снять никак. Сегодня увидел тему https://github.com/vvip-68/GyverPanelWiFi/issues/24 и там тоже Сяоми у человека, буду копать дальше.

vvip-68, 2020-12-04:
@247760 - там реально после переезда с Thunkable на AppInventor вкралась ошибка со снятием этой галочки.
Поправил

247760, 2020-12-04:
@vvip-68 Спасибо! завтра утром обновлюсь и всё проверю.