vvip-68/GyverPanelWiFi #36: error

matteos1, 2020-11-27:
compilation terminated.
exit status 1
a_def_pass.h: No such file or directory

vvip-68, 2020-11-28:
Read comments placed above this error line.

Last version of firmware requires this file only if you use USE_MGTT==1 or (USE_WEATHER==1 and WEATHER_SYSTEM==1)
Please download latest version from github.

If you turned on these features and do not want store passwords in private file 'a_def_pass.h', set MQTT params and WEATHER API KEY in sketch directly in lines mentoined in comment above this line
Otherwise you may create this file in sketch folder and put lines mentoined above in this file
I use this file and not publish it because I do not want show passwords for my account to all followers.

I do not understand sentensies about ESP32 and related link
Yes, you may use ESP32 in this project instead of ESP8266

closed 2020-11-28 by vvip-68

vvip-68, 2020-11-28:
ESP32 from AliExpress much more cheaper than from Amazon:
https://www.aliexpress.com/item/4000534485134.html
[image]