vvip-68/GyverPanelWiFi #37: Гирлянда 18*10 GyverPanelWiFi

Evgeni198, 2020-11-29:
Добрый вечер! Спасибо большое за проект, сам новичок в этом деле, но гирлянду, которую от alexgyver запустил и она меня очень радовала на балконе))) Решил перейти на новый уровень и запустить GyverPanelWiFi. Плата та-же Wemos D1 mini и возникло очень много неудач( Может кто на пирожках расскажет и поможет? Вопросы:
 1) При компиляции **какую плату указывать**? В пошаговой инструкции сказано плат Wemos D1 mini / Wemos D1 pro в настройках компиляции выбирайте "NodeMCU 1.0(ESP-12E Module)", но при такой настройке вообще ничего с гирляндой не происходит. В компиляции поставил Wemos D1 R1- работает, а спустя пару минут зависает, пробую ставить LOLIN(WEMOS) D1 R2 /mini -  пару минут работает и зависает все, такая-же ситуация и с LOLIN(WEMOS) D1 mini pro.
2) При включении гирлянды рандомно включаются режимы, как настроить их очередность и тут же вопрос как активизировать эффекты дэмо игры? На сколько хватило сил перезагружать плату - ни разу игр не видел...
 В гайде блютуз матрица - гирлянда на балконе все было понятно в части как добавлять удалять эффекты, здесь на ум ничего не приходит. Прошу прощения если пишу какой-то бред, но сам не програмист, а красоту хотелось-бы сделать на балконе. Всем спасибо за внимание.

vvip-68, 2020-11-29:
1. Все верно - указывать "NodeMCU 1.0(ESP-12E Module"; У меня при выборе Wemod D1 mini - случалось много разных глюков, при указании NodeMCU - работает стабильно. На какой пин прицеплен выход на ленту? В проекте и согласно данному выбору вывод идет на D2; Если указывать Wemos - при тех же настройках вывод коммутируется на D4;
Паяйте на D2 - должно заработать. Про зависания - смотрите что в мониторе порта в данный момент выводится - дальше в зависимости от - нужно думать
2. Если в приложении на телефоне убрать галку "Случайный режим" - будут включаться последовательно как прописано в прошивке. Чтобы при старте выбирался не случайный режим, а, скажем, с самого начала - в основном файле в строке 254 поставьте вместо ```setRandomMode2(); ``` -> ```setEffect(0)```
Если у вас выбран DEVICE_ID==4 - игр и не будет, они для этого ID отключены. Если другой - нужно разбираться.
3. Удаление эффектов:
в a_def_soft.h  в строках 38-82 прописаны идентификаторы эффектов (сквозная нумерация) и в строке 84 - общее количество; Удаляете ненужные эффекты - или переносите их номера за номер MAX_EFFECT, правите MAX_EFFECT на правильное количество эффектов. В работе останутся только эффекты от 0 до MAX_EFFECT-1
Если хотите вовсе удалить их из прошивке - ищите по всему коду где встречается указанный эффект (например MC_SNOW)  и удаляете всё что с ним связано
4. Добавление эффекта:
Добавляете его в список, дав ему новый уникальный идентификатор, например MC_NEW, правите MAX_EFFECT
В effects.ino прописываете функцию, которая формирует отображение эффекта. Потом берете один из существующих эффектов (например MC_SNOW) и ао всему тексту ищите где он встречается. По образу и подобию делаете примерно то же самое для вашего нового эффекта.

Evgeni198, 2020-11-29:
> 1. Все верно - указывать "NodeMCU 1.0 (модуль ESP-12E"; У меня при выборе Wemod D1 mini - случалось много разных глюков, при указании NodeMCU - работает стабильно. на D2; Если указывать Wemos - при тех же настройках вывод коммутируется на D4; 
>    Паяйте на D2 - должно заработать. Про зависания - смотрите что в мониторе порта в данный момент выводится - дальше в зависимости от - нужно думать
> 2. Если в приложении на телефоне убрать галку "Случайный режим" - будут включаться последовательно как прописано в прошивке. Чтобы при старте выбирался не случайный режим, а, скажем, с самого начала - в основном файле в строке 254 поставьте вместо `setRandomMode2(); `-> `setEffect(0)`
>    Если у вас выбран DEVICE_ID == 4 - игр и не будет, они для этого ID отключены. Если другой - нужно разбираться.
> 3. Удаление эффектов: 
>    в a_def_soft.h в строках 38-82 прописаны экологические эффекты (сквозная нумерация) и в строке 84 - общее количество; Удаляете ненужные эффекты - или переносите их номера за номер MAX_EFFECT, правите MAX_EFFECT на правильное количество эффектов. В работе останутся только эффекты от 0 до MAX_EFFECT-1, если 
>    хотите вовсе удалить их из прошивке - ищите по всему коду где встречается эффект (MC_SNOW).
> 4. Добавление эффекта: 
>    Добавляете в список, дав ему новый уникальный идентификатор, например MC_NEW, правите MAX_EFFECT 
>    В effects.ino прописываете функцию, которая формирует представление эффекта. Потом берете один из используемых эффектов (например, MC_SNOW) и всему тексту ищите где он встречается. По образу и подобию делаете примерно то же самое для вашего нового эффекта.

Перепаял, на D2 -гирлянда включилась, но спустя пару минут зависла(. по поводу: Про зависания - смотрите что в мониторе порта в данный момент выводится....Можно про это поподробнее пожалуйста

vvip-68, 2020-11-29:
Когда плата подключена к USB и вы ее прошили (загрузка завершена) / или до того как загружать прошивку - нажмите Ctrl-M или кнопку в правом верхнем углу Arduno IDE или через меню "Инструменты"

Каждый проект скетча имеет основной файл, к которому подключаются все отальные. С него начинается сборка и загрузка проекта, в нем расположены процедуры setup() и loop() - базовые процедуры проекта.

В данном случае это файл GyverPanelWiFi_v1.07.ino
А вообще - с такими вопросами вам лучше начать с изучения основ. Хотя бы - с уроков на сайте Гайвера - alexgyver.ru
Я могу дать консультации по проекту, но мне не хотелось бы заниматься разжевыванием основ...

closed 2020-11-29 by vvip-68

Evgeni198, 2020-11-29:
> Когда плата подключена к USB и вы ее прошили (загрузка завершена) / или до того как загружать прошивку - нажмите Ctrl-M или кнопку в правом верхнем углу Arduno IDE или через меню "Инструменты" [...]

В основном файле  GyverPanelWiFi_v1.07.ino первая вкладка строка 254 выглядит так:
  
250 if (!isDfPlayerOk) {
251      Serial.println(F("MP3 плеер недоступен."));
252    }
253  #endif
**254**
255 // Проверить соответствие позиции вывода часов размерам матрицы

То есть она пустая. Вы же писали что на это 254 строке необходимо заменить setRandomMode2(); -> setEffect(0). 

Извините заранее что напрягаю Вас.( Но я не программист, а сделать красоту хочется. Если не Вы то кто тогда поможет?

vvip-68, 2020-11-29:
Я даю рекомендации, используя актуальную версию прошивки. Какая у тебя - я не знаю.
Мог бы просто поиском поискать где эта aqyrwbz используется в этом файле, осмыслить логику и поправить.

Вообще, если у тебя не последняя версия (а видимо это так) - в прежних версиях была ошибка, связаннае с переполнением буфера оферлея отображения погоды, которая выражалось в периодических перезагрузках контроллера, если:
- Используется получение погоды
- Часы малого размера, когда погода отображается вместе с часами
- Часы двигаются по матрице.

Тогда либо:
- Отключить погоду
- Зафиксировать часы, чтоб не двигались
- Обновить проект до последней версии, где эта ошибка была исправлена

Evgeni198, 2020-11-29:
> Я даю рекомендации, используя актуальную версию прошивки. Какая у тебя - я не знаю.
> Мог бы просто поиском поискать где эта функция используется в этом файле, осмыслить логику и поправить.
> 
> Вообще, если у тебя не последняя версия (а видимо это так) - в прежних версиях была ошибка, связанная с переполнением буфера оверлея отображения погоды, которая выражалось в периодических перезагрузках контроллера, если:
> 
> * Используется получение погоды
> * Часы малого размера, когда погода отображается вместе с часами
> * Часы двигаются по матрице.
> 
> Тогда либо:
> 
> * Отключить погоду
> * Зафиксировать часы, чтоб не двигались
> * Обновить проект до последней версии, где эта ошибка была исправлена

Весь архив качал три дня назад в четверг, тоесть это уже старая версия?

vvip-68, 2020-11-29:
Сейчас идет доработка прошивки и программы - она часто меняется...
Но та ошибка вроде была исправлена раньше. Кажется 27.11
Так что в мониторе порта при зависаниях? И в чем выражаются зависания после пары минут работы?

Evgeni198, 2020-11-30:
> Сейчас идет доработка прошивки и программы - она часто меняется...
> Но та ошибка вроде была исправлена раньше. Кажется 27.11
> Так что в мониторе порта при зависаниях? И в чем выражаются зависания после пары минут работы?

Сегодня вечером продолжу мониторить, спасибо Вам за помощь!

247760, 2020-11-30:
Спасибо за проект. Собрал, предварительно прописал все нужные вводные. Самое не понятное для меня, хоть и расписан целый раздел, это про написание бегущего текста. Время есть, надеюсь разберусь.

Evgeni198, 2020-11-30:
> > Сейчас идет доработка прошивки и программы - она часто меняется...
> > Но та ошибка вроде была исправлена раньше. Кажется 27.11
> > Так что в мониторе порта при зависаниях? И в чем выражаются зависания после пары минут работы?
> 
> Сегодня вечером продолжу мониторить, спасибо Вам за помощь!

> Сейчас идет доработка прошивки и программы - она часто меняется...
> Но та ошибка вроде была исправлена раньше. Кажется 27.11
> Так что в мониторе порта при зависаниях? И в чем выражаются зависания после пары минут работы?

Зависания ввражаются в замерании эффектов.

vvip-68, 2020-11-30:
И причина вполне может быть видна в мониторе порта. Что там видно в этот момент, когда замирает?
Какой DEVICE_ID определен в вашем проекте?

Evgeni198, 2020-11-30:
> И причина вполне может быть видна в мониторе порта. Что там видно в этот момент, когда замирает?
> Какой DEVICE_ID определен в вашем проекте?

Ловлю момент

Evgeni198, 2020-11-30:
> 1. Все верно - указывать "NodeMCU 1.0(ESP-12E Module"; У меня при выборе Wemod D1 mini - случалось много разных глюков, при указании NodeMCU - работает стабильно. На какой пин прицеплен выход на [...]

По поводу второго пункта, исправил на setEffect(0):
[image]
Все так-же рандомно  появляются эффекты(

Evgeni198, 2020-11-30:
Если у вас выбран DEVICE_ID==4 - игр и не будет, они для этого ID отключены. Если другой - нужно разбираться.

Где наблюдать- DEVICE_ID==4 ???

vvip-68, 2020-11-30:
1. Ну, то есть базовых вещей ты так и не нашел времени поизучать...
2. Видишь в окошке монитора порта скорость 9600 бод в правом нижнем углу. А в настройках  в setup() скорость порта ставится 115200. Несоответствие скоростей и ты видишь то, что видишь.
3. Рандомные эффекты - ты в приложении снял галочку "Случайный порядок"?
4. А как ты вообще настраивал прошивку под себя, если не знаешь какой DEVICE_ID и что это вообще такое? Или ты скачал прошивку, ничего не читая и не настраивая нажал на кнопку "Загрузить"?  Так оно скорее всего работать не будет.
DEVICE_ID определяется в файле настроек конфигурации твоей матрицы a_def_hard.h где-то в самом начале файла...

Evgeni198, 2020-11-30:
DEVICE_ID определяется в файле настроек конфигурации твоей матрицы a_def_hard.h где-то в самом начале файла...
Да тупанул( 
#define DEVICE_ID   4         // 0 - тестовый стенд
                              // 1 - часы на кухне
                              // 2 - часы на столе
                              // 3 - панель на стене
                              // 4 - гирлянда на балконе
                              // 5 - часы у Даши
Стоит 4 гирлянда, тоесть в гирлянде на балконе нет игр?

Evgeni198, 2020-11-30:
> * Видишь в окошке монитора порта скорость 9600 бод в правом нижнем углу. А в настройках  в setup() скорость порта ставится 115200. Несоответствие скоростей и ты видишь то, что видишь.
> * Рандомные эффекты - ты в приложении снял галочку "Случайный порядок"?

2 и 3 пункт сори не понял. По поводу скорости может инструкцию проморгал, не в видел где ее менять. По поводу галочки "Случайный порядок"ничего нигде не трогал только поменял все как Вы говорили:
в основном файле в строке 254 поставьте вместо setRandomMode2(); -> setEffect(0)

vvip-68, 2020-11-30:
> Стоит 4 гирлянда, тоесть в гирлянде на балконе нет игр?
Да, и это крупными буквами рядом с определение DEVICE_ID в комментариях написано.
Почему-то 95% повторяющих проект ВООБЩЕ не читаю комментариев. Загрузил с гитхаба проект, ничего не правя нажал кнопку "Загрузить", потом сплошные вопросы - "почему у меня ничего не работает?"

[image]

Попробуй всё же внимательно читать то, что я тебе пишу?

> По поводу галочки "Случайный порядок"ничего нигде не трогал только поменял 
Ну так ты попробуй поменять. Авось сработает. Что то ты поменял - это чтобы при запуске прошивки она начиналась не со случайного эффекта, а всегда с самого первого (нулевого то есть)

Evgeni198, 2020-11-30:
> > Стоит 4 гирлянда, тоесть в гирлянде на балконе нет игр?
> 
> Да, и это крупными буквами рядом с определение DEVICE_ID в комментариях написано.
> Почему-то 95% повторяющих проект ВООБЩЕ не читаю комментариев. Загрузил с гитхаба проект, ничего не правя нажал кнопку "Загрузить", потом сплошные вопросы - "почему у меня ничего не работает?"
> 
> [image]
> 
> Попробуй всё же внимательно читать то, что я тебе пишу?
> 
> > По поводу галочки "Случайный порядок"ничего нигде не трогал только поменял
> 
> Ну так ты попробуй поменять. Авось сработает. То, что ты поменял - это чтобы при запуске прошивки она начиналась не со случайного эффекта, а всегда с самого первого (нулевого то есть)

Не ругайся, друг. У меня изначально стояло 4 я ее по инструкции назначал сам(.

Evgeni198, 2020-11-30:
[image]
Зависло на тетрисе(
и змейки не было

vvip-68, 2020-11-30:
> Не ругайся, друг. У меня изначально стояло 4 я ее по инструкции назначал сам(.

Покажи мне это место в инструкции? Возможно я где-то забыл это поправить. Поправлю, чтоб остальные не путались.

vvip-68, 2020-11-30:
Ну вот - при твоем DEVICE_ID=4 где нет татриса и др. игр - включается режим "Тетрис", который на экране ничего не рисует (потому что заглушка стоит), там остаются следы от от предыдущего эффекта. Эффект по умолчанию длится минуту - значит минуту на матрице ничего нового формироваться не будет. Далее, если эффекты идут последовательно - будут другие игры. Значит еще три минут не будет ничего происходить.

Evgeni198, 2020-11-30:
> > Не ругайся, друг. У меня изначально стояло 4 я ее по инструкции назначал сам(.
> 
> Покажи мне это место в инструкции? Возможно я где-то забыл это поправить. Поправлю, чтоб остальные не путались.

Изначально указывал цифру 4 сам.

Evgeni198, 2020-11-30:
> Ну вот - при твоем DEVICE_ID=4 где нет татриса и др. игр - включается режим "Тетрис", который на экране ничего не рисует (потому что заглушка стоит), там остаются следы от от предыдущего эффекта. Эффект по умолчанию длится минуту - значит минуту на матрице ничего нового формироваться не будет. Далее, если эффекты идут последовательно - будут другие игры, которых тоже нет. Значит еще три минут не будет ничего происходить.

так игры нельзя наблюдать на DEVICE_ID=4??? или все же можно запустить их?

vvip-68, 2020-11-30:
Без правки скрипта - нельзя. Если хочешь извращений - иди в файлы скетчей игр - g_maze.ino и и .п.
и убирай директивы условной компиляции - #if (DEVICE_ID == 4 && defined(ESP8266)) или ставь тут другой ID ил условие при котором это не выполняется. Если будешь удалять - не забудь в самом конце файла убрать #endif

Evgeni198, 2020-11-30:
> Без правки скрипта - нельзя. Если хочешь извращений - иди в файлы скетчей игр - g_maze.ino и и .п.
> и убирай директивы условной компиляции - #if (DEVICE_ID == 4 && defined(ESP8266)) или ставь тут [...]

Блин, я точно это не потяну((( а так хотел тетрис на окне со змейкой как у alexgyver((( Печаль

vvip-68, 2020-11-30:
Ну, если тяму не хватает DEVICE_ID на другой поправить, ничем помочь не могу... Страдай дальше...

Evgeni198, 2020-11-30:
> Ну, если тяму не хватает DEVICE_ID на другой поправить, ничем помочь не могу... Страдай дальше...

Тоесть ставим DEVICE_ID- панель на стене и все эффекты будут? и игры и время и бегущая строка???
И вопрос с рандомом отпишите пожалуйста поподробнее

vvip-68, 2020-11-30:
> Тоесть ставим DEVICE_ID-3 панель на стене и все эффекты будут? и игры и время и бегущая строка???

Это смотря где ты ставить будешь. Если в файлах g_maze.ino g_snake.ino g_tetris.ino, g_arkanoid.ino в первых строках где
```#if (DEVICE_ID == 4 && defined(ESP8266))``` поменяешь 4 на 3 - будет работать без дополнительных правок.
Ну или в a_def_hard.h меняешь DEVICE_ID 4 на DEVICE_ID 3 и переносишь свои настройки в блок который #IF DEVICE_ID == 3

С рандомом. Открываешь приложение на телефоне. Страничка основных настроек (там где ползунок яркости) Вверху экрана находишь чекбокс "Случайный порядок" и снимаешь с него отметку

Evgeni198, 2020-11-30:
> Это смотря где ты ставить будешь. Если в файлах g_maze.ino g_snake.ino g_tetris.ino, g_arkanoid.ino в первых строках где
> `#if (DEVICE_ID == 4 && defined(ESP8266))` поменяешь 4 на 3 - будет работать [...]

По поводу эффектов, Если ставим DEVICE_ID-3 панель потом  файлах g_maze.ino g_snake.ino g_tetris.ino, g_arkanoid.ino в первых строках где
> `#if (DEVICE_ID == 4 && defined(ESP8266))` поменяю с 4 на 3 - будут работать все теже 40 эффектов что и DEVICE_ID == 4 но и к тому-же  игры  g_maze.ino g_snake.ino g_tetris.ino, g_arkanoid.ino???

К телефону еще не подключился , это видимо еще не скоро)

vvip-68, 2020-11-30:
Ну ты мозги то хоть маленько включай?
Если ты и в играх и потом еще в определении поменяешь DEVICE_ID с 4 на 3 - у тебя снова равенство будет выполняться и вместо игр будет работать заглушка, а не игры.
Либо первый способ, либо второй. Не надо все мешать в кучу.

Нескромный вопрос - а сколько тебе лет вообще?

Evgeni198, 2020-11-30:
> Ну ты мозги то хоть маленько включай?
> Если ты и в играх и потом еще в определении поменяешь DEVICE_ID с 4 на 3 - у тебя снова равенство будет выполняться и вместо игр будет работать заглушка, а не [...]

35 мне не стыдно отвечать на данный вопрос

vvip-68, 2020-11-30:
Ууу как всё запущено... По вопросам и реакции я думал лет 12 максимум....
И - запустил? Работает?

Evgeni198, 2020-11-30:
Стыдно не узнавать что-то новое. Считаю, что игры из 90-х можно было и оставить в  DEVICE_ID с 4, а не убивать их(. Рабочий профиль не буду уточнять, но для понимания я очень далек от этой темы. И поверьте, те вопросы, которые я у Вас спрашиваю мне на самом деле непонятны.

Evgeni198, 2020-11-30:
> Ууу как всё запущено... По вопросам и реакции я думал лет 12 максимум....
> И - запустил? Работает?

Спасибо за комментарий...
Нет, это нам не по силу:
Это смотря где ты ставить будешь. Если в файлах g_maze.ino g_snake.ino g_tetris.ino, g_arkanoid.ino в первых строках где
#if (DEVICE_ID == 4 && defined(ESP8266)) поменяешь 4 на 3 - будет работать без дополнительных правок.
Ну или в a_def_hard.h меняешь DEVICE_ID 4 на DEVICE_ID 3 и переносишь свои настройки в блок который #IF DEVICE_ID == 3

Evgeni198, 2020-11-30:
Давайте сначала, у меня стоит в a_def_hard.h  DEVICE_ID 4 - гирлянда на балконе, мне необходимо что бы в ней были игры... Ваш комментарий :убирай директивы условной компиляции - #if (DEVICE_ID == 4 && defined(ESP8266)) или ставь тут **другой ID** ил условие при котором это не выполняется. Если будешь удалять - не забудь в самом конце файла убрать #endif
Какой другой, например?
Установили допусти какой-то другой... Игры заработали.... Все 40 эффектов остались?

vvip-68, 2020-11-30:
> Считаю, что игры из 90-х можно было и оставить в DEVICE_ID с 4, а не убивать их(. Рабочий профиль не буду уточнять, но для понимания я очень далек от этой темы.

Этот проект - МОЙ проект - в смысле, что все эти DEVICE_ID описывают реально имеющиеся у меня устройства. Кjulf я обновляю в них прошивку - я не меняю кучу параметров типа размера матрицы, углы подключения и всё такое. Я просто меняю в начале #definе DEVICE_ID X и загружаю прошивку в соответствующее устройство

Устройство с DEVICE_ID работает с 1000 диодов, что расходует много памяти и если включить туда игры - прошивка просто вылетает (перестает работать) по нехватки памяти. Именно для этого устройства (большой гирлянды) я отключил игры, т.к на ЭТОЙ гирлянде они не смотрятся и я их на ЭТОЙ гирлянде все равно не использую. Впрочем, это все описано в комментариях возле определения DEVICE_ID - почему не нужно использовать 4. Но кто ж комментарии то читает?

Раз уж ты по какой-то неизвестной причине выбрал DEVICE_ID 4 и настроил размеры и угол подключения в соответствующем блоке и хочешь игры (на твоих размерах 18х12 они будут работать без проблем), то вот инструкция.
В ArduinoIDE идешь во вкладки g_maze, g_snake, g_tetric и g_arkanoid и в строке номер 2 правишь
```#if (DEVICE_ID == 4 && defined(ESP8266))```
на
```
#if (DEVICE_ID == 44 && defined(ESP8266))
```

Все это, впрочем, я уже написал выше.

vvip-68, 2020-11-30:
>Установили допусти какой-то другой... Игры заработали.... Все 40 эффектов остались?

Да, для DEVICE_ID 4 отключены только игры, методом описанным выше - вместо игр для этого DEVICE_ID стоят заглушки

Evgeni198, 2020-11-30:
Запуск - успешный, осталось подключить телефон и убрать рандом) Спасибо!

vvip-68, 2020-11-30:
Поздравляю. Предчувствую, с телефоном будет масса вопросов.
Попробуй предварительно прочитать ВЕСЬ текст на страничке проекта, в том числе раздел "Вопросы и ответы"...

Evgeni198, 2020-11-30:
> Поздравляю. Предчувствую, с телефоном будет масса вопросов.
> Попробуй предварительно прочитать ВЕСЬ текст на страничке проекта, в том числе раздел "Вопросы и ответы"...

Спасибо, обязательно прочтём.

Evgeni198, 2020-12-01:
Скиньте пожалуйста ссылку на инструкцию по подключению телефона к гирлянде, спасибо!

vvip-68, 2020-12-01:
>Подскажите, другого способа как убрать рандомные эффекты с телефона существует?

Ну почему же не существует - надо в программе поправить, чтобы вызывался следующий эффект, а не рандомный. Правда, тогда будет несоответствие с программой - галочка есть, но она не работает.

> Или в любом случаи надо настраивать подключения к MQTT серверу ?

Если вам MQTT не нужно - не нужно ничего настраивать, просто не включать эту фичу. Если нужно - нужно настраивать.

Evgeni198, 2020-12-01:
> Ну почему же не существует - надо в программе поправить, чтобы вызывался следующий эффект, а не рандомный. Правда, тогда будет несоответствие с программой - галочка есть, но она не работает.
>
>
> Если [...]

Задача такая: убрать рандомные эффеты, которые повторяются через раз. Если есть возможность поправить в программе то как?

vvip-68, 2020-12-01:
Ну, попробуйте в основном файле в setup() в строке, скажем, 260 поставить
useRandomSequence = false;

Evgeni198, 2020-12-01:
> Ну, попробуйте в основном файле в setup() в строке, скажем, 260 поставить
> useRandomSequence = false;

260 строка:
[image]

Можете скрин прислать?

vvip-68, 2020-12-01:
Ну, можете и тут поставить. Работать будет. Я строки даю по актуальной версии прошивки.
[image]

Evgeni198, 2020-12-01:
Верно?:
[image]

Evgeni198, 2020-12-01:
> есть результат?

пойду загружать

Evgeni198, 2020-12-01:
отлично!

Evgeni198, 2020-12-01:
Решил убрать ненужные эффекты например "Часы" убрал его из строки 36. Также в строке #define MAX_EFFECT  уменьшил на один эффект:
[image]

Evgeni198, 2020-12-02:
Добрый вечер, я понимаю что Вас очень напрягаю ( Прошу помощи в последний раз. Если возникнут новые вопросы, то свой новый непокорённый проект сверну в трубочку и вернусь к  GyverMatrixBT и буду хотя-бы этим проектом любоваться(