vvip-68/GyverPanelWiFi #38: Форма панели

Michael-79, 2020-11-30:
Добрый день! Выражаю огромную благодарность автору!
Собираю гирлянду на окно. Возник один вопрос. Расстояние между светодиодами - 40 мм. Окно состоит из двух рам. Ширина импоста - 200 мм. Соответственно некоторые эффекты будут отображаться некорректно. Возможно ли как-то это предусмотреть?
[image]

vvip-68, 2020-11-30:
У меня нет идей.

closed 2020-11-30 by vvip-68

247760, 2020-11-30:
Очень уж плотно у Вас всё. В этом моменте с разрывом выход один, делать это расстояние (можно на пару см меньше) между светодиодами, иначе Ваша гирлянда будет выглядеть более менее цельно, с учётом разрыв, на расстоянии минимум 100 метров, а то более.