vvip-68/GyverPanelWiFi #39: английский при отображении времени до события?

vaicekas, 2020-11-30:
Здраствуйте, не нашёл есть ли вазможност переключить язык на английский при отображении времени до события? На пример вместо 30 дней - 30 days.

_Originally posted by @vaicekas in https://github.com/vvip-68/GyverPanelWiFi/issues/31#issuecomment-735973883_

vvip-68, 2020-11-30:
Нет, только исправив этот вывод в скетче. Файл utils.ino, функция WriteDays() и около нее - часы минуты, названия дней недели и т.п. Если нужны оба языка - нужно создавать, например, новый макрос вместо {R...} и при его обработке вызывать другие функции, например не WriteDays(), a WriteDaysEng()

vaicekas, 2020-11-30:
Спасибо. Нужен только один язык, подкоректирую себе в скече.
Спасибо за такой шыкарный функционал!

closed 2020-12-01 by vvip-68