vvip-68/GyverPanelWiFi #40: text and mqtt

matteos1, 2020-11-30:
the characters displayed appear to go from right to left and backwards. what needs to be corrected?
Mqtt can't get it to work

vvip-68, 2020-12-01:
If you want characters move right to left and back (bidirectional move) - in need write appropriate code in sketch. Current code allows move text unidirectional rigt to left. If you want move characters mirrored - left from right unidirectional and mirrored -- you need change matrix connection angle (CONNECTION_ANGLE) and direction from angle STRIP_DIRECTION

MQTT need appropriate setting to be working. Did you set appropriate server and  name and user password, server port and so on? Which server do you use? Are you able read article in project wiki related to mqtt settings (article in russian)?

matteos1, 2020-12-01:
Mosquito

vvip-68, 2020-12-01:
So, you need change settings in script regarding connection to your MQTT server.

matteos1, 2020-12-01:
I have tried several times. I wanted to make it communicate with home assistant but I can't.
i changed the parameters but nan goes.

vvip-68, 2020-12-01:
I know nothing about mosquito and its settings.
I know only thd sketch has server name defined in DEFAULT_MQTT_SERVER (line 293 of a_def_soft.h), server user in DEFAULT_MQTT_USER line 297, password in DEFAULT_MQTT_PASS line 301 and server port in DEFAULT_MQTT_PORT
All these settings are used on first time sketch start when params were not save into EEPROM). Next start these params rerieved from saved EEPROM and use to establish connection to MQTT server. You may change EEPROM_OK constant to another value after you changed one of definition described above. Or upload sketch with params "erase flash: all). After that saved ino EEPROM mqtt settings will be overwrite with defined default values. Also you may set MQTT connection parameters from amartphone application, page 'Connections'
Flag MQTT_USE_PREFIX lead to how topic request is constructed.
When MQTT_USE_PREFIX==1 topic is user_name/device_id/cmd
When MQTT_USE_PREFIX==0 topic is device_id/cmd
Here user_name is name of user defined in DEFAULT_MQTT_USER 
device id is - DEVICE_ID value
For my settings to first case topic is 'user_af7cd12a/WiFiPanel-0/cmd', in second case - 'WiFiPanel-0/cmd'
Free MQTT servers usually have limitation - how often it can process request from one device, usually 1 request per second. More often request are skipped. Delay before send next request is defined in MQTT_SEND_DELAY
Check if your settings match to description

matteos1, 2020-12-01:
> Non so niente di zanzara e delle sue impostazioni.
> So solo che lo schizzo ha il nome del server definito in DEFAULT_MQTT_SERVER (riga 293 di a_def_soft.h), utente del server in DEFAULT_MQTT_USER riga 297, password in DEFAULT_MQTT_PASS riga 301 e porta del server in DEFAULT_MQTT_PORT
> Tutte queste impostazioni sono utilizzate sui primi parametri non salvare in EEPROM). Quindi avviare questi parametri recuperati dalla EEPROM salvata e utilizzare per stabilire la connessione al server MQTT. È possibile modificare la costante EEPROM_OK con un altro valore dopo aver modificato una delle definizioni descritte sopra. Oppure carica lo schizzo con i parametri "cancella flash: tutto). Dopodiché le impostazioni mqtt salvate in EEPROM verranno sovrascritte con i valori predefiniti definiti. Inoltre è possibile impostare i parametri di connessione MQTT dall'applicazione dello smartphone, pagina 'Connessioni'
> Il flag MQTT_USE_PREFIX porta a come viene costruita la richiesta di argomento.
> Quando MQTT_USE_PREFIX == 1 argomento è nome_utente / id_dispositivo / cmd
> Quando MQTT_USE_PREFIX == 0 argomento è id_dispositivo / cmd
> Qui nome_utente è il nome dell'utente definito in DEFAULT_MQTT_USER
> ID dispositivo è - DEVICE_ID valore
> Per le mie impostazioni per il primo caso l'argomento è 'usera / WiFi- 0 / cmd ', nel secondo caso -' WiFiPanel-0 / cmd '
> I server MQTT gratuiti di solito hanno delle limitazioni: quanto spesso può elaborare la richiesta da un dispositivo, di solito 1 richiesta al secondo. Più spesso le richieste vengono ignorate. Il ritardo prima di inviare la richiesta successiva è definito in MQTT_SEND_DELAY in millisecondi / Se non è necessario un ritardo, impostalo su 0
> Controlla se le tue impostazioni corrispondono alla descrizione.

ok i try

closed 2020-12-02 by vvip-68