vvip-68/GyverPanelWiFi #41: Не подключается к WiFi

AndreyLyamkin, 2020-12-02:
Если в файле "a_def_soft.h", в параметре #define NETWORK_PASS прописан пароль в 20 знаков (может и это критично), то при включении пытается найти **_сеть с другим именем_**.

Например:
#define NETWORK_SSID "WiFi-2.4G"       // Имя WiFi сети
#define NETWORK_PASS "7e88sqT;brDbtbkh6lrb"          // Пароль для подключения к WiFi сети
Подключается к сети с именем "lrb" (т.е. к последним трем символам из пароля)

21:54:27.844 -> LED-Panel-WiFi v.1.07.2020.1130
21:54:27.844 -> 
21:54:27.844 -> Инициализация EEPROM...
21:54:27.878 -> Сохранение строк выполнено.
21:54:27.878 -> Ипользованы адреса EEPROM 800 - 837
21:54:27.878 -> Свободно ячеек 3258
21:54:27.911 -> Настройки сохранены в EEPROM
21:54:27.911 -> 
21:54:27.977 -> 
21:54:27.977 -> Подключение к lrb
21:54:28.010 -> ..................................................

247760, 2020-12-02:
Под каким адресов роутер видит "гирлянду"?

AndreyLyamkin, 2020-12-02:
> Под каким адресом роутер видит "гирлянду"?

Как роутер может её увидеть, если она пытается подключиться геройски к какой сети. 

Если в пароль прописать "12345668", перепрошить, то все работает (ну, с учетом того, что на роутере тоже пароль сменить на 12345678)

vvip-68, 2020-12-02:
Длина пароля - не более 16 символов

AndreyLyamkin, 2020-12-02:
А нельзя как-то увеличить длину пароля? Ну прям надо очень

vvip-68, 2020-12-02:
Можно. Увеличьте в a_def_soft.h размер массива pass в строке 279 до 21 - pass[21]
Далее идите в eeprom.ino - там карта распределения ячеек EEPROM где что хранится - строка 60 говорит о том, что для пароля отводятся ячейrи 104-119 и для чтения / записи функции getPass()/setPass.
Далее два варианта - поскольку ячейки с 120 и далее уже заняты под другое - вы сдвигаете ВСЕ ячейки на 4, переписывая ВСЕ функции чтения/записи других параметров указывая новые адреса, либо переносите место хранения пароля со 104 ячейки в какую-нибудь неиспользуемую область - например с 248 + 20 ячеек.
Потом находите функции getPass / setPass и изменяете в них адрес начальной ячейки пароля и длину строки, не забыв поправить во всех местах их вызовы (где они вызываются в скетче). После такого исправления нужно очистить EEPROM при прошивке или поменять EEPROM_OK на другое значение. Потом фойти в программе на телефоне подключившись к созданной точке доступа и в программе настроив параметры подключения к сети.

closed 2020-12-02 by vvip-68