vvip-68/GyverPanelWiFi #43: Спасибо и несколько мелких вопросов

JohnyDeath, 2020-12-04:
Огромное спасибо за замечательный проект и отличную документацию. На это не у каждого разработчика хватит сил. 

Появилось ряд вопросов от меня и моих детей:

1. Им очень нравилось "рисовать" на матрице/гирлянде. Сейчас все игры и рисование, судя по описанию, переехало в отдельное приложение WiFiPlayer. Но где его найти, я так и не понял. Оно даже толком и не гуглится.
2. Не было мысли интегрироваться с Телеграмом? Как мне кажется, для рядового гражданина телеграм-бот намного удобнее, чем MQTT. И API у телеграма достаточно простое и понятное. Как мне кажется, для управления панелью очень хорошо подойдет.

P.S. на сбер тоже закинул "спасиб" ;)

JohnyDeath, 2020-12-04:
На первый вопрос кажется нашел ответ.
Скорее всего имелось ввиду GyverPanelWiFi: https://github.com/vvip-68/GyverPanelWiFi/blob/master/Android/AppInventor/GyverPanelWiFi.apk
Правильно?

vvip-68, 2020-12-04:
@JohnyDeath - Спасибо за "спасибо"
Да, именно это приложение в проекте для рисования
У меня нет телеграмма, я про него ничего не знаю (да и знать не хочу, если честно).
Если и - то в очень долгия ящик. У меня пока других идей и проектов хватает...

JohnyDeath, 2020-12-04:
@vvip-68 по-моему приложение GyverPanelWiFi и GyverPlayerWiFi в текущем репозитории одинаковые.
По крайней мере в телефон встало две "панели". Сравнение внутренностей тоже показывает, что отличия в основном по подписям. Можете проверить? 

Про телеграм - понял. Возможно что-то и сам смогу дополнить в этом направлении, но совсем не силен в С++ и микроконтроллерах. В связи с этим вопрос: можете показать пример, как в прошивке можно выполнять Get/Post HTTP запросы?

vvip-68, 2020-12-04:
Ну, размеры разные, а что там внутри...
Мне проще "перегенерить" чем разбираться в отличиях...
Минут через 10...

vvip-68, 2020-12-04:
Попробуйте сейчас

vvip-68, 2020-12-04:
> В связи с этим вопрос: можете показать пример, как в прошивке можно выполнять Get/Post HTTP запросы?

Да. Arduino IDE меню Файл -> Примеры -> ESP8266 ->BasicHttpClient, PostHttpClient

JohnyDeath, 2020-12-04:
> Попробуйте сейчас

Теперь всё нормально (по крайней мере иконка и название приложений изменились). Еще раз спасибо.

closed 2020-12-04 by vvip-68