vvip-68/GyverPanelWiFi #44: Касательно шрифтов

north63, 2020-12-04:
Добрый день, благодарю за прекрасный проект. Это огромный прыжок от прошлогоднего. Мой вопрос в следующем, некорректно отоброжается буква ё, Ё. пытался решить самостоятельно, но понял что Ё собирается как то по особому, к ней пририсовываются точки. а по итогу на выходе нечто похожее на большую букву Б получается.
[image]

vvip-68, 2020-12-04:
Насколько мне известно - в прошивке нет поддержки буквы Ё и ё - они вообще отсутствуют в СТАНДАРТНОМ наборе кириллицы. Эти буквы расположены где-то в расширенном наборе символов, который прошивкой не обрабатывается и не поддерживается. (Там еще много каких символов не поддерживается). А то что написание как буквы Б  видимо после вычитания старшего байта UTF-8 как раз остается код, указывающий на букву Б.

north63, 2020-12-04:
{0x7c, 0x50, 0x50, 0x20, 0x7c}, //ы      154
  {0x7c, 0x50, 0x50, 0x20, 0x00}, //ь      155
  {0x28, 0x44, 0x54, 0x54, 0x38}, //э      156
  {0x7c, 0x10, 0x38, 0x44, 0x38}, //ю      157
  {0x08, 0x54, 0x34, 0x14, 0x7c}, //я 255  158 

  {0x06, 0x09, 0x06, 0x00, 0x00}, //°      159
 т.е. если сюда добавить прорисовку буквы ё, то все будет работать?
else if ((modif == 208 || modif == 209) && font >= 112 && font <= 159) {           // и пизд*ц для русских
    return pgm_read_byte(&(fontHEX[font - 17][row]));
и тут 159 заменить на 160.
Заранее извиняюсь за может быть глупые вопросы, просто уровень программирования на уровне школьного курса Паскаль. Кстати букву Й он нормально прорисовывает. Да и всю остальную кириллицу вроде тоже.

vvip-68, 2020-12-04:
Теоретически - да, практически - не всё так просто.
Нужно соответственно в drawLetter() выбирать код для букв Ёё - он скорее всего в последовательный диапазон не попадает. Трансляция кода буквы в индекс в массиве - в getFont(). Далее - все образы букв размещены в массиве 5 байт (ширина) - байт - 8 бит - т.е 8 точек.  Стандартная Е как раз по верхнему ряду знакоместа. Значит ее либо сдвигать вниз, чтобы разместить точки - и она будет выпадать из общего ровного ряда букв, либо обрабатывать специальным образом, чтобы точки рисовать НАД знакоместом Е. В общем - есть заморочки.

vvip-68, 2020-12-04:
> Кстати букву Й он нормально прорисовывает. Да и всю остальную кириллицу вроде тоже.

Ну, вот такой прикол разработчики кода ASCII (в плане кириллицы) сделали, что буква Й есть в стандартном ряду, а буквы Ё - нету. Она в группе расширенных символов и уже не ASCII, а UTF-8

north63, 2020-12-04:
> > Кстати букву Й он нормально прорисовывает. Да и всю остальную кириллицу вроде тоже.
> 
> Ну, вот такой прикол разработчики кода ASCII (в плане кириллицы) сделали, что буква Й есть в стандартном ряду, а буквы Ё - нету. Она в группе расширенных символов и уже не ASCII, а UTF-8

Тоже заметил что в UTF-8 уже добавили буквы ё, Ё. попробую все же приляпать её отдельным символом, спасибо.

vvip-68, 2020-12-04:
Я добавил буквы Ё и ё - обновите проект

vvip-68, 2020-12-04:
ш

closed 2020-12-04 by vvip-68