vvip-68/GyverPanelWiFi #46: Ох уж этот WIFI

AkmonLS, 2020-12-05:
Прошивка 
GyverPanelWiFi
плата Wemos (прошивал как положено, да и как не положено тоже пробовал)

Проблема: плата работает(в режиме точка доступа), но стоит подключиться к роутеру(а подключение проходит норм, пингуется с командной строки винды даже), приложение смартфона упорно не хочет подключаться.

vvip-68, 2020-12-05:
Что делаете для подключения со смартфона? Подробности по шагам.
Какой адрес получает плата в сети? Нет ли в сети другого устройства с таким же адресом?

vvip-68, 2020-12-05:
Из личного опыта. Пришел в сеть, Ноутбук с которого прошивал плату. Задал плате адрес 192.168.0.100. В логе вижу что загружается, подключается к роутеру, а из приложения к плате подключится - никак. Потом выяснилось, что ноут регистрируясь в сети тоже получил адрес 192.168.0.100

AkmonLS, 2020-12-05:
Залил прошивку, подключился к точке доступа Panel, зашел в приложение подсоединился к панели, прописал в настройках имя WiFi и пароль.
Монитор порта включен и он отображает что соединение с моим домашним вайфай установлено. адрес 192.168.1.135 (в настройках роутера по MAC адресу зафиксил его, чтобы он один и тот же адрес всегда подключал)
на этом адресе болльше никого нет.
Захожу опять в приложение(уже будучи подключенный к домашнему вифи) ввожу адрес 192.168.1.135 не подключается.
Однако через комп: ping -a 192.168.1.135 пакеты летают!!

AkmonLS, 2020-12-05:
13:25:09.203 -> LED-Panel-WiFi v.1.07.2020.1130
13:25:09.203 -> 
13:25:09.203 -> Загрузка строк выполнена.
13:25:09.249 -> Ипользованы адреса EEPROM 800 - 837
13:25:09.249 -> Свободно ячеек 3258
13:25:09.249 -> 
13:25:09.249 -> Инициализация SD-карты...
13:25:09.436 -> SD-карта недоступна
13:25:09.529 -> 
13:25:09.529 -> Подключение к Akmon
13:25:09.622 -> .............
13:25:13.676 -> WiFi подключен. IP адрес: 192.168.1.135
13:25:13.676 -> 
13:25:13.676 -> UDP-сервер на порту 2390
13:25:13.676 -> 
13:25:13.676 -> Подключаемся к MQTT-серверу 'srv1.clusterfly.ru:9123'; ClientID -> 'WiFiPanel-3' ....NTP-сервер ru.pool.ntp.org -> 91.207.136.50
13:25:13.767 -> Отправка NTP пакета на сервер ru.pool.ntp.org
13:25:13.814 -> UDP << ip='91.207.136.50:123'; ntp sync
13:25:13.814 -> UDP пакeт размером 48
13:25:13.814 -> Разбор пакета NTP
13:25:13.814 -> Секунд с 1970: 1607189113
13:25:13.814 -> Текущее время: 05.12.2020 17:25:13
13:25:13.814 -> --------------------
13:25:16.231 -> ...................

vvip-68, 2020-12-05:
А у вас реально MQTT используется и настроен?
Я вижу что он не может подключиться к MQTT.
При этом каждая попытка подключения блокирует прочую передачу по канала на время пока не отвалится по таймауту.
Вероятнее всего поэтому пакеты с вашего телефона даже не доходят до платки - все ресурсы канала занимаются попытками подключиться к MQTT серверу

AkmonLS, 2020-12-05:
LED-Panel-WiFi v.1.07.2020.1130

Загрузка строк выполнена.
Ипользованы адреса EEPROM 800 - 837
Свободно ячеек 3258

Подключение к Akmon
.............
WiFi подключен. IP адрес: 192.168.1.135

UDP-сервер на порту 2390
Режим: 'Тетрис'
NTP-сервер ru.pool.ntp.org -> 128.0.142.251
Отправка NTP пакета на сервер ru.pool.ntp.org
UDP << ip='128.0.142.251:123'; ntp sync
UDP пакeт размером 48
Разбор пакета NTP
Секунд с 1970: 1607189596
Текущее время: 05.12.2020 17:33:16
--------------------
Режим: 'The Matrix'
Режим: 'Радужные переливы'
Режим: 'Тетрис'
Режим: 'Арканоид'
Режим: 'Смена цвета'

vvip-68, 2020-12-05:
И?

AkmonLS, 2020-12-05:
оключил и все так же не подсоединяется, но по прежнему пингуется с компа

vvip-68, 2020-12-05:
Тогда у меня больше нет идей

AkmonLS, 2020-12-05:
[image]

AkmonLS, 2020-12-05:
с прошивкой Gyver**Matrix**WiFi такая же петрушка....

vvip-68, 2020-12-05:
Значит дело в телефоне. Есть возможность другой телефон попробовать?
Есть возможность с телефона попинговать?

AkmonLS, 2020-12-05:
Сейчас буду пробовать другой телефон и прошивки apk так как не пингуется(((

vvip-68, 2020-12-05:
А если не телефоне в этот момент отключить Мобильные данные?

AkmonLS, 2020-12-05:
они отключены, но роутер и компьютер пингуются с телефона

vvip-68, 2020-12-05:
И микроконтроллер тоже пингуется с телефона?

AkmonLS, 2020-12-05:
Нет микроконтроллер не пингуется с телефона почему то!

vvip-68, 2020-12-05:
Может у вас на телефоне какой-нибудь прокси или VPN стоит?

AkmonLS, 2020-12-05:
нет, ничего нет, остальные "абоненты" вай фая и локальной сети с телефона пингуются то! а именно микроконтролеер нет.
Зато микроконтролеер прекрасно виден из роутера и ПК.

vvip-68, 2020-12-05:
Я не знаю причину, но следствие очевидно - пока микроконтроллер не будет виден с телефона - программа не сможет к нему подключиться. Попробуйте сбросить микроконтроллер, чтобы он снова подключался в режиме точки доступа, потом зайдя в программу заново прописать параметры подключения задав другой адрес (не 135) в поле явным образом. И снова пробовать

vvip-68, 2020-12-05:
А у вас роутер в сети имеет какой адрес? 192.168.1.1 ? Или 192.168.1.100?

AkmonLS, 2020-12-05:
192.168.1.1

vvip-68, 2020-12-05:
И все же проверьте - если выключить микроконтроллер - он перестанет пинговаться с компа? Может все же второе устройство с таким адресом? Может телевизор какой...

AkmonLS, 2020-12-05:
Что очень кстати примечательно у меня дома есть вайфай разетка магазинная она работает, все норм(ESP_FC0CB4  так видит ее роутер), но с телефона она так же не пингуется а с компа пингуется

AkmonLS, 2020-12-05:
> И все же проверьте - если выключить микроконтроллер - он перестанет пинговаться с компа? Может все же второе устройство с таким адресом? Может телевизор какой...

да если отключить микроконтролеер он перестает пинговаться с компа

vvip-68, 2020-12-05:
С другим телефоном - ничего не поменялось?

vvip-68, 2020-12-05:
Телефон - не Xiaomi? Именно они по отзывам пользователей часто чудят почему-то...

AkmonLS, 2020-12-05:
Нет телефона 2 и оба самсунг, щас попробую со старого Хонора

vvip-68, 2020-12-05:
А розетка чем управляется? Программой на телефоне? И управляется?

AkmonLS, 2020-12-05:
> А розетка чем управляется? Программой на телефоне? И управляется?

Да прогой с телефона, и все работает, но к ней доступ есть и глобального интернета, я думаю что там обмен информации организован не на прямую через локальную сеть а через сервера. поэтому не показатель...

AkmonLS, 2020-12-05:
> Нет телефона 2 и оба самсунг, щас попробую со старого Хонора

ну увы с хонора тоже не пингуется ни розетка ни микроконтроллер, больше телефонов нет(((

vvip-68, 2020-12-05:
а на телефоне кроме ping возможность есть аналога виндовой команды tracert - посмотреть куда они пакеты отправляют?

vvip-68, 2020-12-05:
На роутере нет никакой настройки куда отправлять пакеты? Может он получив с телефона отправляет не в свою сеть, а куда-нибудь в интернет? Может на роутере чего настроить нужно?

vvip-68, 2020-12-05:
Другого роутера нет попробовать?

AkmonLS, 2020-12-05:
> а на телефоне кроме ping возможность есть аналога виндовой команды tracert - посмотреть куда они пакеты отправляют?

не нашел(

> На роутере нет никакой настройки куда отправлять пакеты? Может он получив с телефона отправляет не в свою сеть, а куда-нибудь в интернет? Может на роутере чего настроить нужно?

маршруты никакие на роутере не натроены, можно попробовать пробросить порты, например обращаясь к роутеру по порту UTP 2390  перенаправлять к микроконтроллеру.... надо изучать как это делать

> Другого роутера нет попробова

нужно поискать где то был старенький D-link

AkmonLS, 2020-12-05:
**Все проблема решена!!**
Итог в настройках роутера нужно отключить "изоляцию точки доступа" - она включена по умолчанию!!!
эта опция запрещает напрямую общаться только вайфай клиентам!Хотя она странно работает скрывало только ESP модули от других устройств, а телевизоры и прочаяя техника(телеки, смартфоны и т.д.) вифи попрежнему пинговала друг друга.

Надеюсь это кому-нибудь еще поможет

vvip-68, 2020-12-05:
Спасибо за информацию.

vvip-68, 2020-12-05:
А что за модель роутера то?

AkmonLS, 2020-12-05:
Asus RT-AC58U

closed 2020-12-05 by vvip-68