vvip-68/GyverPanelWiFi #47: Сохранение настроек между прошивками

JohnyDeath, 2020-12-06:
Добрый день. 
Не совсем понимаю принцип работы загрузки настроек между перепрошивками.
Судя по коду и описанию, чтобы все введенные настройки подгружались с устройства, необходимо не изменять значение в строке
`#define EEPROM_OK 0xA7`
файла **a_def_soft.h**
Внутри устройства было это значение (предыдущая прошивка).
Сегодня качнул новую прошивку, в ней тоже самое значение, но при заливке в устройство поимел сброс всех параметров : тайминги режимов, тексты строк, часовой пояс и т.п.
Что я делаю не так и как не терять настройки между прошивками?

P.S. при выключении/включении устройства все настройки сохраняются нормально. Устройство: NodeMCU CH340 V3

vvip-68, 2020-12-06:
Скорее всего в меню "Инструменты" в настройках заливки прошивки стоит Erase Flash: all.

JohnyDeath, 2020-12-06:
Точно! Подозревал, что всё дело в параметрах заливки.
Стояло вот так
[image]
Как я понимаю, надо установить "Only sketch"?

vvip-68, 2020-12-06:
Да, Only Sketch - тогда EEPROM не стирается, все настройки остаются пока не поменяется EEPROM_OK

closed 2020-12-06 by vvip-68