vvip-68/GyverPanelWiFi #49: Помогите, подскажите

badbear1987, 2020-12-07:
Здравствуйте. У меня на плате nodemcu v3. При установке времени автоматического переключения времени с 1-11секунд все норм. 12 и выше гирлянда висит на одном из выбранных режимах( не переключает) температура в помещении около +10°С. Как можно исправить? Память переполнена ? (Хотя при компиляции всего 50%) или холодно плате и кварц гонит? Матрица 24х10

vvip-68, 2020-12-07:
Не знаю. Вряд ли память и температура. У меня на балконе гирлянда 50x20 при температуре -15 прям сейчас исправно работает. Могу только предложить обновить версию прошивки до последней. У меня такого не наблюдается.

closed 2020-12-07 by vvip-68

jdc85, 2020-12-07:
Возможно ли получить ссылку на проект?

vvip-68, 2020-12-08:
А как вы сюда пишете, если у вас нет ссылки на проект?
 
https://github.com/vvip-68/GyverPanelWiFi

> 
> 
>Возможно ли получить ссылку на проект?
 
 
С уважением,
Вадим Пузанов

badbear1987, 2020-12-12:
Вместо тетриса может игру гонки?))) На что была на тетриса. Только не по
вертикали а по горизонтали.
Я слаб в этом плане программирования (((

vvip-68, 2020-12-12:
> Я слаб в этом плане программирования (((

Можно. Но если слаб в программировании - то заказы (за деньги) принимаются на сайте Гайвера в разделе "Ищу разработчика" - https://community.alexgyver.ru/forums/ischu-razrabotchika-ispolnitelja.21/

Там сформулируете техническое задание, договоритесь об оплате, вам сделают.