vvip-68/GyverPanelWiFi #50: Вопрос по надписям

tih66, 2020-12-07:
Добрый день!
Ни как ни пойму, как включить чтобы надписи были на черном фоне (эффекты в это время отключались)? подскажите пожалуйста!

vvip-68, 2020-12-07:
В строке надписи указать макрос {B#000000}

closed 2020-12-07 by vvip-68

247760, 2020-12-11:
@vvip-68 Добрый вечер. Возможно данный вопрос подымался, я не видел, возможно ли в эффекты добавить режим бегущей строки (текст) т.е. всё та же бегущая строка но без эффектов (и не по верх эффектов), как отдельная единица, в том же демо режиме будет идти чисто текст. Спрашиваю, так как иногда текст трудно читаем на некоторых эффектах в виду переливания\перемешивания цветов, не помогает даже определение цвета да строки. Спасибо.

vvip-68, 2020-12-11:
@247760 нет, строка отдельным эффектом не предусмотрена. Но вы можете самостоятельно дописать текст - там не сложно. А чем не устраивает просто бегущий текст, включающийся по заданному интервалу на черном (или другого цвета) фоне? Чтобы текст не был трудно читаем на фоне эффектов - у каждого эффекта в настройках есть ползунок "Контрастность".  Его можно вывести ползунком влево - яркость этого эффекта будет значительно яркости часов/текста бегущей строки и она не будет теряться на фоне этого эффекта. Поиграйтесь с настройками.

247760, 2020-12-11:
> А чем не устраивает просто бегущий текст, включающийся по заданному интервалу на черном (или другого цвета) фоне?
Хм, возможно это то что мне нужно. Т.е. я задаю своим скриптам чёрный фон и через заданный 240сек интервал я получаю что хочу? Тогда сразу вопрос, что будет с эффектом, когда этот текст  будет выведен поверх эффекта (ведь я не могу как то выбирать, что эти идут по верху текста, а другие отдельно через заданный интервал) он часом не "поглотит" его? И ещё вопрос, у меня забито уже 4 скрипта текста, как они выводятся в плане очерёдности (просто по очереди или рандомно как то)?

vvip-68, 2020-12-11:
1. При выводе текста на черном фоне текущий эффект перестает показываться, когда текст пробежит - его показ возобновится (ну, или сменится на другой очередной, если пришло время)
2. В тексте бегущей строки вместо фона нужного цвета можно указать эффект на фоне которого отображать текст. Тогда на момент бегущей строки включится этот эффект, потом вернется как в п.1.
3. Содержимое ячейки 0 в наборе текстов определяет как будут отображаться тексты из списка.
- Если текст в 0-й ячейке НЕ начинается с # - он отображается как обычный текст и далее все не пустые (и не отключенные) строки по порядку.
- Если текст в 0-1 ячейке содержит (начинается с) двух решеток '##' - не пустые и не отключенные строки отображаются в случайном порядке.
- Текст в 0-й строке может начинаться с одинарной решетки #, тогда за ней должны быть указаны индексы строк в той последовательности, в которой они должны быть отображены, например '#1357ABCDZF7575'
Ты вообще в разделе wiki на странице проекта на гитхабе читал описание параметров и макросов настройки бегущей строки?

247760, 2020-12-11:
Мой косяк за очерёдность, Вы это всё расписали в мануале, проглядел.
Крайний вопрос. Если я убираю все галки, где по верх эффектов можно проигрывать текст, то я получу строгое проигрывание текста через заданные 240 секунд без эффектов на чёрном (мною установленном) фоне?

vvip-68, 2020-12-11:
Нет. Текст будет бежать только поверх тех эффектов, где он разрешен галкой. Если галка снята - когда играется эффект текст не будет его прерывать. Текст побежит сразу как только эффект с запрещенным текстом сменится на эффект с разрешенным. Если у всех эффектов снять галки - текста не будет вообще.

Отсюда побочный эффект - если у всех сняты галки, и только у некоторых установлены - тест будет бежать только когда эти эффекты "проигрываются"

247760, 2020-12-11:
@vvip-68 Спасибо! Буду подстраиваться.