vvip-68/GyverPanelWiFi #51: Проблема ArduinoOTA.setHostname(host_name.c_str())

jdc85, 2020-12-08:
Добрый вечер. Скажите с чем может быть проблема призагрузке скетча версии 1.07  'class ArduinoOTAMdnsClass<WiFiServer, WiFiClient, WiFiUDP>' has no member named 'setHostname'
и выделяет эту строку ArduinoOTA.setHostname(host_name.c_str())

vvip-68, 2020-12-08:
1. Старая версия библиотек
2. Старая версия ядра
3. Старая версия Arduino IDE

vvip-68, 2020-12-08:
4. Выбрана не та плата в настройках сборки

jdc85, 2020-12-08:
Wemos d1 mini. Только не esp12 модуль, а esp8266. Какую в таком случае выбирать. Библиотеки все с раздачи сегодня скачал с Вашего архива. Очень благодарен Вам за помощь!

vvip-68, 2020-12-08:
Я выбираю NodeMCU v1.0 - с этим глюков не замечено, хотя реально платка Wemos d1 mini
Кроме библиотек еще причина может быть в версии IDE и в версии ядраESP8266

jdc85, 2020-12-08:
Скорей всего в. IDE давно не обновлял. Попробую, отпишусь. Может у кого-то возникнет вопрос. Почитает и расставит точки.

jdc85, 2020-12-08:
Всё получилось. Проблема была в версии IDE как Вы и говорили. Есть вопрос, немного не в этом посте-как согласовать 12 вольтовую ленту? Тоже адресная.  А RGB меняется. Здается в моем случае не совпадает, надо изменить вместо красного зелёный. Спасибо

vvip-68, 2020-12-08:
Так при инициализации же ленты указывается ее тип и порядок цветов

 FastLED.addLeds<WS2812, LED_PIN, **COLOR_ORDER**>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );

где

```#define COLOR_ORDER GRB          // порядок цветов на ленте. Если цвет отображается некорректно - меняйте. Начать можно с RGB```

closed 2020-12-08 by vvip-68

citypeople, 2020-12-13:
Подобная беда с NodeMCU v3, пытаюсь разобраться, но тщетно. Буду раз если есть какие-то соображения по этому поводу.

vvip-68, 2020-12-13:
Вероятно подобная беда решается подобными же методами, что описаны выше...

citypeople, 2020-12-13:
Да дело в том что уже удалял ide, чистил все libs которые могли быть на ПК, дело сдвинулось и теперь встал на
'class ArduinoOTAClass' has no member named 'getCommand'

vvip-68, 2020-12-13:
1. Какая версия библиотеки ArduinoOTA используется?
2. Какая плата выбрана в настройках компиляции в Arduino IDE?

vvip-68, 2020-12-13:
Точнее - вопрос неправильный - т.к. в данном случае ArduinoOTA - часть ядра esp8266 - какая версия ядра ESP8266 используется?

citypeople, 2020-12-13:
1.  1.0.5
2. NodeMCU  1.0 (ESP-12E Module)

citypeople, 2020-12-13:
Ядро пробовал 2.2.0 и 2.3.0

vvip-68, 2020-12-13:
Ну так видимо поэтому и не работает. Текущая версия ядра - 2.7.4

citypeople, 2020-12-13:
Спасибо пробую, вот даже мысли не было посмотреть последнюю версию, брал ссылку из интернетов на json

vvip-68, 2020-12-13:
Так менеджер устройств по этой ссылке json и берет актуальную версию м Arsuino IDE - менеджер плат

citypeople, 2020-12-13:
Уже компилится. И вроде все удачно, спасибо за отзывчивость и быстрые ответы.