vvip-68/GyverPanelWiFi #52: Ошибка кодировки в Мониторинге порта и вопрос о комбинации макросов в бегущей строке.

CrazyGhostRider, 2020-12-10:
Здравствуйте
1) Столкнулся с проблемой, что мониторинг порта выдает это:
12:55:31.795 -> Ƥ⸮⸮ĉ%⸮J3⸮⸮⸮!⸮⸮qz⸮⸮⸮⸮a⸮⸮Z,⸮+⸮⸮⸮⸮⸮L⸮⸮Pl0)⸮⸮#⸮⸮⸮⸮ ⸮y⸮1-⸮+)⸮;⸮⸮⸮ƌL%⸮⸮-E+⸮i⸮Ǵ9@⸮qk@-7=⸮⸮q⸮⸮0)⸮9⸮L⸮)p⸮1⸮⸮⸮⸮⸮!⸮⸮⸮0)⸮⸮⸮L!⸮!
12:57:26.850 -> @⸮⸮!⸮⸮L⸮)⸮1⸮!U⸮1⸮)9⸮⸮⸮0)⸮E⸮Le ⸮1⸮⸮9⸮⸮⸮L⸮)9Q⸮9⸮⸮L⸮)9⸮9⸮⸮1⸮⸮⸮9⸮⸮⸮0)⸮E⸮L⸮⸮⸮L⸮)9Q⸮9⸮⸮1⸮⸮⸮9Ѩ⸮@젔⸮d⸮⸮!⸮⸮0⸮⸮d⸮⸮!⸮U⸮1⸮⸮9⸮⸮)0⸮⸮d⸮⸮!⸮⸮0)b

Это появляется, и на чистой прошивке, и с настройками.
Плата: ESP8266 NodeMcu v3

2) Можно ли комбинировать макросы?
textLines[0]  = "До Нового Года осталось{R01.01.****#1}";
textLines[1]  = "-С новым годом {S01.01.****#07.01.****}"; (Будет ли она действовать только в эти дни?)

 "С РОЖДЕСТВОМ! {S06.01.****}{R07.01.****}";  (Только 6 января начнется отсчет?)

И Большое спасибо за Вашу работу!

vvip-68, 2020-12-10:
1. Скорость порта в окне монитора порта настройте. В правом нижнем углу. На 115200
2. Думаю что эти макросы будут работать вместе. Не вижу причин, чтобы не работали.
3. Макрос R не понимает звездочек в дате.

CrazyGhostRider, 2020-12-10:
Ошибся в написании. Имел ввиду "До рождества осталось {S06.01.* * * }{R07.01.2021}

И в чем принципиальное отличие между R и P?
Если {P01.01.****} то это будет точно также, как и {R01.01.2021}?
Но если {P01.01.****#1#60#60}, то независимо от работы за 60 секунд начать непрерывный отсчет и потом 60 секунд непрерывно показывать строку 1

closed 2020-12-10 by vvip-68