vvip-68/GyverPanelWiFi #54: Изменить размер шрифта

Diemon1, 2020-12-10:
Добрый день!
Моя гирлянда на балкон имеет размеры 30 высота *14 ширина (14 штук метровых полос по 30 светодиодов/метр).  Получается что по высоте расстояние между светодиодами ~ 3 сантиметра, но вот по ширине будет ~15 сантиметров между полосами.

Как я могу увеличить высоту шрифта, чтобы скомпенсировать количество светодиодов по вертикали? По моим подсчетам мне нужна высота букв/цифр в 22-25 диодов при той же ширине.
Наверное мну нужно ручками поменять шрифт в fonts.h? Есть какая-то программа, которая сможет визуализировать progmem array (0x3e, 0x51, 0x49, 0x45, 0x3e) ?

Спасибо!

vvip-68, 2020-12-10:
Хм... Гирлянда с неквадратными, но узкими ячейками - так себе решение... Буквы может и исправишь, но эффекты то будут так себе выглядеть... Да, надо переписать шрифты в fonts.h и/или процедуру, которая рисует эти буквы из массива шрифта - drawLetter(), getFont(). Если брать точку отрисовки по вертикали и рисовать вместо нее две (или три) - получатся буквы 16 или 24 точки высотой на том же самом массиве букв.
Ей-богу проще взять ножнички, почикать ленточки по одному диоду и проводками спаять с размером сетки 15x15 см
Мне кажется быстрее будет и по итогу выглядеть красивее...
Про программу - не знаю. Были, которыми в клетках шрифт рисуешь и она в этот progmrm конвертит, а вот чтобы наоборот... Ручками наверное только в биты разворачивать...

closed 2020-12-10 by Diemon1

fastdron, 2020-12-10:
Паял с шагом 10см с помощью шлейфа, потом приклеил получившиеся ленты на крышку с кабель-канала тонкого (10мм) вышли такие полоски жёсткие , спаял их между собой змейкой с параллельным питанием (шина внизу) и закрепил это все на окно с помощью присосок 2см с Али. Матрица вышла 25*11.[image]
[image]
[image]
[image]

vvip-68, 2020-12-10:
Ну, ты крут! А где результат и видео работы? Можно куда-нибудь на гугль-диск и расшарить ссылку...

fastdron, 2020-12-10:
Жду ещё ленту с Китая, хочу удвоить ширину матрицы ,будет 50*11( через перегородку гипсокартонную такое же окно, хочу оба закрыть гирляндой. Запилю видос с монтажом и сборкой и поделюсь.

vvip-68, 2020-12-10:
Надеюсь повезет. У меня была одна матрица 16x16? решил еще одну добавить. Заказал, пришла. Подключал.
Цвета оччень сильно отличаются....

fastdron, 2020-12-10:
[image]
[image]
Хочу ещё доработать немного, нижний шлейф питания спрятать в тонкий кабель канал и провода от шлейфа к столбикам окультурить, когда все будет готово - выложу) Спасибо Вам за Мега-крутой проект !👍👍👍

fastdron, 2020-12-10:
> Цвета оччень сильно отличаются....

Заказал у того-же продавана - посмотрим . Главное успеть к НГ)

mcvall, 2020-12-10:
> > Цвета оччень сильно отличаются....
> 
> Заказал у того-же продавана - посмотрим . Главное успеть к НГ)

Подскажите, как вы так ровно провода зачистили? У меня стриппером не получается так ровно, т.к. каждый отдельно зачищаю.

vvip-68, 2020-12-11:
Если это кабель три спаянных провода) - можно же вместе зачищать. А по глубине зачистке - у стриппера  же упор  есть на какую глубину вставлять зачищаемый провод.

mcvall, 2020-12-11:
> Если это кабель три спаянных провода) - можно же вместе зачищать. А по глубине зачистке - у стриппера же упор есть на какую глубину вставлять зачищаемый провод.

Может у меня стриппер такой, если три сразу вставлять, то рвёт. По одному норм, но я честно уже замахался на 500 светодиодов зачищать и паять)

vvip-68, 2020-12-11:
До Нового года осталось 20 дней. Время еще есть.
Терпение и труд - всё перетрут.

fastdron, 2020-12-11:
> Может у меня стриппер такой,

Я вообще зачищал вручную обычными тонкими бокорезами \кусачками ))) приловчился ровно снимать изоляцию сразу с 3 жил шлейфа. Изначально нарезал шлейф кусочками по 7см и потом каждый кусок с двух сторон по 3-4 мм снимал изоляцию кусачками. Провода не лудил, лудил только контактные площадки на светодиодах, не жалея припоя,жирненько ,потом легче провод паять.

mcvall, 2020-12-11:
> Если это кабель три спаянных провода) - можно же вместе зачищать. А по глубине зачистке - у стриппера же упор есть на какую глубину вставлять зачищаемый провод.

Может у меня стриппер такой, если три сразу вставлять, то рвёт. По одному норм, но я честно уже замахался на 500 светодиодов зачищать и паять)

> Я вообще зачищал вручную обычными тонкими бокорезами \кусачками ))) приловчился ровно снимать изоляцию сразу с 3 жил шлейфа. Изначально нарезал шлейф кусочками по 7см и потом каждый кусок с двух [...]

А кусачки какие? У меня шаг между диодами 6.5см, паяю сразу, предварительно на провод намазав флюс-пасту, при хорошей температуре всё сразу лудится и паяется, при том, что контакты на ленте покрыты лаком. Припой sn97ag3 (олово/серебро) без свинца.

fastdron, 2020-12-12:
> А кусачки какие?

Тонкие в радиомагазине покупал , типо таких 
https://sxema.com.ua/p725643681-kusachki-bokorezy-tonkimi.html?gclid=Cj0KCQiAzsz-BRCCARIsANotFgNqCS7__wvYF1NdYTwhxdtnxAD-nyw7iKnKIt1yn_GfDIvKBVT28WYaAmnVEALw_wcB
[image]

mcvall, 2020-12-12:
> А ты тоже на шлейф паяешь? У тебя матрица какого размера будет, на балкон (остекление от пола до потолка?)

Матрица 26х19 на окно(2х1,3м) Провод такой https://dilux.com.ua/p550772977-dilux-kabel-ploskij.html

mcvall, 2020-12-12:
> Окно 2 м шириной? У вас от центра до центра светодиода будет примерно 7 см? А импост на окне не мешает? У меня например шаг между центрами светодиодов 10см ,а ширина импоста (на створке окна) 14-15 см примерно, соответственно в этих местах шаг между столбцами на 50% больше чем на всей матрице (на матрице 10см ,на импосте 15) ,тоже хотел как можно с более плотным шагом паять матрицу ,планировал 8см, но этот нюанс остановил ,и решил паять 10см чтоб не так видна разница была на импосте окна.
> Хотя в итоге эти 50% разницы в ширине между светодиодами не заметны вообще с улицы на любых эффектах, думал на тексте будет искажать -но тоже нет, не видно вообще разницы ,возможно из за того что квартира на 10м этаже и с такого расстояния изображение не искажается

У меня тоже 14см импост, окно 2 метра, между светодиодами 6.5см расстояние. Я ещё не цеплял матрицу, паяю, немного осталось