vvip-68/GyverPanelWiFi #55: Бегущая строка низ головой

19Andrey, 2020-12-10:
Добрый день, у меня текст бегущей строки идёт низ головой, подскажите как поправить

vvip-68, 2020-12-11:
Указать правильно угол подключения матрицы и направление из угла

closed 2020-12-11 by vvip-68

19Andrey, 2020-12-11:
В прошивке у меня часы показывают нормально, в матрицу выводит  нормально, а вот когда текст в программе пишу идёт низ головой, подскажите где менять угол матрицы в программе на телефоне что бы текст шёл нормально

vvip-68, 2020-12-11:
У меня нет ответа. Я не знаю как добиться, чтобы часы шли нормально, а текст вниз головой. По-моему в прошивке вообще это не предусмотрено. Возьмите с хранилища прошивку, измените только размер матрицы и угол подключения. Залейте ее в микроконтроллер и убедитесь что часы и текст отображаются одинаково. Потом вносите свои изменения дальше и смотрите какое из них приводит к эффекту как у вас. Программа на телефоне вообще не имеет никаких настроек ориентации текста.