vvip-68/GyverPanelWiFi #60: Эффекты не на всю доступную матрицу

tarasifua, 2020-12-12:
Здравствуйте! Заметил такую особенность, некоторые эффекты, такие как вышиванка, тетрис отображаются не на всю доступную площадь, а только в квадрате по центру гирлянды. Размер гирлянды у меня 12х20. Это так и должно быть, или баг? Я пользовался Вашей прошивкой в прошлом году, там все нормально было, все эффекты работали "на весь экран". У меня просто окно с двух половинок, и получается что эти эффекты по центру сильно перекрываются рамой. Если есть возможность, поправьте, пожалуйста их, чтобы были на всю площадь.

И еще, огромное спасибо за поддержку MQTT. Теперь есть возможность интегрировать ее в системы умного дома. Есть только одно пожелание. Если можете, было бы намного проще с ней работать, если бы она отдавала все свои параметры в отдельном топике и постоянно, как это делают практически все MQTT-устройства. К примеру, чтобы был топик states, в котором матрица публиковала бы все свои параметры сразу после их изменения, без отдельного запроса, чтобы в любой момент можно было их считать. Общие параметры можно складывать прямо в этом топике, а для эффектов, будильника, других функций, где несколько параметров, можно сделать субтопики (states/effect_id1, states/effect_id2, states/alarm, states/running_textN и т. д. Это очень упростит работу с матрицей, а то сейчас приходится костыли городить. Еще и баг один заметил, когда начинаешь отправлять какие-то команды в устройство, отключается автоматический режим, надо его после включать. А чтобы в систему умного дома попадали актуальные параметры, надо постоянно их запрашивать с определенным интервалом, и из за этого автосмена отключается. Плюс парсить их с топика еще нужно, если запрашивать пакетно. Поэтому отдельный топик решил бы эти проблемы.

Заранее извиняюсь, если мои просьбы будут неуместными, я бы сам с радость помог это сделать, но, к сожалению я далек от программирования, чтобы буквы добавить, я четыре часа пытался понять как это сделать. Так что если вы сможете, буду очень признателен

vvip-68, 2020-12-13:
Про украинские буквы - а тем разве еще нет буквы типа Г, но с хвостиком сверху?

> Заметил такую особенность, некоторые эффекты, такие как вышиванка, тетрис отображаются не на всю доступную площадь, а только в квадрате по центру гирлянды. Размер гирлянды у меня 12х20. Это так и [...]

Ну, положим, "вышиванки" в прошлом году вообще не было. Сам алгоритм эффекта - рисование по диагоналям квадрата.
Поэтому и поле используется квадратное. Чтобы было симметрично - центрируется по ширине матрицы. Вообще эффект глючный - порой рисует нормально, порой просто точки четные/нечетные мерцают. Почему так - я так и не отловил. Возможно вообще выкину его. Не нравится он мне.

Тетрис - у меня широкая панель 50x20 (да и 48х16 во второй тоже) - во всю ширину тетрис абсолютно не смотрится, да и играть в него становится невозможным. Поэтому сделано - ширина стакана равна его высоте. Но, хорошо, я добавил в настройки вверху файла тетриса параметр FULL_SCREEN - если он 1 - тетрис будет во всю ширину экрана

Про MQTT - ничего не понял. Меня человек попросил написать для управления в умном доме. Я написал. У него претензий не было. Сам я этим не пользуюсь. Напишите точное ТЗ - будет время - сделаю. А так - ничего не понятно. 

Что такое "отдавать постоянно"?  Забивать канал? На бесплатных серверах интервал отправки - не чаще 1 раза в секунду, есть ограничения на количество отправляемых сообщений в сутки. Это тоже нужно учитывать. Сейчас вот как у меня в андроидном приложении - управление разбито на логические страницы. Когда я что-то изменяю на этой странице - присылается все состояние текущей страницы. Там ничего не меняется пока нет воздействия. Зачем слать постоянно? К тому же будет следующий эффект: вы что-то редактируете (если это приложение) - изменили данные в полях. А тут раз - статус самопроизвольно в очередной раз пришел - все что вы вводили сотрется и заменится пришедшим.

Остальные оповещения о том что происходит без участия со стороны пользователя - присылаются (побежал текст, получена погода, сменился эффект и т.п.)

К тому же тот кто просил, для кого я это делал - скорее всего уже интегрировал то что есть в свою систему. Если что-то менять - у него все сломается, ему придется переписывать все заново у себя.

vvip-68, 2020-12-13:
> Еще и баг один заметил, когда начинаешь отправлять какие-то команды в устройство, отключается автоматический режим, надо его после включать.

Расскажите что вы при этом делаете?
У меня подобного не происходит - я отправляю запросы из публичных панелей типа http://client.mqtt.4api.ru/
Да и по коду выключение авторежима происходит только в двух случаях - явно командой включить ручной режим '$16 0;' или при включении указанного пользователем эффекта '$8 0 N;'

Может быть у меня взгляд замылен и я действую "по правилам" как привычно, а вы что-то делаете по другому?

tarasifua, 2020-12-13:
Добрый день!

> Про украинские буквы - а тем разве еще нет буквы типа Г, но с хвостиком сверху?
Есть, "Ґ", в принципе можно добавить, но по факту она почти никогда не используется, все пишут через "Г".

> "вышиванки" в прошлом году вообще не было
Да, не было, но эффект прикольный) А за тетрис спасибо за параметр) Кстати, в эффекте "Пейнтбол" такая же проблема. Можно там так же сделать, или там алгоритм тоже под квадратное поле сделан?

> Что такое "отдавать постоянно"? 
Извиняюсь, не так выразился. Не слать параметры постоянно, а один раз при включении матрицы, а дальше только при их изменении. Включили матрицу - она загрузилась и отправила все свои настройки на сервер в отдельный топик (например states). Пользователь сменил какой-то параметр - он сразу же поменялся в топике. Таким образом все параметры будут всегда доступны для получения, без необходимости отправлять команду с запросом настроек.
То есть слать параметры не постоянно, а только после включения, если было полное обесточивание, и после изменения настроек слать измененный параметр. Хотя, учитывая количество параметров, при первоначальной загрузке все равно будут проблемы с бесплатными серверами.
В таком случае можно сделать чтобы параметры слались одной строкой, как сейчас.

> Расскажите что вы при этом делаете?
Вот я попытался добавить матрицу в систему УД Home Assistant. Пока только добавил Вкл/Выкл, яркость и лимит по току. Чтобы в нем всегда были актуальные параметры матрицы, система каждые 30 секунд отправляет команду "$6 7|DM|BR|PW|; для получения параметров, и, если они изменились, они меняются и в Home Assistant. Так вот из за постоянной отправки этой команды матрица не переключает режимы, а стоит на одном и все. Если я это отключаю - работает без проблем.

vvip-68, 2020-12-13:
> Кстати, в эффекте "Пейнтбол" такая же проблема. Можно там так же сделать, или там алгоритм тоже под квадратное поле сделан?

Нет. Там в настройках эффекта есть галочка "Сегменты" - если ее снять - шарик бегает по всему полю. Тоже было сделано для широких матриц - одиноко бегающий шарик по всему большому полю - не впечатлительно. Поэтому было сделано разделение на квадратные сегменты - так живее смотрится. Но - выключите в настройках эффекта - будет одинокий шарик (несколько шариков)  по всему полю

MQTT
Мне почему-то думалось, что включение матрицы происходит в разы реже чем, скажем, подключение управляющего клиента, который просто при своем включении просто запрашивает необходимые ему параметры. Я не очень разбираюсь в MQTT, кажется там есть параметр retain - вы его имели в виду, когда при включении матрицы слать параметры? Помню давно у меня были с ним проблемы - мне всё время приходили с ним устаревшие неактуальные данные...

Но если слать при загрузке полный набор параметров просто в никуда в MQTT - авось кто-нибудь услышит - то это не проблема, могу добавить такой вариант. Ну, и видимо, - одной строкой - из за ограничений бесплатных серверов.
А в чем смысл слать параметр при его изменении? MQTT-клиент же знает ЧТО он изменил. Разве что, если несколько MQTT-клиентов и один меняет, а другие знают... Надо это обдумать на досуге...

> каждые 30 секунд отправляет команду "$6 7|DM|BR|PW|; для получения параметров, и, если они изменились, они меняются и в Home Assistant. Так вот из за постоянной отправки этой команды матрица не переключает режимы, а стоит на одном и все.

Странно... Посмотрю этот момент...
Здесь на гитхабе (да и в списке Stars  этого проекта есть пользователь shostko
Это он просил меня сделать управления для умного дома. Вроде бы он тоже через Google Assiatant (хотя, может и через Алису - не помню). Вы бы с ним пообсуждали, а мотом мне выкатили что-то типа ТЗ, чтобы я знал что реально нужно сделать... А так это будут опять мои фантазии.

tarasifua, 2020-12-13:
> Там в настройках эффекта есть галочка "Сегменты" - если ее снять - шарик бегает по всему полю.

Действительно есть, не заметил.

> слать при загрузке полный набор параметров просто в никуда в MQTT - авось кто-нибудь услышит - то это не проблема, могу добавить такой вариант.

> А в чем смысл слать параметр при его изменении? 

Смысл как раз в том, чтобы, к примеру, тот же Home Assistant, всегда получал актуальные данные о состоянии матрицы без костыля в виде отправки команды с запросом параметров с некоторым интервалом. Параметры же можно сменить и с приложения, а чтобы они изменились и в Home Assistant, надо слать запросы. 
Все похожие системы умеют сами отслеживать состояние заданного топика и получать актуальные данные сразу же после их изменения.

> Вы бы с ним пообсуждали, а мотом мне выкатили что-то типа ТЗ, чтобы я знал что реально нужно сделать... А так это будут опять мои фантазии.

В принципе Вы можете ничего старого не ломать, просто сделать еще один топик, где будут храниться все актуальные настройки. Этого будет достаточно.

vvip-68, 2020-12-13:
> В принципе Вы можете ничего старого не ломать, просто сделать еще один топик, где будут храниться все актуальные настройки. Этого будет достаточно.

Это какое-то слабое ТЗ. Я не понимаю ЧТО нужно сделать.

tarasifua, 2020-12-13:
У Вас сейчас есть 5 топиков: "cmd", "dta", "nfo", "err" и "ack". Просто добавьте еще один топик, к примеру "settings", куда после загрузки панели или после изменения настроек будут отправляться ВСЕ настройки панели в таком же виде как и сейчас:
`$18 CE:0|CC:0|CO:0|CK:0|NC:0|SC:105|C1:255,255,255|DC:1|DD:3|DI:240|NP:1|NT:60|NZ:2|NS:[pool.ntp.org]|DW:1|OF:0;`
В таком виде, только все возможные параметры и всегда актуальные.
Прошу прощения если плохо объяснил

vvip-68, 2020-12-13:
Ну, это не проблема. Чуть позже сделаю.

tarasifua, 2020-12-13:
Отлично, спасибо большое!

vvip-68, 2020-12-13:
Еще просьба - когда всё настроите и будет работать - не могли бы вы написать инструкцию по интеграции с Home Assistant (или другими системами, если знаете) - чтоб я потом это в wiki положил - и люди бы знали как и что нужно делать для.
Для меня вот это - вообще темный лес.

tarasifua, 2020-12-13:
Обязательно, если разберусь со всеми командами настроек. Пока есть некоторые непонятки. К примеру:
```
15 - скорость таймеров
  - $15 D N; D - скорость 0..255; 
              N - таймер, где N
                  0 - таймер эффектов
```
Таймер эффектов - это имеется ввиду длительность эффекта в авторежиме?
Или это:
`8 - управление эффектами
  - $8 0 N; включить эффект N
  - $8 1 N D;   - D -> параметр #1 для эффекта N;
  - $8 2 N X;   - вкл/выкл использовать в демо-режиме; N - номер эффекта, X=0 - не использовать X=1 - использовать 
  - $8 3 N D;   - D -> параметр #2 для эффекта N;
  - $8 4 N X;   - вкл/выкл оверлей текста поверх эффекта; N - номер эффекта, X=0 - выкл X=1 - вкл 
  - $8 5 N X;   - вкл/выкл оверлей часов поверх эффекта; N - номер эффекта, X=0 - выкл X=1 - вкл 
  - $8 6 N D;   - D -> контрастность эффекта N;`
Что значит "параметр #1" и "параметр #2"?

vvip-68, 2020-12-13:
1. Нет. Это 255-xx - цикл задержки следующей итерации отрисовки эффекта - фактически - скорость эффекта
2. Изначально у эффекта был только один параметр (#1) - для каждого эффекта имел свое значение? например цвет огна в режиме "Огонь", интенсивность снега, зеленых полосок в Matrix и т.п.
По мере развития появились эффекты, которым одного параметра стало мало. Для них появился дополнительный параметр #2 - например галочка "сегменты"  в циклоне, пейнтболе, водовороте. Вариант отображения радуги в Радуге, номер ролика в эффектах SD-карты и т.п.

vvip-68, 2020-12-13:
Кстати, команда эта про скорость $15 - применяется к текущему активному эффекту.
Когда-то в самом начале разработки она была общая для всех эффектов и имела ползунок в приложении ниже ползунка яркости. Потом выяснилось что каждому эффекту неплохо бы иметь собственнуб настройку скорости. И ползунок переехал на страничку настройки эффектов, а команда осталась. И применяется теперь к параметрам активного эффекта.

tarasifua, 2020-12-13:
Вот как все сложно.. А можно как-то в исходниках приложения посмотреть какие команды отправляются при изменении настроек. Я *.aia-файл в инвентор загрузил, но что-то не могу найти где логика прописана

vvip-68, 2020-12-13:
Логика в блоках. Кнопка "Блоки" в верхнем правом углу рядом с кнопкой "Дизайн". Который кстати есть в виде отдельной картинки.

vvip-68, 2020-12-13:
Кстати, в мониторе порта же видно какие команды приходят из приложения при нажатии той или иной кнопки / галочки...

vvip-68, 2020-12-13:
Я исправил ситуацию когда после команды $6 7 переставали меняться режимы.
Отправку параметров при старте и при изменении сделаю чуть позже.

tarasifua, 2020-12-13:
Мда, там в логике еще сложнее разобраться. 
А мониторе порта видно только отправку параметров в приложение, но не из него. Только пишет "UDP пакeт размером 6" и все. 

> Я исправил ситуацию когда после команды $6 7 переставали меняться режимы.

О, спасибо!

vvip-68, 2020-12-13:
> А мониторе порта видно только отправку параметров в приложение, но не из него. Только пишет "UDP пакeт размером 6" и все.

Страннно... Вот скриншот монитора порта - отправлена команда "Авторежим", затем "Ручной режим", затем "Выкл случайный режим". Каждый раз после приход команды в, наприемр
'''18:34:07.395 -> UDP << ip='192.168.0.107:2390'; cmd='$16 1;''''
Отправляется ответ из - в данном случае состояние элементов на странице откуда пришла команда
''18:34:09.038 -> UDP 192.168.0.107:2390 >> $18 W:48|H:16|DM:1|PD:15|IT:0|AL:0|RM:0|PW:15000|BR:32|WU:1|WT:15|WR:62|WS:1502026|WC:1|WN:0|WZ:1;'''

[image]

vvip-68, 2020-12-13:
А отправку состояния в топик MQTT как делать - строкой для парсинга как сейчас в ответ на $6 7 или строкой в формате json?

tarasifua, 2020-12-13:
Странно, у меня почему-то так:
[image]

> отправку состояния в топик MQTT как делать

В принципе без разницы, делайте как Вам проще. Я уже для этого формата парсер сделал, главное чтобы последовательность параметров в топике не менялась

vvip-68, 2020-12-13:
>  Я уже для этого формата парсер сделал, главное чтобы последовательность параметров в топике не менялась

Это как? Там же параметры в любой последовательности теоретически могут приходить. У меня уж точно нет никакого определенного параметра...

tarasifua, 2020-12-13:
Тогда лучше сделайте в json-формате, там по названию ключа можно будет любой параметр получить.
А по поводу монитора порта. Что у меня с ним не так?

vvip-68, 2020-12-13:
> А по поводу монитора порта. Что у меня с ним не так?

Я даже не знаю чем вы пользуетесь. Но это точное не ArduinoIDE с его монитором порта.
Могу предположить, что ваш терминал принимает строку до первой точки с запятой, все остальное отбрасывает.
Ну, чисто визуально это выглядит так.

tarasifua, 2020-12-13:
Да, это PlatformIO в Visual Studio Code. Но это без разницы, в ArduinoIDE все аналогично
[image]

vvip-68, 2020-12-13:
Тогда не знаю.  Всё это выводится в строках 2089-2102 файла a_main.ino
```
Serial.print(F("UDP << ip='"));
      IPAddress remote = udp.remoteIP();
      Serial.print(remote.toString());
      Serial.print(":");
      Serial.print(udp.remotePort());
      Serial.print("'");
      if (udp.remotePort() == localPort) {
        Serial.print(F("; cmd='"));
        Serial.print(incomeBuffer);
        Serial.print("'");
      }
      if (udp.remotePort() == 123) {
        Serial.print(F("; ntp sync"));
      }
```
Если у вас в этих строках этого нет - вероятно у вас какая-то оооочень древняя версия, полагаю не менее чем двухмесячной давности. Что в основном файле в первых строках, где номер текущей версии. Типа такого:
```#define FIRMWARE_VER F("WiFiPanel-v.1.07.2020.1213")```

vvip-68, 2020-12-13:
Или в вашем случае не выполняется udp.remotePort() == localPort

tarasifua, 2020-12-13:
Моя версия
`#define FIRMWARE_VER F("LED-Panel-WiFi v.1.07.2020.1206")`
Код присутствует
[image]

vvip-68, 2020-12-13:
Значит не выполняется udp.remotePort() == localPort
Это я не знаю почему. По идее команда должна обрабатываться только когда она приходит на определенный порт UDP.
Если у вас порт не совпадает - оно вообще не должна воспринимать никакие команды...

vvip-68, 2020-12-13:
Уберите этот if() - будет печататься входная команда. И, видимо, время от времени - мусор, потому что приходят в буфер данные синхронизации NTP? возможно еще от погоды что-то -- не помню...

vvip-68, 2020-12-13:
Я смотрю у вас порт - 56351. По умолчанию - 2390. Переопределяли?
Возможно это мой косяк - надо посмотреть - что в данном случае в localport - считанное из настроек или (по-видимому) все же значение по умолчанию, что ошибка...

vvip-68, 2020-12-13:
Хотя, по коду - именно то, что в localPort (даже если переопределяли) используется в ufp.begin(localPort);
Nffr что вообще ничего не понятно почему так...

tarasifua, 2020-12-13:
Нет, я никакие порты не переопределял. Но приложение для управления у меня запущено в эмуляторе Андроида на компе. Может в этом проблема?

vvip-68, 2020-12-13:
Не знаю. Я не запускал все это под эмулятором...

tarasifua, 2020-12-13:
В любом случае я убрал if, как Вы говорили и команды появились. Мне этого достаточно, чтобы перенести их в HA. Заметил еще 2 особенности. У меня установлена длительность эффекта 15 секунд. Все эффекты так и работают, кроме Тетриса, он почему-то отображается до "проигрыша", то есть пока область не заполнится до конца. Это так и должно быть?
И второе. В авто режиме, когда эффекты доходят до конца, до "Рассвета", дальше они начинают воспроизводится со случайного номера. Как можно сделать, чтобы они начинались сначала? А то некоторые эффекты никогда не показываются

vvip-68, 2020-12-13:
1. Файл custom.ini, строка 541-449. Там определяется - нужно ли переключать эффект пока бежит текст.
Сейчас там для игр - если режим "игра" и она не окончена или если режим MC_TEXT (это отображение IP alhtcf) и строка не пробежала до конца - fullTextFlag - ok сбросить в false - эффект не менять и далее.
Измените это проверку как вам нужно. Или отключите проверку на тетрисе, чтобы он не дожидался конца игры или включите на любом эффекте, чтобы он не менялся пока строка не пробежит

2.Тот же файл. Строка 385 - переход к следующему режиму, 419 - к предыдущему. Исправляйте там логику перехода с следующему режиму - к случайному ли или к режиму по порядку. Если дошли до последнего - thisMode ставьте в ноль. Если в обратную сторону - после достижения ноля - thisMode в кол-во эффектов - 1

vvip-68, 2020-12-13:
И - судя по описанному что после рассвета включается случайный режим - у вас направильно выставлено кол-во эффектов MAX_EFFECT. То есть при переходе к следующему счетчик на границу MAX_EFFECT не достигнут - номер режима увеличивается, но такого режима нет. А когда указан не существующий - он включает случайный.
Именно это по описанию у вас и происходит.

vvip-68, 2020-12-13:
Скорее - это у меня проблема в коде в плане определения -
```#define MAX_EFFECT              45         // количество эффектов, определенных в прошивке```
Нужно поменять (я поправлю)
```#if (USE_SD==1)
#define MAX_EFFECT              45         // количество эффектов, определенных в прошивке
#else
#define MAX_EFFECT              45         // количество эффектов, определенных в прошивке
#endif
```

tarasifua, 2020-12-13:
Спасибо, поправил, теперь работает

vvip-68, 2020-12-13:
Чё-то с JSON по ресурсам довольно жирно выходит и стандарты JSON соблюсти не удается простым путем...
Может, все-таки строку парсить будешь?

tarasifua, 2020-12-13:
Ну если нельзя, то выбор невелик) Пусть будет строка. Может, если возможно, сделайте не одну, а несколько строк, где параметры будут рассортированы. К примеру 1 строка - одна вкладка параметров в приложении. Чтобы не парсить километровую строку. 
Еще заметил баг. Режим "Анимация" отображается поверх эффектов. Стандартный Марио у меня то в огне бегает, то под дождем :-)

vvip-68, 2020-12-13:
Могу и по частям - там проще - буфер точно не переполнится. Однако - там примерно 7 страниц. Интервал между отправками - минимум 1 сек.  Статусы будут приходить за 7-10 секунда вся пачка. А чем парсинг километровой строки отличается от коротких? Тот же цикл, от разделителя до разделителя, извлекаем строку, делим на части по двоеточию. Слева ключ, справа значение...

Это не баг, это фича. Если хочешь чтоб Марио по черному фону бегал - файл bitmap1.h, строка 124-130 вместо фоновых эффектов поставь '-1'

tarasifua, 2020-12-13:
Разницы действительно нет в парсинге, но есть в том, что если поменяется 1 параметр, система будет парсить все параметры из строки и сверять их с имеющимися. И я не уверен что не будет глюков, если, к примеру быстро поменять несколько параметров. Система еще не успеет после первой смены считать данные, а тут еще раз надо считывать.

> Может тебе как для приложения сделать - там при получении команды на странице или переключения на страницу отдается состояние элементов этой страницы.

Так в том то и идея, чтобы параметры были доступны всегда, без дополнительных запросов. Тогда тот же Home Assistant сможет моментально реагировать на их изменение

За подсказку с Марио спасибо, я думал что это глюк)

Vovaka1, 2020-12-13:
Привет. После переустановки приложения  GyverPanelWiFi.apk (с исправлениями). не получается изменить "Время режима". Ставлю прошлое приложение эта функция активна. Или у меня какой то косяк, я ведь только третий день вникаю.

vvip-68, 2020-12-14:
@Vovaka1, мой косяк. Недосмотрел. Исправил. Выложил. Обновитесь.

vvip-68, 2020-12-14:
@tarasifua - так запросов дополнительных нет. Если по аналогии с приложением - отправил команду со страницы (там галочку поставил или tit что) - тебе в ответ на эту команду в приложение присылается текущее состояние элементов этой страницы. В идеале конечно да - при старте отправлять полную картину, далее при изменении параметра отправлять только для измененных значений. Но в текущей архитектуре это довольно трудоемко реализовать.

sanyafifa, 2020-12-14:
> Еще просьба - когда всё настроите и будет работать - не могли бы вы написать инструкцию по интеграции с Home Assistant (или другими системами, если знаете) - чтоб я потом это в wiki положил - и люди бы знали как и что нужно делать для.
> Для меня вот это - вообще темный лес.

Доброго времени суток! Спасибо за ваши труды.

sanyafifa, 2020-12-14:
> Еще просьба - когда всё настроите и будет работать - не могли бы вы написать инструкцию по интеграции с Home Assistant (или другими системами, если знаете) - чтоб я потом это в wiki положил - и люди бы знали как и что нужно делать для.
> Для меня вот это - вообще темный лес.

Доброго времени суток! Спасибо за ваши труд!
Присоединяюсь к запросу функции отдельного топика MQTT получения статуса и тек. настроек. 
Формат JSON всеяден для любых серверов УД, У меня например Node-Red. могу помочь с инструкцией по интеграции.

vvip-68, 2020-12-14:
@sanyafifa - по статусам - в работе.
Инструкция - это здорово. Да, нужна. Выложу потом все инструкции в wiki проекта
Если есть возможность - писать сразу в формате md-документа, чтобы просто выложить текст без переработок.

tarasifua, 2020-12-14:
> запросов дополнительных нет. Если по аналогии с приложением - отправил команду со страницы (там галочку поставил или tit что) - тебе в ответ на эту команду в приложение присылается текущее состояние элементов этой страницы. 

Понял, но такой вариант не уверен что подойдёт. Алгоритм действий Home Assistant следующий: он видит изменение данных в топике > выполняет какое-то действие. В нашем случае парсит строку с топика и сверяет данные с имеющимися, каждый параметр. Команда парсинга у меня приблизительно такая:
`{trigger.payload.split(" ")[1].split(";")[0].split("|")[1].split(":")[1]}}'`
Как видите, тут строка обрезается от лишнего спереди и сзади, а дальше делится на части и прописывается в HA, без проверки названия параметра. Просто знаем, что если, к примеру яркость приходит первая в строке, значит первое значение прописываем в параметр яркости в HA. Если каждый раз будет приходить разная строка, алгоритм не сработает.
Я просто не уверен, разрешены ли там циклы, чтобы проверять по названию параметра.
ПС: пока интеграция выглядит как-то так 
[image]
[image]

vvip-68, 2020-12-14:
@tarasifua - предположение, что приходящий параметр будет в этом месте - крайне неверная и порочная практика (предполагать, что яркость BR будет на первом месте). Завтра мне понадобится что-то изменить / добавить - и у вас всё сломается.

Вернулся к попыткам отдавать параметры в виде json...

tarasifua, 2020-12-14:
Да, я знаю что так не совсем правильно. Когда я тестировал с вариантом запроса
`$6 7|DM|BR|PW|;`
то параметры приходили именно в том порядке, в котором я их запрашивал. Потому я и думал что тут будет так же

vvip-68, 2020-12-14:
@tarasifua, да, в случае, когда ты САМ запрашиваешь через $6 7 - будут приходить в том же порядке, в каком запрошено.
Но ты же просишь переделать, чтобы скетч тебе сообщал о значениях параметра и их изменениях. А значит, порядок уже на совести скетча и ты не можешь контролировать в каком порядке тебе придут параметры - это уже на усмотрение разработчика прошивки.

tarasifua, 2020-12-14:
Да я то могу под Ваш порядок настроить. Я имел ввиду то, чтобы порядок не менялся при каждом обновлении параметров.

vvip-68, 2020-12-14:
Вот как раз стоит ориентироваться на последний вариант. В случайном порядке. И не всегда и не все 50 параметров. 
Если поменялся только один или два - зачем присылать всю пачку из 50? Если на это не ориентировано - ваша программа легко ломается при любом изменении автором прошивки.

sanyafifa, 2020-12-14:
Может отдельную тему стоит создать. 
тема интересная и обширная.

shostko, 2020-12-14:
Моя конфигурация на данный момент поскромнее будет: https://github.com/shostko/home/blob/4401eaa85280b38de2fccf92f209564abef7f3f2/packages/balcony_panel.yaml

Обратная связь сделана только на список режимов.
Хотя в целом я не вижу для себя необходимости переносить все настройки панели в Home Assistant, но с удовольствием бы глянул на конфиги, если @tarasifua может ими поделиться)

tarasifua, 2020-12-14:
Поделюсь обязательно. На данный момент не могу разобраться с парсингом строки с параметрами. Проблема в том, что разные значения надо хранить в разных сущностях (input_boolean, input_number, input_state). И я никак не могу придумать как сделать автоматизацию, которая бы выполняла разные действия в зависимости от типа значения

shostko, 2020-12-14:
> Команда парсинга у меня приблизительно такая

@tarasifua тут мне кажется более уместным будет использовать регулярные выражения. и не потребуется точный порядок

shostko, 2020-12-14:
> И я никак не могу придумать как сделать автоматизацию, которая бы выполняла разные действия в зависимости от типа значения

Если я вас правильно понял - это по идее надо делать отдельные автоматизации, с одинаковыми тригерами и разными кондишенами (для каждого типа своя автоматизации со своим кондишеном)

[63 more replies]

closed 2020-12-14 by vvip-68