vvip-68/GyverPanelWiFi #61: MQTT не выключается из програмы

vaicekas, 2020-12-13:
В скетче включил МQTT, а в програме на телефоне снимаю галочку, но она всё равно появляеться, и МQTT стартует.
В английской версии аппса нет есчё этих настроек, и настроек погоды также.
И такой вопрос: если установлено короткое время показа эфекта, и поверх него бегущая строка-как добиться полного показа всего текста? Теперь обрубаеться по истичении времени эфекта.

vvip-68, 2020-12-13:
> В скетче включил МQTT, а в програме на телефоне снимаю галочку, но она всё равно появляеться, и МQTT стартует.

Да, действительно - при загрузке эта галочка не считывалась с EEPROM... Исправил

> В английской версии аппса нет есчё этих настроек, и настроек погоды также.

Английская версия была сделана по просьбе отдельных англоязычных пользователей, поскольку все исходники приложения были в Thunkable и редактировать его могли только те, ку кого был аккаунт в этом конструкторе. Но регистрация новых пользователей в Thunkable была закрыта его разработчиками и те, кому нужна версия на других языках не могли ничего предпринять. Теперь приложение перенесено в AppInventor, регистрация там открыта - все кому нужно могут самостоятельно переводить интерфейс на любой нужный ему язык. Особенности конструктора - там нет dll-ек ресурсов, в которых хранятся строки. Как на экране нарисовал - так и будет.  Я не готов за бесплатно поддерживать версии на разных языках.

> И такой вопрос: если установлено короткое время показа эфекта, и поверх него бегущая строка-как добиться полного показа всего текста? Теперь обрубаеться по истичении времени эфекта.

Это что ж за текст у вас такой длинный? Или почему время смены эффектов такое короткое?
Какой-то непрактичный момент, который даже не принимался во внимание при написании.
А что по итогу то нужно? Чтобы эффект менялся, а текст продолжал бежать? Или чтоб эффект не менялся, пока текст не пробежит?

vvip-68, 2020-12-13:
Хм... А я вот сейчас проверил - пока бежит строка - эффекты под ней меняются по своему установленному времени, а строка продолжает бежать? Что я делаю не так?
Один нюанс - если в настройках очередного эффекта снята галочка, что поверх него можно отображать строку (то есть показ строки на этом эффекте отключен) - то строка, естественно, перестает показываться. А как иначе то?
Вот поверх Анимации или Погоды бегущий текст вообще не смотрится - там еще и наслоение текста эффекта на текст строки - каша получается. А в тетрисе или змейке - вообще невозможно - ибо там анализ пикселей игрового поля с матрицы считывается - а там строка бежит и все портит.

closed 2020-12-13 by vvip-68

vaicekas, 2020-12-13:
Втом то и дело, если поставить смешывание эфектов то следуюшщий эфект(неразрешён показ текста на нём) обрубает текст. Правильней было бы показать текст до конца и только тогда переключить эфект.

vaicekas, 2020-12-13:
С английской версией всё понятно. По руски всё понимаю, вот только писать трудновато ;)
А есть возможность слить все настройки и после обновления програмы залить обратно?

vvip-68, 2020-12-13:
Файл custom.ini, строка 541-449. Там определяется - нужно ли переключать эффект пока бежит текст. 
Сейчас там для игр - если режим "игра" и она не окончена или если режим MC_TEXT (это отображение IP alhtcf) и строка не пробежала до конца - fullTextFlag - ok сбросить в false - эффект не менять и далее.
Измените это проверку, чтобы на ЛЮБОМ эффекте - пока флаг fullTextFlag не установился в true - эффект не менять.

Настройки прошивки никакой связи с приложением не имеют. Прошивка все текущие параметры отправляет в приложения. Будь они на русском, английском или еще каком языке.

Вообще - все настройки сохраняются в постоянной памяти  EEPROM. Она не стирается при замене прошивки в режиме SketchOnly и пока значение EEPROM_OK вы не смените на другое.

Значения инициализируется по-новой, если считанное из ячейки EEPROM с номером 0 не соответствует значению константы EEPROM_OK или если вы в настройках загрузки прошивки не поставите режим Erase Flash: all contnt.

А вот так вот, чтобы куда-то экспортировать настройки, а потом их загрузить - такого нет...

vaicekas, 2020-12-13:
Большое спасибо за разяснение про настройки. У меня стояло в настройках Erase Flash... теперь не надо перебирать все настройки эффектов занова.
Залил последную прошивку: с MQTT всё отлично! Но грусть и печяль- Вы убрали Литовские буквы (я из Литвы). Дополнил их в fonts.h из бывшей прошивки, но их не выводит на гирлянду, пустые места только :(
Буду пробовать с fullTextFlag в custom.ini, может разберусь самостоятельно.
Пожалуйста, верните буквы :)
СПАСИБО!

vvip-68, 2020-12-14:
@vaicekas - там не только в fonts.h буквы прописаны были. Это был опытный вариант по просьбе одного парня из Литвы - спецпроект. Что-то я думал это не надо никому - убрал. Ладно, верну... В следующем обновлении будут.

vvip-68, 2020-12-14:
@vaicekas, вернул буквы. Забирайте, проверяйте.

vvip-68, 2020-12-14:
@vaicekas, вы MQTT для чего используете? Интеграция с каким-то умным домом? С каким?
Вы будете третьим, кто использует MQTT. Если вы что-то уже сделали - не могли бы составить инструкцию по интеграции, чтобы я мог выложить ее в раздел wiki - может еще кому интересно будет.

С MQTT будут изменения. Один из пользователей просит меня добавить топик состояний и отправлять туда значение [изменившихся] параметров, а также начальный набор состояний при старте прошивки...

А вас то всё с MQTT устраивает? Или чего-то не хватает?

vaicekas, 2020-12-14:
Спасибо за буквы! Вечером попробую.
По правде MQTT я неиспользую. Думал что будет вазможность управлять гирляндой с улицы, но немножко покапавшысь в настройках- понял что слишком сложно это для меня, и успокоился :)
А так всё устраивает! Класный проект! Уже третий год укрошаю жылой раион :) Спасибо Вам за это!
Если кому то есче пригодиться, я подправил себе на литовский язык слова даты и времени для бегушчей строки, могу запостить.

vaicekas, 2020-12-14:
Обновился до последней версии. Буквы проверить не удалось, так как что-то не так с регулировкой яркости, она падает до нуля за несколько секунд. 
Не компилируется, если в настройках гирлянды поставить SD CARD 0
Если указать SD CARD 1 все проходит без ошибок.

vaicekas, 2020-12-14:
Я не пользуюсь кнопкой. Может надо резистор включить?

vvip-68, 2020-12-14:
Не, не надо никакого резистора. У меня тоже кнопки сейчас нет, но все работает, ничего с яркостью не происходит - какую поставил, та и светит...

vaicekas, 2020-12-14:
Переключил все заново и исправилось. 
Гирлянда снова говорит по-литовски! БЛАГОДАРЮ ВАС!

vaicekas, 2020-12-14:
вот что выдает при смене яркости ползунком

UDP << ip='192.168.0.122:2390'; cmd='$4 0 188.96001;'
UDP пакeт размером 15
UDP << ip='192.168.0.122:2390'; cmd='$4 0 181.34;'
UDP пакeт размером 12
UDP << ip='192.168.0.122:2390'; cmd='$4 0 176.25999;'
UDP пакeт размером 15
UDP << ip='192.168.0.122:2390'; cmd='$4 0 173.72;'
UDP пакeт размером 12
UDP << ip='192.168.0.122:2390'; cmd='$4 0 168.64;'
UDP пакeт размером 12
UDP << ip='192.168.0.122:2390'; cmd='$18 0;'
UDP пакeт размером 6
UDP << ip='192.168.0.122:2390'; cmd='$18 0;'
UDP пакeт размером 6
UDP << ip='192.168.0.122:2390'; cmd='$4 0 163.56;'
UDP пакeт размером 12
UDP << ip='192.168.0.122:2390'; cmd='$4 0 166.10001;'
UDP пакeт размером 15
UDP << ip='192.168.0.122:2390'; cmd='$4 0 168.64;'
UDP пакeт размером 12
UDP << ip='192.168.0.122:2390'; cmd='$4 0 171.17999;'
UDP пакeт размером 15
UDP << ip='192.168.0.122:2390'; cmd='$18 0;'
UDP пакeт размером 6

не много цифер после запятой?

vvip-68, 2020-12-14:
Ну оно ж никому не мешает? Работает? Когда программа была в Thunkable - не было возможности вставлять лишние блоки, которые бы усекали число до целого. Сейчас в AppInventor есть - но это ж нужно иди искать где это передается, искать как сделать усечение дробной части / округление до целого. В монитор же вообще никто не смотрит практически. Даже те, ку кого что-то не в порядке - все время "а почему" и даже в лог заглянуть не удосужатся, чтобы посмотреть.
А вы перфекционист, батенька... :)

vaicekas, 2020-12-14:
есть немножко :)....
не мешает - так немешает. Подумал может это как то нагружает проц. 
Пользуем дальше ;)

fastdron, 2020-12-14:
По поводу яркости - тоже замечал что иногда яркость сбрасывается почти на минимум, но так и не понял когда именно это происходит, после вывода одной из строк текста (не уверен , не смог повторить , но яркость реально сбрасывается частенько)

AndreyLyamkin, 2020-12-15:
> @vaicekas, вы MQTT для чего используете? Интеграция с каким-то умным домом? С каким?
>
> Вы будете третьим, кто использует MQTT. Если вы что-то уже сделали - не могли бы составить инструкцию по [...]

Я использую MQTT свой. В интеграции умного дома Home Assistant и тоже был бы рад расширенной поддержке прошивки с MQTT.

vvip-68, 2020-12-15:
@fastdron - ну, если выясните причину - дайте знать. У меня такого не происходит. Я не могу чинить то, что не сломалось...