vvip-68/GyverPanelWiFi #62: Работа с MQTT

RadioLed2000, 2020-12-15:
Добрый день.
Уважаемый @vvip-68, раз Вы написали, что планируете делать изменения в работе с MQTT, то с вашего позволения открою отдельную тему для этого.

**Первое.** Функция MQTT Last Will and Testament.
В файле mqtt.ino в функции void checkMqttConnection() делаем изменение:
```
    if (mqtt.connect(mqtt_client().c_str(), mqtt_user, mqtt_pass, mqtt_topic("state").c_str(), 0,  true, "offline")) {
	  mqtt.publish(mqtt_topic("state").c_str(), "online", true);
      Serial.println(F("\nПодключение к MQTT-серверу выполнено."));
```
После этого в топике state мы всегда получим online, если устройство в сети, и offline, если нет связи с сервером, или устройство кто-то варварски выдернул из розетки. Это очень ценная функция для мониторинга удалённой гирлянды ))
Этот функционал используется многими системами умного дома. Этот код я лично проверил, всё работает как описано.

**Второе.** Я посмотрел новую функцию void SendCurrentState(). Мне кажется хорошей идеей сделать такую функцию, а выбор ключей предоставить пользователям непосредственно в коде. Всем не угодишь, каждому нужны разные значения, пусть каждый и выберет то, что ему надо.
Единственное замечание: а чем обусловлено использование вложенных массивов? Получается громоздкая конструкция, которую не каждый парсер поймёт. Я предлагаю использовать такой вариант попроще:
```
// Отправка в MQTT канал - текущие значения переменных
void SendCurrentState() {  
  if (!useMQTT) return;
  
  String out, key, s_tmp, keys = "W|H|DM|PD|IT|AL|RM|PW|BR|EF|IP";
  
  DynamicJsonDocument doc(1024);
 // DynamicJsonDocument obj(256);
//  doc["act"] = F("STATE");
//  JsonArray arr = doc.createNestedArray("state");  

  int16_t pos_start = 0;
  int16_t pos_end = keys.indexOf('|', pos_start);
  int16_t len = keys.length();
  if (pos_end < 0) pos_end = len;

  // Строка keys содержит ключи запрашиваемых данных, разделенных знаком '|', например "CE|CC|CO|CK|NC|SC|C1|DC|DD|DI|NP|NT|NZ|NS|DW|OF"

  while (pos_start < len && pos_end >= pos_start) {
    if (pos_end > pos_start) {      
      key = keys.substring(pos_start, pos_end);
      if (key.length() > 0) {
        s_tmp = getStateValue(key, thisMode, false);        
        if (s_tmp.length() > 0) {
         // JsonObject object = obj.to<JsonObject>();        
         // obj[key] = s_tmp;
         // arr.add(obj);
         doc[key] = s_tmp;
        }
      }      
    }
    pos_start = pos_end + 1;
    pos_end = keys.indexOf('|', pos_start);
    if (pos_end < 0) pos_end = len;
  }
  serializeJson(doc, out);      
  String data = out;
  putOutQueue(mqtt_topic(TOPIC_STT).c_str(), data.c_str());  
}
```

Вот пример полученного ответа: 
{"W":"16","H":"10","DM":"1","PD":"60","IT":"0","AL":"0","RM":"1","PW":"2000","BR":"1","EF":"30","IP":"192.168.100.119"}

**Третье.** Предлагаю добавить ключи для получения состояния гирлянды включено/выключено и названия текущего эффекта текстом. Достаточно использовать переменную isTurnedOff и сделать глобальной переменную effect_name.

tarasifua, 2020-12-15:
И так, резюмируя, сейчас мне видится следующая картина:
1. Существующие топики 
```
#define  TOPIC_CMD      "cmd"            // Топик - получение команды управления от клиента
#define  TOPIC_DTA      "dta"            // Топик - отправка запрошенных данных клиенту
#define  TOPIC_NFO      "nfo"            // Топик - отправка информационных уведомлений клиенту
#define  TOPIC_ERR      "err"            // Топик - отправка уведомлений об ошибке клиенту
#define  TOPIC_ACK      "ack"            // Топик - отправка уведомлений клиенту
```
и сообщения, которые летят в них лучше всего оставить без изменений.
2. Добавить топик "state", где будет значение, определяющее включена ли матрица или нет. 
3. Добавить топик "parms", где будут выводиться параметры матрицы при ее загрузке и обновляться при их изменении. Выбор списка параметров сделать в коде, как предложено выше.
4. Ну и добавьте, пожалуйста, параметр, который показывает включен ли черный экран. Это не совсем то, что в пункте 2, матрица может быть физически включена, но ее включение/выключение производится включением-отключением этого режима ($14 0).

Возможен такой вариант? Или идти искать губозакаточную машинку?

vvip-68, 2020-12-15:
Да возможен, наверное. Работаю. По мере появления чего-нибудь вразумительного - выложу.

tarasifua, 2020-12-15:
Отлично, спасибо большое, будем ждать

sanyafifa, 2020-12-15:
> #define  TOPIC_CMD      "cmd"            // Топик - получение команды управления от клиента
> #define  TOPIC_DTA      "dta"            // Топик - отправка запрошенных данных клиенту
> #define [...]

> 4. Ну и добавьте, пожалуйста, параметр, который показывает включен ли черный экран. Это не совсем то, что в пункте 2, матрица может быть физически включена, но ее включение/выключение производится включением-отключением этого режима ($14 0).

Имеется ввиду с клиента на сервер должно приходить? 
Может сделать более информативным `TOPIC_ACK` , ввел команду - прилетело что изменилось...

tarasifua, 2020-12-15:
В топик "parms" по задумке должен прилетать список необходимых параметров и обновляться, если что-то изменилось. С него уже достаточно легко спарсить необходимые параметры.

vvip-68, 2020-12-15:
Вся проблема, что в си нет "свойств"  как во всех нормальных языках, чтобы изменил значение свойства - выполнился сеттер, который отправил сообщение что изменилось и какое новое значение. поэтому процедура отслеживания изменений параметров - тот еще геморрой. Параметры от которых зависит изменение состояния меняются во многих местах скетча, поэтому приходится либо править весь текст без гарантии что ничего нигде не пропустил, либо извращаться, что требует значительного количества ресурсов программы. Уже не уверен, что на ESSP8266 вообще хватит памяти на работу с MQTT. Я пока ковыряюсь с ESP32 - там памяти пока есть. 

Топик ACK как сообщали выше - для MQTT бесполезный - убрал.
Планируется (как просили) два режима - для бесплатных MQTT где ограничения на кол-во запросов - отправка параметров пакетом - длинная строка как в самом первом сообщении в этом топике, содержащая все параметры (retain) и для быстрых серверов - отправка только изменившихся - по одному значению (на самом деле та же пара ключ-значение, но одна) и тоже с retain.

По ресурсам - параметров сейчас 84+ штуки. Значит очередь на отправку должна быть больше, иначе часть сообщений вероятно будет теряться если сервер медленный с ограничением отправки и режим отправки не пакетами.. Это расходы памяти. По описанным выше размышлениям - определить что именно изменилось в момент изменения - затруднительно. Поэтому придется хранить состояние предыдущих значений (опять расходы памяти), потом сравнивать, понимать что изменилось и отправлять только изменившееся.

Далее придется переделывать работу с редактором текстов бегущей строки. Пока одно андроид приложение - там выбирается какая ячейка текста активная, прошивка это знает и работает именно с этой ячейкой. В случае канала MQTT клиентов может быть несколько и каждый может редактировать свою ячейку текста. Значит нужно переделывать систему редактирования текстов. 

А как в MQTT вообще принято - вот параметр изменился, мне надо его отправить на сервер, а сервер не отвечает / не принимает посылку. И что с этим сообщением мне делать?

RadioLed2000, 2020-12-15:
> А как в MQTT вообще принято - вот параметр изменился, мне надо его отправить на сервер, а сервер не отвечает / не принимает посылку. И что с этим сообщением мне делать? Я пока в очередь складываю - сервер ответит - очередь отправится. Но ее может и не хватить. Получается рассинхронизация состояний что реально в устройстве и что об этом знает сервер. Как в этом случае поступают?

Как я понимаю, если сервер не отвечает, значит с ним утеряно соединение. Соответственно производим попытку соединения заново. А при успешном подключении производим отправку параметров.

vvip-68, 2020-12-15:
> Как я понимаю, если сервер не отвечает, значит с ним утеряно соединение. Соответственно производим попытку соединения заново. А при успешном подключении производим отправку параметров.

Ну то есть - либо тормозим весь скетч и в цикле пытаемся отправить, сервер не отвечает 10 минут, весь скетч завис. 
Или складываем в очередь (расходы памяти), очередь переполняется, сообщение теряется - рассинхронизация состояний.

vvip-68, 2020-12-15:
У меня раньше было просто mqtt.publish(...)
Потом выяснилось, что бесплатные сервера не принимают чаще одного в секунду. Или сервер просто не отвечает и рвет соединение (так было с srv2). Пришлось организовывать очередь. Пока предполагалось, что клиент сам будет спрашивать  что ему нужно (знаю - неправильно) - очередь вполне вмещалась в 8-32 элемента. Если отправлять пакетами - может и хватит. Но тогда при любом изменении параметра клиенту отправляются все 84 элемента, слепленные в строку почти 3-4 килобайта. И клиенту все это нужно будет распарсить. Или вычленять что изменилось (хранение предыдущих состояний - память) и проверка ЧТО изменилось - время. Плюс если сервер с задержками (бесплатный), а включен режим не пакетной отправки, а каждого измененного значения - на очередь может потребоваться уже около 90 элементов - снова память.

RadioLed2000, 2020-12-15:
> Ну то есть - либо тормозим весь скетч и в цикле пытаемся отправить, сервер не отвечает 10 минут, весь скетч завис.
> Или складываем в очередь (расходы памяти), очередь переполняется, сообщение теряется - рассинхронизация состояний.
Тут я не знаю, как правильно... Возможно стоит почитать описание функции QoS протокола MQTT. Там задаётся контроль приёма/передачи пакетов. Но я с ним не заморачивался, у меня в локальной сети и так всё работает.

tarasifua, 2020-12-15:
Такой вопрос? А если использовать бесплатный MQTT без ограничения раз в 1 сек?

vvip-68, 2020-12-15:
QoS то тут причем? Просто приоритезация сообщений. Что в первую очередь, что не очень срочно. И никак не решает если сервер вообще перестал отвечать и состояние рассинхронизировалось.

vvip-68, 2020-12-15:
>Такой вопрос? А если использовать бесплатный MQTT без ограничения раз в 1 сек?

Жду предложений. Ч в поисках таких бесплатных серверов перерыл кучу вариантов. Ограничение в одно сообщение в секунду - самый безобидный вариант. Остальные были куда как с более неприятными ограничениями.

tarasifua, 2020-12-15:
Ну, если с ходу, http://mqtt.by/, например

RadioLed2000, 2020-12-15:
> 
> 
> QoS то тут причем? Просто приоритезация сообщений. Что в первую очередь, что не очень срочно

QoS в MQTT - это подтверждение доставки сообщений. 
https://ipc2u.ru/articles/prostye-resheniya/chto-takoe-mqtt/#kachestvo

tarasifua, 2020-12-15:
Есть еще такой
http://www.dioty.co/mqtt
50000 сообщений в месяц лимит, для нашего случая этого с головой, даже если матриц 10 штук

shostko, 2020-12-15:
словил интересную ситуацию:
в моей [текущей конфигурации](https://github.com/shostko/home/blob/6403f147617e417e44f7cc6f3405446c6e067b98/packages/balcony_panel.yaml) получается так, что при включении автоматической смены режимов смена происходит только один раз (потом переходит в ручной мод)
происходит это потому, что при смене режима по MQTT сервер получает изменение, отрабатывает одна автоматизация - меняет локальное состояние дропдауна с режимами, из-за этого (состояние сменилось) отрабатывает другая автоматизация - отправляет по MQTT команду на смену режима, панель принимает команду (режим тот же), и отключает авторежим смены режимов.

я пока не придумал, можно ли это решить средствами HA (@tarasifua у вас такой проблемы нету?)
@vvip-68 может быть имеет смысл сделать какую-то настройку в прошивке, которая будет запрещать отключение авторежима при ручной смене режима? ну или не отключать его если фактически режим выбран тот же, что думаете?

vvip-68, 2020-12-15:
> которая будет запрещать отключение авторежима при ручной смене режима? ну или не отключать его если фактически режим выбран тот же, что думаете?

То есть я вручную сменил режим (ну нравится он мне - поставил чтоб постоянно крутился), поставил нужны мне, а оно все равно хрясь - и сменила его на следующий? Не, это неправильно.

У меня в андроиде та же проблема. Я решал её так - когда мне надо установить значение комбобокса на пришедшей с прошивки состояние - я устанавливал флаг - valueChanging, и присваивал новое значение. При этом срабатывала обработка, которая проверяла - если флаг установлен - ничего не делать. После того как новое значение установлено и обработка проигнорирована - снимал флаг .

vvip-68, 2020-12-15:
Чё-т пока всё не радужно.... Для формирования строки (пакета) вот этих вот переменных состояния для отправки на сервер в формате json я использую библиотеку ArduinoJson - и в ней DynamicJsonDocument, куда складываются вот эти вот Key-Value-Pairs. Только глючит она. То ли из-за недостатка, то ли из-за фрагментации памяти (хотя на ESP32 пишет использовано 19% оперативной памяти) довольно часто кладешь туда значение, а считываешь - возвращает null. В результате текущее и предыдущее значения не равны и оно шлет-и-шлет-и-шлет этот параметр, хотя фактически он не изменился... Да и эффекты подтормаживают на проверке не изменилось ли значение.

Вопрос - как можно быстро обнаружить какой параметр изменился, чтобы отправить измененное значение?
Какое-нибудь не ресурсоемкое и быстрое хранилище пар ключ-значение...

Иначе мне придется полностью переписывать весь скетч, заменяя во всех местах переменные от которых зависит состояние на вызов функции - сеттера (т.к. нормальных properties с си нету... А это долго и муторно писать и чревато ошибками при которых я что-то где-то пропущу...

Кстати, обнаружил что бесплатный сервер на который я ориентируюсь в прошивке в настоящий момент отменил ограничение на 1 сообщение в секунду. Сейчас вся очередь без задержки влет отправляется. И приходит на сервер без пропусков.

tarasifua, 2020-12-15:
> @tarasifua у вас такой проблемы нету?

Нет, но у меня и такой автоматизации нету. Не вижу смысла видеть какой сейчас эффект проигрывается. Если очень надо, сделайте отдельную переменную и пишите туда. 
ПС: А в HA нету какой-то возможности узнать, чем было изменено состояние? Просто если в Инструментах разработчика смотреть историю изменений, то там явно видно, кто или что изменил состояние

tarasifua, 2020-12-15:
> Кстати, обнаружил что бесплатный сервер на который я ориентируюсь в прошивке в настоящий момент отменил ограничение на 1 сообщение в секунду. Сейчас вся очередь без задержки влет отправляется. И приходит на сервер без пропусков.

Так может в таком случае слать каждый параметр в отдельный топик при изменении и все? Да, будет куча топиков, но за то, как я понимаю, не надо ничего отслеживать и формировать километровые json-ы. Параметр изменился - параметр улетел.

vvip-68, 2020-12-15:
А еще подумайте как решать следующий конфликт. По аналогии вот с приложением в Андроид, если соберетесь писать "аналог" только в HA с полным управлением.
И это связано со снятием авторежима тоже при ручной установке эффекта.
Я захожу на страничку настройки эффектов, выбираю его значение. На пвнкль уходит выбранный эффект, он включается, состояния ползунков настроек изменяется. Далее я что-то настраиваю, а в прошивке сменился автоматически эффект и мне прилетело уведомление о новом эффекте и его параметрах. И все что я настраивал для выбранного мной - заменилось в элементах управления на новые полученные значения. ТО же самое и если два клиента одновременно будут пытаться что-то делать. То же самое, когда я в настройках текста что-то редактирую, а тут хрясь - с другого клиента прилетело обновление этой строки...

vvip-68, 2020-12-15:
> Так может в таком случае слать каждый параметр в отдельный топик при изменении и все? Да, будет куча топиков, но за то, как я понимаю, не надо ничего отслеживать и формировать километровые json-ы. Параметр изменился - параметр улетел.

Такой вариант я тоже сделал. На выбор. Но это никак не отменяет проблемы как понять что изменилось.

shostko, 2020-12-15:
> У меня в андроиде та же проблема. Я решал её так - когда мне надо установить значение комбобокса на пришедшей с прошивки состояние - я устанавливал флаг - valueChanging, и присваивал новое значение. [...]

@vvip-68  Ну как крайний вариант, но создавать в умном доме для этого глобальную переменную - не хочется. буду смотреть

> Не вижу смысла видеть какой сейчас эффект проигрывается

@tarasifua Ну этом есть определнный смысл, пусть и не большой. но обратная связь для меня в любом случае в итоге важна.

> А в HA нету какой-то возможности узнать, чем было изменено состояние? Просто если в Инструментах разработчика смотреть историю изменений, то там явно видно, кто или что изменил состояние

@tarasifua А вот это, кстати, идея хорошая. Пошел курить документацию...

> Но это никак не отменяет проблемы как понять что изменилось.
@vvip-68 ну я более адекватного варианта не вижу, пошел бы писать сеттеры

vvip-68, 2020-12-15:
>  ну я более адекватного варианта не вижу, пошел бы писать сеттеры

Это на неделю минимум... И потом долгая отладка и обнаружение глюков...
К февралю что-то получится. Может быть...

tarasifua, 2020-12-15:
> А вот это, кстати, идея хорошая. Пошел курить документацию...

Если найдете, дайте знать. Думаю, мне тоже пригодится

shostko, 2020-12-15:
@tarasifua не нашел такой возможности, но придумал другой вариант:

```
        action:
          - service: automation.turn_off
            data:
              entity_id: automation.balcony_panel_effect_with_mqtt
          - service: input_select.select_option
            ...
          - service: automation.turn_on
            data:
              entity_id: automation.balcony_panel_effect_with_mqtt
```

пока вроде работает норм. вот [тут](https://github.com/shostko/home/blob/352236d04c073ec0df52a40215c8e3f4ad515098/packages/balcony_panel.yaml) вся конфигурация

tarasifua, 2020-12-15:
Нда, костыль, конечно, но как вариант

AndreyLyamkin, 2020-12-16:
Я что-то не пойму, кто и зачем использует сторонние MQTT сервера с кучей ограничений? По обсуждениям видно, что все, кто заинтересован в улучшении работы прошивки с MQTT имеют свои собственные (локальные) сервера. А тем, у кого нет , может легко поставить на компьютере.

tarasifua, 2020-12-16:
Ну вот, к примеру, у ув. @vvip-68 некоторыми матрицами надо управлять удаленно. Не ставить же возле каждой комп с белым IP

vvip-68, 2020-12-16:
@AndreyLyamkin, я например.  У меня на удаленной квартире / в удаленном офисе висит гирлянда. Работает себе по расписанию. Бываю я там редко. Никаких серверов там нет - просто роутер с выходом в интернет. Ставить для этой (этих) гирлянд отдельно свой сервер у меня нет ни опыта ни желания. Мне достаточно бесплатного общедоступного сервера, чтобы видеть что гирлянда работает и удаленно настроить какие-нибудь параметры.

AndreyLyamkin, 2020-12-16:
так это решается запуском одного "своего" сервера.... Ну да, нужен комп...
Ну тогда, быть может, привести все это к общедоступному формату? Например, есть прошивка для умных устройств "Tasmota" (https://tasmota.github.io/docs/MQTT/), она из коробки посылает на сервер всю информацию, которые забирают все MQTT клиенты в понятном виде. Может там можно взять всю конфигурацию? Этим самым все будут удовлетворены.

vvip-68, 2020-12-16:
@AndreyLyamkin? возможно. Предлагается Вам это реализовать, интегрировать с гирляндой и предоставить нам на обсуждение.

AndreyLyamkin, 2020-12-16:
Эх если бы у меня было столько познаний в этой области, как у Вас!!!!

AndreyLyamkin, 2020-12-16:
> @AndreyLyamkin, возможно. Предлагается Вам это реализовать, интегрировать с гирляндой и предоставить нам на обсуждение. 

Эх если бы у меня было столько познаний в этой области, как у Вас!!!!

vvip-68, 2020-12-16:
Так я потому и мучаюсь (и чудю), потому что познаний недостаточно. А познание - дело наживное. Только этим нужно заниматься (вот что-то понадобилось - взял и сделал, приобрел опыт), а не искать готовые решения. Ну или искать, прать за основу и допиливать под себя.

tarasifua, 2020-12-16:
Тут наверное имелось ввиду программирование в принципе. С этим так просто не сядешь и не сделаешь)

vvip-68, 2020-12-16:
@tarasifua, путь в тысячу ли начинается с первого шага

tarasifua, 2020-12-16:
Извиняюсь, что не по теме, но тут возник вопрос по API. Что делает запрос $13 0 N;? Его надо отправлять перед редактированием текста строки? Просто вижу что приложение постоянно его шлет, хотя текст и без него прекрасно отправляется

vvip-68, 2020-12-17:
В следующей версии эта команда вообще удалена. Ибо предполагает работу один-один. ТО есть приложение шлет контроллеру индекс строки с которой типа будет работать. Далее прошивка отправляет в приложение состояние что эта кнопка активна (желтая) - ответ "TA" (который тоже будет удален). И текст выбранной строки "TX". И после изменения "TY" чтобы заменить строку на новую в списке в приложении.

Работа же MQTT подразумевает, что работа с текстом может вестись с нескольких клиентов, причем каждый со своей "ячейкой" -  следовательно такой подход порочен. Значит будет переделываться. Как - пока не знаю. Завяз на переделке присваивания простым переменным в работу через сеттеры, чтобы понимать ЧТО поменялось и ЧТО нужно отправлять на сервер. Это много, долго и муторно.

tarasifua, 2020-12-17:
А может не стоит так заморачиваться? Ну кто будет одновременно с 2-х устройств менять настройки?

vvip-68, 2020-12-17:
Даже тут уже писали - про одновременно с телефона (с приложения) и с HA... Не помню в связи с чем, но писали.

tarasifua, 2020-12-17:
> Даже тут уже писали - про одновременно с телефона (с приложения) и с HA...

Наверное это я писал в соседней теме, но немного неточно выразился. Имелось ввиду следующее. Если управлять матрицей только из HA, в принципе можно обойтись и без полной обратной связи, ведь все параметры сохраняются в HA. Разве что только сделать обратную связь на on/off, ну или на текущий эффект, если кому надо. Это можно сделать без проблем в текущей реализации работы с MQTT. А строчку с параметрами или отдельные топики с параметрами я просил сделать для случая, когда произошла "рассинхронизация" настроек в HA и матрице (к примеру из-за изменения настроек через приложение или по другой причине). Назначение этой строки - поддержка актуальных параметров в HA в любой момент, так как он может при изменении этой строки выполнять считывание актуальных данных. А об одновременной настройка с двух клиентов я и не думал, какой смысл.
Но если все-же надо обойти этот момент, может проще будет блокировать прием команд, к примеру, с приложения, пока идут команды с MQTT и наоборот?

tarasifua, 2020-12-17:
> А строчку с параметрами или отдельные топики с параметрами я просил сделать для случая, когда произошла "рассинхронизация" настроек в HA и матрице (к примеру из-за изменения настроек через [...]

Можно, конечно, и этот момент реализовать сейчас. Просто заставить HA постоянно слать запрос вида "$6 7|CE|CC|CO|CK......" с необходимым набором параметров, к примеру, раз в 30 секунд, и проверять, не изменилось ли что-то. Но это, как по мне, дикий костыль. Если реализовать строку с полным набором параметров сложно, может проще будет сделать отправку какого-то информационного сообщения, к примеру в топик "nfo", которое будет сигнализировать HA что произошло какое-то изменение, и надо обновить параметры?

vvip-68, 2020-12-17:
> Но это, как по мне, дикий костыль.

Это дикий костыль. И отправка что что-то произошло с просьбой спросить - тоже дикий костыль.
И ограничение на одновременно только одного клиента и блокировка другого - тоже костыль...

Про костыли и чужой или забытый код. Если потом кто-то захочит переписать...

Вот представь, что тебе доверили достроить за другим прорабом лабораторию на острове. Ты приходишь на объект, а там кроме недостроенного здания: огромный вентилятор (размером со здание), большой воздушный шар и комната набитая швабрами. Почесав голову, ты разбираешь этот хлам и доделываешь лабораторию. Сдаешь объект ученным, но через 5 минут они выбегают с криком: "УТЕЧКА ЯДОВИТОГО ГАЗА!!!".

- Как так-то, блять! Должно же работать! - в отчаянии кричишь ты и звонишь прошлому прорабу:
- Вася, у нас ядовитый газ потёк! В чем проблема?
- Не знаю, должно было все работать. Что-то в проекте менял?
- Немного, швабры вынес...
- Швабры потолок держали!
- Что??? Что, блять, извините???
- Говорю, швабры потолок держали. Над ними цистерны с газом были. Очень тяжелые, пришлось в комнату снизу швабры напихать.
- Ты хотя бы записку на двери повесил бы, что швабры для держания потолка! У нас тут ядовитый газ течет! Что нам делать?
- Включай вентилятор. Он сдует газ с острова.
- Я его, блять, демонтировал сразу же!
- Зачем?
- Зачем ты построил 120 тонный вентилятор? Ты не мог положить ящик блядских ПРОТИВОГАЗОВ?
- Ящик противогазов искать нужно, а вентилятор у меня с прошлого заказа оставался.
- Вася, я убрал твой вентилятор! Мы тут задыхаемся!
- Херли вы тогда там делаете? Садитесь на воздушный шар и уебывайте!

sanyafifa, 2020-12-17:
> А строчку с параметрами или отдельные топики с параметрами я просил сделать для случая, когда произошла "рассинхронизация" настроек в HA и матрице (к примеру из-за изменения настроек через приложение или по другой причине)

По логике я себе слабо представляю управление "и от туда и отсюда ". 
Если у человека есть сервер умного дома - то будет пользоваться его фронт эндом. (конечно если сможет перетянуть весь доступный в API функционал)
Я планирую все нужные команды завязать с телеграмм ботом. И управление панелью всеми членами семьи будет с "одной точки входа". 
Ну это так, мысли...

vvip-68, 2020-12-17:
@sanyafifa? не все такие умные. У меня вот не телефоне приложение. У жены, у детей. Может потом РФ будет, если расскажете как сделать де еще полнофункциональное управление прикрутите. И каждый со своего телефона из своей комнаты, предположим, начнет делать. Нормальная система и такой вариант должна учитывать тоже. См. законы Мерфи выше.

RadioLed2000, 2020-12-17:
> Если у человека есть сервер умного дома - то будет пользоваться его фронт эндом. (конечно если сможет перетянуть весь доступный в API функционал)

Я, например, использую приложение, когда нужно произвести начальную настройку гирлянды. А после этого в HA используется только вкл/выкл и яркость, плюс управление Алисой. А она понимает даже ребенка, который половину звуков не выговаривает. 

Так что я лично не вижу нужды дублировать функционал приложения в Home Assistant. Он нужен для удобного управления любым членом семьи. Туда сведены все гирлянды и освещение и привязано голосовое управление. А приложение реализует сервисные функции, и в него полезу только я чтобы что-то настроить. ИМХО))

vvip-68, 2020-12-17:
А мне вот опять же нужно полнофункциональное приложение. Для все тех же гирлянд, которые висят чёрти-где и до них ни приложением, ни докричаться...

tarasifua, 2020-12-17:
У каждого свои потребности. У меня например все устройства на iOS, а приложения под нее нет. Потому чтобы что-то настроить, надо ставить эмулятор Андроида на комп. Держать эмулятор ради того, чтобы что-то когда-то настроить такое себе...

RadioLed2000, 2020-12-17:
Согласен. В этом случае это будет самая крутая прошивка с самым обширным управлением по MQTT. Только наверное сразу стоит договориться со спецом по HA, чтоб сделал пример по полноценной интеграции для wiki

tarasifua, 2020-12-17:
> А мне вот опять же нужно полнофункциональное приложение.
Имеется ввиду приложение на Андроид с управлением через MQTT?

vvip-68, 2020-12-17:
> Имеется ввиду приложение на Андроид с управлением через MQTT?

Мне без разницы. Может и приложение под андроид с управлением через MQTT
Может какая-то панель (MQTT-panel) на телефоне, которая это умеет.
Я слабо представляю себе что такое HomeAssistant, как он выглядит и какими возможностями обладает.
Напишете инструкцию по интеграции, или еще лучше - готовую интеграцию, которую можно повторить и использовать - прекрасно.

tarasifua, 2020-12-17:
Интеграцию я выложу, когда доделаю. Но в Вашем случае она не будет применима. Home Assistant надо устанавливать на какой-то сервер, типа Raspberry Pi, и оттуда уже можно управлять любыми устройствами или через его веб-интерфейс или через его приложение

vvip-68, 2020-12-17:
Печалька...

RadioLed2000, 2020-12-17:
Я раньше использовал приложение MQTT Dash для управления устройствами, пока на HA не перешёл. https://play.google.com/store/apps/details?id=net.routix.mqttdash

vvip-68, 2020-12-17:
Знаю я приложения этого типа. У меня линейная гирлянда под таким работает
https://github.com/vvip-68/NodeMCU-MQTT-WS2812

Но это такой геморрой все это настраивать на телефоне, писать куча JavaScript`a...
И интерфейс не сильно разнообразен и функционален... 
В общем - так себе решение для проекта с более чем двумя кнопками "включить" и ""выключить"

vvip-68, 2020-12-18:
Ну, что, господа MQTT-шники. Есть у меня бета-версия прошивки с переработанным протоколом MQTT. Хотя, скорее, даже - альфа-версия. Документацию по переделкам еще не обновил - не привел в соответствие. Есть желание потестить, поковыряться? Или буду потихоньку писать доки и тестить сам?

tarasifua, 2020-12-18:
Я могу, в принципе, если вкратце напишете что поменялось и что тестить

vvip-68, 2020-12-18:
Вкратце если...
Частично поменялось API - обновленный файл в файлах проекта - api,md - убраны/изменены некоторые команды, связанные с текстом бегущей строки
На сайт еще не выгружал
Из MQTT - убрана галочка "Использовать префикс" 0 вместо нее поле prefix 30 симв - префикс, который будет добавляться к топику. Старый метод соответствовал правилам только одного сервера и к тому же mqtt.by уже не подходил - поэтому теперь префикс задается вручную. Приложение для смартфона поменялось тоже

В настройках MQTT появилась галочка - пакетный режим. Это как раз для медленных серверов - все состояние параметров отправляется пакетом в виде одной длинной строки. Если сервер свой быстрый - галку снять, задержку отправки поставить в 0

Добавлено последняя воля - когда прошивка отключается - сервер посылает топик с сообщением "offline" type "hard"
При соединении - "online" с типом "hard"/ При программном вкл/выкл - тип "soft"

Если режим не пакетный - все измененные параметры отправляются в соответствующие топики сразу же и по одному. В мониторе порта видно что куда.

При старте программы отправляется полный набор (и с флагом retain) - далее - только измененные

[176 more replies]

closed 2020-12-27 by vvip-68