vvip-68/GyverPanelWiFi #63: Эффект Салюта & Жирный шрифт & Доп.Эффекты

MaiklNem, 2020-12-15:
Добрый день! 
Спасибо за ваш огромный труд и такую крутую доработку проекта alexgyver. В этом году своевременно заказал из Китая комплект светодиодов и контроллеров NodeMCU. Собрал и разместил на окнах балконов две гирлянды. Одну размером 20x10 (ШxВ), вторую 45x10 (ШxВ). Прошивка Ваша стала без проблем, неделю почитав Ваши инструкции, комментарии к коду, вопросы и ответы других пользователей, с минимально необходимыми настройками разобрался и настроил гирлянды под свои требования. Пока у меня есть к вам только несколько вопросов и пожеланий. 
1. По ссылке https://www.youtube.com/watch?v=m_-9sD2Bofw в гирлянде есть два эффекта, которые хотелось бы увидеть и в составе вашей прошивки: это Салют и вывод текста в бегущей строке жирным шрифтом (возможно это реализовано у Вас, тогда просто подскажите где это настроить).
2. Текст бегущий строки у меня выводится правильно на черном фоне. Только в приложении для этого надо поставить галочку выводить поверх всех эффектов, в самом скрэйтче я не нашел где она ставится. Но судя потому что она записывается в eeprom, наверное на этой же вкладке должна и устанавливаться. Подскажите пожалуйста где именно нужно внести изменения, чтобы не устанавливать галочку каждый раз в приложении.
3. Если я создам несколько дополнительных эффектов в Jinx, смогу ли я их записать в контроллер без наличия у меня модуля для SD карт. Ведь в используемом мной контроллере NodeMCU еще остается свободное место в памяти.

vvip-68, 2020-12-15:
@MaiklNem
1. Все эффекты из этого ролика (салют и жирный шрифт и много других) сгенерированы в программе Jinx!, сохранены на SD карту и проигрываются оттуда. В прошивке есть поддержка эффектов с SD-карты и в wiki - инструкция как в настроить Jinx, сгенерить эффекты и поместить из на карточку, чтобы прошивка могла их ситать и воспроизводить. Сама программа Jinx! тоже есть в репозитории в папке Tools
2. Галочка в приложении устанавливается и сохраняется в EEPROM. Не понимаю зачем ее нужно устанавливать каждый раз. Ее состояние запоминается в постоянной памяти и после перезагрузки микроконтроллера восстанавливается оттуда.
Но если вы именно хотите имя переменной - в a_def_soft.h - textOverlayEnabled
Только вам придется по всему скетчу пройтись и сделать то что вам нужно, когда она загружается из памяти или изменяется из программы, чтобы установить нужное вам значение вместо считанного
3. Памяти в котроллере ESP (в файловой системе - 4 мегабайта максимум. Часть из них используется для собственных нужд контроллера. Эффект из Jinx!  (только один эффект!) занимает от 80 мегабайт и больше. Сможете вы такой объем поместить в контроллер?

closed 2020-12-15 by vvip-68

MaiklNem, 2020-12-15:
Большое Вам спасибо за оперативный и подробный ответ!
По вопросам 1 и 3 все понятно. Буду заказывать модуль для поддержки работы с SD картами. Только жаль, что в этом году он уже ко мне не успеет приехать и придется довольствоваться теми эффектами которые присутствуют в прошивке, без Новогоднего салюта.
По вопросу 2 небольшое уточнение: для того что бы записать и сохранить настройки подключения к WiF и поднятия своей точки доступаi пришлось в редакторе прошивок Arduino включить режим полной перезаписи "All Flash Contens" (по другому почему то не получалось), поэтому eeprom у меня каждый раз переписывается и приходится каждый раз ставить галочку в приложении. Почему значение переменной надо менять в a_def_soft.h, ели она записывается в eeprom.  Нельзя ли его прописать и заменить сразу во вкладке eeprom?
Вопрос 4: Не получается прописать статический адрес IP для контроллера. Почему-то когда прописываю его здесь WiFiUDP udp;  byte   IP_STA[]   = {192, 168, 100, 100};
то DHCP-сервер присваивает контроллеру все-равно свой адрес. Что я делаю не так?

vvip-68, 2020-12-15:
1 и 3 - а рядом в городе магазина нет который всеми этими ардуинными хохоряшками торгует? Где живешь то?

3 и 4 - все настройки подключения к сети задаются в программе и сохраняются в EEPROM. Оттуда же они загружаются при старте прошивки. То есть то, что прописано в скетче (вот там, например где ты в Вопрос 4 написал) - используется только при первой загрузке скетча в микроконтроллер. Все другое время они игнорируются. И там вроде даже в комментариях это все написано. "При первой прошивке" определяется так - из ячейки 0 EEPROM считывается байт и сравнивается с константой EEPROM_OK. Если значения не совпадают - прошивка считает что EEPROM не содержит валидных значений и инициализирует его значениями по умолчанию. Также в ячейе 0 оказывается не значение EEPROM_OK, если заливаешь с Erase Flash - all content.

> Почему значение переменной надо менять в a_def_soft.h, ели она записывается в eeprom. Нельзя ли его прописать и заменить сразу во вкладке eeprom?

А оно еще где-то используется. И если его прописать на в основной области, а в EEPROM - компилятор его просто не находит. По крайней мере в одной из версий было так. Да и на мой взгляд логичнее хранить ключевые значения преимущественно в одном месте скетча, чем разбросанным по разным вкладкам. Большинство пользователей и основных то комментариев не читает, что уж говорить о т ом, чтобы они весь скетч просмотрели, вникли где что лежит и для чего...

MaiklNem, 2020-12-15:
Живу в Могилеве (Беларусь). Магазины с радиокомпонентами есть в городе, а вот комплектующих для Arduino я там не встречал. Попробую поискать в Интернет магазинах Минска, возможно там найду и тогда успею до НГ сделать и свои собственные эффекты.
По  вопросу 4, ответ так и не понял. Статический адрес я задаю строкой byte IP_STA[] = {192, 168, 100, 100} в закладке a_def_soft.h и после чего при записи скрэйтча в контроллер он записывается в EEPROM, при этом в ячейке 0 EEPROM оказывается не значение EEPROM_OK. При первой перезагрузке контроллера он считывает все параметры с EEPROM, в том числе и статический адрес. Но по факту IP контроллеру присваивает dhcp-сервер роутера. Может стоит попробовать задать этот адрес на прямую в закладке eeprom, командой saveStaticIP(IP_STA[192.168.100.100])?

vvip-68, 2020-12-15:
Ничего не понял, но работает так, как описано выше. Именно этот адрес и записывается в EEPROM при первом старте и в ячейку 0 - тоже записывается 192.168.100.100 как указано. Но ведь одного IP мало - нужно еще имя сети SSID  и  пароль к ней. Далее этот адрес используется при подключении к маршрутизатору. Почему маршрутизатор не принимает и дает другой - смотри в логах маршрутизатора что за запросы там идут.
Вообще - способ какой-то извращенный. На моей памяти им никто никогда не пользуется - ты первый.
Все подключаются к контроллеру когда он точку доступа создает и там уже настраивают параметры входа в сеть, в т.ч. и Iжелаемый IP адрес.

RadioLed2000, 2020-12-15:
>Буду заказывать модуль для поддержки работы с SD картами. Только жаль, что в этом году он уже ко мне не успеет приехать и придется довольствоваться теми эффектами которые присутствуют в прошивке, без Новогоднего салюта.

В этом модуле нет ничего ничего особого. Просто соединяются контакты microSD карточки и пины ESPшки. Если умеете держать паяльник в руках, то сделаете такой за 10 мин. Я себе сам сделал из такой из переходника SD-microSD.

RadioLed2000, 2020-12-15:
> Вопрос 4: Не получается прописать статический адрес IP для контроллера. Почему-то когда прописываю его здесь WiFiUDP udp; byte IP_STA[] = {192, 168, 100, 100};
> то DHCP-сервер присваивает контроллеру все-равно свой адрес. Что я делаю не так?

А зачем такие сложности? Если нужно, чтоб роутер всегда выдавал один и тот же IP, то в роутере достаточно настроить привязку IP к mac-адресу ESPшки.

fastdron, 2020-12-15:
> Буду заказывать модуль для поддержки работы с SD картами.

Это не обязательно ,лови лайфхак))) https://youtu.be/pQo8lZ4CtLA
У меня именно так и работает, а модуль сд карты пришел позже ) но уже было лень переделывать)

MaiklNem, 2020-12-16:
**RadioLed2000** и **fastdron** спасибо Вам за предоставленный лайфхак, я сам как-то об этом не догадался поискать, а как оказалось все очень просто.

По моим вопросам 2 и 4 рабочее решение нашел, меня оно полностью устраивает, хотя на самом деле можно сделать и так как предложил **RadioLed2000**. Но я сделал по своему прописал в закладке EEPROM две строки:
saveStaticIP(192, 168, 100, 100); // Запись статического IP адреса
saveTextOverlayEnabled(true); // Вывод бегущей строки поверх всех эффектов (галочка которую ставится в приложении).

Появился Вопрос 5:  Для проверки вывода бегущей строки по событию добавил следующий текст:
textLines[18] = "До начала рабочего дня осталось {P8:00#19#180#60}{B#000000}";
textLines[19] = "-Успешного рабочего дня!!!{B#000000}";
Макрос отработал исправно, но не так как я предполагал. Я хотел бы что бы за 3 минуты появилось сообщение "До начала рабочего дня осталось" и после по центру моей гирлянды просто пошел обратный отсчет секунд до наступления события. А по факту у меня просто все три минуты бежит бегущая строка с текстом. Можно ли это как-то исправить и реализовать отображение обратного отсчета секунд? Аналогичное потом хочу сделать и до события "Наступление нового года".

vvip-68, 2020-12-16:
> Макрос отработал исправно, но не так как я предполагал. Я хотел бы что бы за 3 минуты появилось сообщение "До начала рабочего дня осталось" и после по центру моей гирлянды просто пошел обратный [...]

Всё в ваших руках, все исходные коды открыты. Дописывайте алгоритм как это нужно именно вам.

MaiklNem, 2020-12-16:
> Всё в ваших руках, все исходные коды открыты. Дописывайте алгоритм как это нужно именно вам.
У Вас опыта больше чем у меня в этом вопросе, поэтому хотел получить от Вас совет как это все же лучше сделать: анимацией записанной на SD карту, либо через gif файл и вкладку bitmap. А может уже это у кого-то это уже реализовано, был бы очень благодарен если поделитесь.

vvip-68, 2020-12-16:
В любом случае вам это придется реализовывать. Не думаю, что есть готовые решения для столь специфических запросов.
Гиф-анимация точно не подойдет - прошивка не умеет работать с гифками. Эффект "Анимация" имеет прописанные кадры с изображениями - вам придется нарисовать 180 полноэкранных кадров на что не хватит никакой памяти.  Ролик на SD-карту конечно сделать можно, но - решить вопрос насколько точно отсчет будет совпадать с реальным временем и написать год, который после показа пробежавшей строки включал бы именно этот эффект SD-карты.

То есть - в любом случае - писать код, но код, написанный просто как обратный таймер будет и менее ресурсоемок и более прост и эффективен чем остальные решения.

MaiklNem, 2020-12-16:
> То есть - в любом случае - писать код, но код, написанный просто как обратный таймер будет и менее ресурсоемок и более прост и эффективен чем остальные решения.

Спасибо за совет **vvip-68** , буду разбираться с кодом, 180 секунд мне в принципе и не надо, достаточно будет и 60. Но если написать код, то тогда можно будет выводить любое количество секунд. 
И все же если у кого-то реализовано уже подобное решение, прошу поделиться, буду очень признателен Вам. 
А также если у кого есть готовый эффект салюта для записи на SD карту, и не жалко, то пришлите пожалуйста файл *.out с ним (и другими интересными эффектами) на мой потовый ящик nemirovv@yandex.ru.

vvip-68, 2020-12-16:
> А также если у кого есть готовый эффект салюта для записи на SD карту, и не жалко, то пришлите пожалуйста файл *.out с ним (и другими интересными эффектами) на мой потовый ящик nemirovv@yandex.ru.

Этот файл крепко-накрепко зависит от размеров матрицы, а также от угла подключения и направления ленты из этого угла. Так что придется самому делать...

fastdron, 2020-12-16:
> есть готовый эффект салюта для записи на SD карту, и не жалко, то пришлите пожалуйста файл *.out

Готовый вам не подойдёт, единственное чем сможем помочь - скинуть примерные параметры jinx для создания эффекта салюта, у меня их три вида разных вышло и все смотрятся шикарно.

vvip-68, 2020-12-16:
@fastdron - там типовой салют из Jinx!-примеров - примерно на 0:27
jnx-файлы - да, интересуют.

MaiklNem, 2020-12-16:
> jnx-файлы - да, интересуют.

Интересуют, буду благодарен если вышлите мне на почту.