vvip-68/GyverPanelWiFi #65: Сенсорная кнопка

mcvall, 2020-12-18:
Кнопка подключена на D4, на плате постоянно горит синий диод, на касания реагирует, но ничего не происходит. Телефон плату не видит. Как только D4 пин отключаю физически, wifi подключается, плата управляется как положено. При подключеном wifi цепляю кнопку на d4, реагирует нормально. Также при подключенной кнопке не загружается скетч, говорит, что плата не подключена, как только d4 отключаю, все прошивается. Так и должно быть?

vvip-68, 2020-12-18:
Да, всё верно. На платах Wemos и NodeMCU пин D4 соответствует GPIO2.
Пин GPIO2 совместно с пинами GPIO0 и GPIO15 определяет режим загрузки микроконтроллера.
В зависимости от комбинации, подаваемой на эти пины в момент подачи напряжения питания контроллер переводится в режим программирования, а не в режим выполнения программы. Комбинацию  состояний ищите в интернете.

Таким образом подключенная вами кнопка на пин D4 (GPIO2) переводит ESP8266 в режим ожидания загрузки программы во Flash-память. Микропроцессор не работает, программа не выполняется. В общем - все как вы описали.

[image]

closed 2020-12-18 by vvip-68

mcvall, 2020-12-18:
Спасибо, но мне что делать?)))
Хочу спрятать плату в корпус, и теперь мне, для того чтобы подключить плату к Wifi и для перепрошивки нужно отпаивать провод на D4?

vvip-68, 2020-12-18:
Вообще поменять D4 на другой пин. С припаянной на D4 сенсорной кнопкой у вас плата даже заводиться не будет. Будет при подачи питания переходить в состояние ожидания прошивки.

mcvall, 2020-12-18:
На D6 нормально будет? И поменять в скетче d4 на d6?

vvip-68, 2020-12-18:
Нормально. Разумеется.

mcvall, 2020-12-18:
Кнопка заработала, теперь когда в приложении ставлю галочку на автоматический режим, все режимы отработали и диоды отключаются, приходится по-новой включать через кнопку. Как сделать, чтоб не выключалось и по кругу работало?

vvip-68, 2020-12-18:
Вообще-то в прошивке не предусмотрено такое вот отключение. Она всегда работает пока есть питание.
Возможно у вас какая-то относительно древняя версия.
Посмотрите файл a_def.soft.h строки 84-88
Там после объявления эффектов должна быть следующая конструкция:
#if (USE_SD == 1)
#define MAX_EFFECT              45         // количество эффектов, определенных в прошивке
#else
#define MAX_EFFECT              44         // количество эффектов, определенных в прошивке
#endif
Если ее нет - добавьте. Если есть, но отключается - не знаю. У меня такого не наблюдается.
А исправлять "ошибки" которые я не могу повторить - я не умею.

mcvall, 2020-12-18:
Версия самая свежая, сегодня скачал.

vvip-68, 2020-12-18:
Тогда не знаю. Если удастся повторить и пойму - поправлю. Пока я такого не наблюдаю.
Или сможете рассказать как вы это получаете

mcvall, 2020-12-18:
84-88 строки так же прописано. Чтобы было быстрее, я включил смену эффектов на 1сек, вот они все быстро отработали и панель отключается, нажимаю на кнопку и все по новой

mcvall, 2020-12-18:
Да, подчеркну - wifi при этом не подключен (роутер) только приложение напрямую.

vvip-68, 2020-12-18:
У меня такого не наблюдается, но судя по коду - была возможная проблемя. Я поправил

vvip-68, 2020-12-18:
Да, сегодняшняя версия 1.08 - глубокая переработка для правильного использования MQTT.
Из-за глубокой переработки - может что-то еще где-нибудь сломаться...

mcvall, 2020-12-18:
Скачал, залил, то же самое. Заканчиваются эффекты, по окончании один короткий блик синего светодиода на плате и плата отключается, в монитор порта тоже перестает слать данные, кнопку нажал, все по новой.

vvip-68, 2020-12-18:
Тогда не смогу помочь. У меня такого не наблюдается. Я даже представить не могу вариант чтобы оно вот так работало.
Если я не могу это наблюдать и не могу это воспроизвести - я не могу это изменить.

Что в момент моргания синего светодиода в мониторе порта пишет?

vvip-68, 2020-12-18:
А без твоей сенсорной кнопки такое происходит? Может обычную тактовую если поставить?
Я в свое время не смог победить эту сенсорную кнопку. Она живет своей жизнью.
И может нажаться самостоятельно когда ей вздумается.
Тогда произойдет короткий "блинк" - прошивка воспринимает как однократное нажатие, а однократное нажатие кнопки - программное выключение устройства...

mcvall, 2020-12-18:
Сейчас залил версию 1.06, всё нормально, не отключается, только команды кнопка не выполняет, при любых вариантах нажатия отключает матрицу. У меня две платы рядом, и две кнопки соответственно, везде одинаково. В крайней прошивке в мониторе порта постоянно идёт создание точки доступа, потом все отключается как и ранее писал. На версии 1.06 в мониторе как положено передаются пакеты, т.е. я вижу, что сработал тот или иной эффект. Вот так пока...

mcvall, 2020-12-18:
Понаблюдал немного за поведением кнопки. Да, у них фантомные нажатия происходят. Сама по себе работает, это видно по красному диоду на ней. Поменяю на тактовые, отпишусь.

mcvall, 2020-12-21:
Такс, заменил кнопки, все работает. Надо бы в описание добавить, что есть траблы с сенсорными кнопками.

mcvall, 2020-12-21:
Я тут решил дисплей подключить, но что-то не компилится, что может быть?

In file included from C:\Arduino\GyverPanelWiFi-master\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.08\GyverPanelWiFi_v1.08.ino:20:0:
a_def_soft.h:504:23: error: 'CLK' was not declared in this scope
 TM1637Display display(CLK, DIO);
                       ^
a_def_soft.h:504:28: error: 'DIO' was not declared in this scope
 TM1637Display display(CLK, DIO);
                            ^
exit status 1
_**'CLK' was not declared in this scope**_

vvip-68, 2020-12-22:
Недостаточно просто поставить USE_TM1637 1
Нужно двумя-тремя стровчrfми ниже раскомментировать #define CLK

mcvall, 2020-12-22:
> 
> 
> Недостаточно просто поставить USE_TM1637 1
> Нужно двумя-тремя стровчками ниже раскомментировать #define CLK

Во вкладке a_def_soft.h нету #define CLK,
#define CLK есть только во вкладке a_def_hard.h
В моем случае с гирляндой на балкон это 253 строка a_def_hard.h
/*
250 #define SRX D4                // не используется, но требуется для компиляции скетча (USE_MP3==1)
251 #define STX D3                // не используется, но требуется для компиляции скетча (USE_MP3==1)
252 #define DIO D5                // D5 TM1637 display DIO pin - не используется, но требуется для компиляции скетча (USE_TM1637 == 1)
253 #define CLK D7                // D7 TM1637 display CLK pin - не используется, но требуется для компиляции скетча (USE_TM1637 == 1)
*/

Праввильно?

mcvall, 2020-12-22:
Всё правильно, работает.