vvip-68/GyverPanelWiFi #67: ESP 32: Прошивка, настройка и мерцание

CrazyGhostRider, 2020-12-19:
Здравствуйте
1) Добавьте, пожалуйста, в вики настройки менеджера плат для ESP32
(https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json)

2) Как настроить стандартные настройки эффектов? (Работа в демо режиме, скорость и т.п.)

3) Так как использую БП 30А от ПК, то подключил питание ESP32 через диод.
Но при активных эффектах на всю матрицу происходит одновременное мерцание всей матрицы. (будто резко яркость повышается) 

Сейчас матрица стабильно работает на Arduino Mega. 
Матрица 26х13

vvip-68, 2020-12-19:
1. Эта ссылка вот прямо в основном файле в строке 10. Но кто ж читает комментарии то?

2. Если имеется в виду, чтобы вот после заливки новой прошивки все параметры уже имели нужные значения - в файле eeprom.ino есть процедура SaveDefaults(). В ней нужно сохранить все нужные вам параметры как значения по-умолчанию. Например в строках 319-335 для некоторых эффектов отключается оверлей часов и бегущей строки. В строках ниже настраиваются скорости прокрутки часов и текстов, ниже - параметры некоторых эффектов, ниже - текстовые строки, и так далее. В строках 249-253 - параметры "Вариант", "Вариант2" и контрастность для эффектов.
В общем - по образу и подобию.

3. - Правильно ли подключен диод? 
    - Даже при включении диодов иногда требуется подстройка напряжения блока питания. 
    - Имеется по крайней мере два случая матриц с тонкими полупрозрачными диодными нитями в которых диоды как в капельке смолы и святят в обе стороны - в них вообще не удается полностью устранить мерцание - видимо особенности китайских диодов.

Вообще - всё что возможно сделать - описано в wiki - https://github.com/vvip-68/GyverPanelWiFi/wiki/%D0%A2%D1%80%D0%B8-%D1%88%D0%B0%D0%B3%D0%B0-%D0%BA-%D1%83%D1%81%D0%BF%D0%B5%D1%85%D1%83.-%D0%9D%D0%B5%D0%BA%D0%BE%D1%82%D0%BE%D1%80%D1%8B%D0%B5-%D0%BE%D1%81%D0%BE%D0%B1%D0%B5%D0%BD%D0%BD%D0%BE%D1%81%D1%82%D0%B8-%D0%BD%D0%B0%D1%81%D1%82%D1%80%D0%BE%D0%B5%D0%BA...

4. Arduino Mega имеет 5-вольтовую логику управляющих сигналов, которая полностью соответствует уровням сигналов, используемых светодиодной лентой. Контроллеры ESP8266 и ESP32имеют сигналы логических уровней 3.3 вольта, что недостаточно для согласования с 5-вольтовой логикой диодов. Именно поэтому используются трюки с диодом, с подбором напряжения. Более надежный способ - поставить между выходом микроконтроллера (управляющим пином на ленту) и входом ленты микросхему согласования уровня или согласователь на полевом транзисторе. Схемы представлены в описании проекта на основной странице.

closed 2020-12-19 by vvip-68

Mosrom, 2021-01-18:
Обнаружил некоторое несоответствие схемы подключения модуля платы с пином ленты. У Вас на схеме изображено подключение контакта din ленты к контакту G13 платы. В скетче настройки матриц в строке 403 указан пин "2U". Верно ли это?

vvip-68, 2021-01-18:
Нет, скорее всего. И схема и скетч, для думающих людей, способных проанализировать и сделать как правильно.

У меня несколько устройств на ESP32, каждые имеют разные пины подключения. В скетче - тот пин, для какого устройства была сборка последним у меня.  Я не в состоянии делать картинки под все варианты скетчей / устройств, или делать для себя однотипные устройства непременно соответствующие схеме (которая, кстати, не моя - кто-то из пользователей прислал для своего устройства, я выложил).

В принципе, могу удалить все схемы, ибо не получится скетч держать все время в соответствии с картинками по указанной причине.

Mosrom, 2021-01-18:
Не надо удалять, пускай думают. И так всё на блюдечке принесли🙂

Mosrom, 2021-01-18:
На плате ESP32 матрица должна работать стабильнее, особенно с большим количеством диодов.

Mosrom, 2021-02-24:
При использовании в схеме реле (без использования MOSFET) значение показателя строки "#define USE_POWER" нужно указывать 1 или 0?

vvip-68, 2021-02-24:
Если 1 - на ногу D1 будет при программном включении устанавливаться уровень 1, на выключение - 0; Впрочем, уровни задаются в a_def_soft.h константами POWER_ON и POWER_OFF
Если 0 - нога D1 не используется и на программное включение/отключение никак не реагирует.

Mosrom, 2021-02-24:
То есть в данном случае нужно устанавливать 1? Если 0, тогда реле работать не будет.

vvip-68, 2021-02-24:
Да.

Mosrom, 2021-03-06:
Как правильно обозначать номера GPIO pin в ESP32, в частности во вкладке "a_def_hard.h" в строках 172-180. Нужны ли скобки и буквы до или после номера? Никак не могу запустить. Скетч в мониторе порта запускается. Но при подключении к GPIO контактов по схеме - нет.

vvip-68, 2021-03-06:
Вот как в тех строках написано - так и можно указывать. Скобки не обязательны. Буква U тоже не обязательна (это всего лишь стандартное обозначение "unsigned" но это и так понятно. Пины - как номера GPIO. Они же на плате тоже как номера GPIO обозначены. Номера GPIO соответствуют указанной схеме, насколько я помню. Вот по этой схеме и с этими номерами пинов устройство собрано и работает. Упаковано всё в корпус - разбирать нет желания пока.

Mosrom, 2021-03-06:
Получилось запустить следующим образом. Подключил "землю" к модулю вместо pin 18 "GND" к pin 14 "GND" (см. схему ESP32_ALL).  Не знаю почему так, но всё тут же заработало. По GPIO поставил номера pin без букв, как на модуле обозначены, но в скобках. Нужно с таким подключением "земли" попробовать сделать.
 как в сетче написано.

vvip-68, 2021-03-06:
Подозреваю, что есть разные модели плат ESP32. У меня вот на моей платке - ног меньше чем на схеме.
Подключал не к номерам ножек, а к ножкам с соответствием надписям на платке. Смотрел в общем распиновку по именно моей версии ESP32. Так что, вероятно, на вашм варианте платки 18 нога - не GND

Mosrom, 2021-03-06:
Я также делал. Подключал как на плате написано. На ней 38 пинов, как на схеме. И возле 18 ноги написано GND. Но может Вы и правы. У меня их несколько. На другой проверю.

vvip-68, 2021-03-07:
Вот на такой у меня собрано. Как видите - пин-аут немного отличается от схемы. 
Но соединение - именно по номерам GPIO сделано. Работает.
[image]